OctoPrint/OctoPrint #2383: gcode analysis took very long or is wrong or wont work

TylonHH, 2018-01-25:
#### What were you doing?
I uploaded a gcode file. This file is in file list on the left. 

#### What did you expect to happen?
Then the gcode analysis should show me lenght and weight of the filament and the extruder (tool0)

#### What happened instead?
The gcode analysis wont get any results. I don't see any weight or lenghts. It's neccesary for the [filament manager plugin](https://github.com/malnvenshorn/OctoPrint-FilamentManager). See https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/73

#### Did the same happen when running OctoPrint in safe mode?
Not tested, yet.

#### Version of OctoPrint
OctoPrint 1.3.6 auf OctoPi 0.13.0 

#### Operating System running OctoPrint
Raspberry with OctoPi 0.13.0 

#### Printer model & used firmware incl. version
CoreXY with Marlin 1.1.8

#### Browser and Version of Browser, Operating System running Browser
Firefox 64bit 58.0

#### Link to octoprint.log
[https://pastebin.com/nXcTLeLC](https://pastebin.com/nXcTLeLC)

#### Link to contents of terminal tab or serial.log
serial.log

#### Link to contents of Javascript console in the browser
https://user-images.githubusercontent.com/18073299/35137409-872c00ea-fce9-11e7-944c-c1b5d4a3b158.PNG

#### Screenshot(s)/video(s) showing the problem:
see above

I have read the FAQ.

foosel, 2018-01-25:
Please provide a gcode file with which to reproduce.

TylonHH, 2018-01-25:
Here is a small one
[Pencils.gcode.txt](https://github.com/foosel/OctoPrint/files/1665215/Pencils.gcode.txt)

After uploading and waiting a while it still show this
[image]

Here are all the error of the console
```
Less has finished and no sheets were loaded.
less.min.js:13:12757
Starting dependency resolution...
packed_core.js:11569:9
... dependency resolution done
packed_core.js:11679:9
Initial application setup done, connecting to server...
packed_core.js:11982:9
Connected to the server
packed_core.js:9965:9
Finalizing application startup
packed_core.js:11960:13

Error calling fromCurrentData on view model PreheatViewModel : PreheatViewModel/self.updateButton@http://192.168.178.44:88/static/webassets/packed_plugins.js?109b8a9b:3805:25
PreheatViewModel/self.fromCurrentData@http://192.168.178.44:88/static/webassets/packed_plugins.js?109b8a9b:3824:47
callViewModelIf@http://192.168.178.44:88/static/webassets/packed_core.js?3c21de80:11188:17
callViewModelsIf/<@http://192.168.178.44:88/static/webassets/packed_core.js?3c21de80:11137:13
Pn@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10658:530
ur/<@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10679:66
callViewModelsIf@http://192.168.178.44:88/static/webassets/packed_core.js?3c21de80:11135:5
callViewModels@http://192.168.178.44:88/static/webassets/packed_core.js?3c21de80:11129:5
DataUpdater/self._onCurrentData/<@http://192.168.178.44:88/static/webassets/packed_core.js?3c21de80:9982:13
DataUpdater/self._ifInitialized@http://192.168.178.44:88/static/webassets/packed_core.js?3c21de80:10080:13
DataUpdater/self._onCurrentData@http://192.168.178.44:88/static/webassets/packed_core.js?3c21de80:9981:9
OctoPrintSocketClient.prototype.propagateMessage/<@http://192.168.178.44:88/static/webassets/packed_client.js?aedb4fd9:380:17
Pn@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10658:530
ur/<@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10679:66
OctoPrintSocketClient.prototype.propagateMessage@http://192.168.178.44:88/static/webassets/packed_client.js?aedb4fd9:379:13
onMessage/<@http://192.168.178.44:88/static/webassets/packed_client.js?aedb4fd9:470:17
Zt/<@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10675:23
_t@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10663:392
Vt/<@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10674:369
ur/<@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:10679:73
onMessage@http://192.168.178.44:88/static/webassets/packed_client.js?aedb4fd9:469:13
[5]</EventTarget.prototype.dispatchEvent@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:21977:5
[14]</</SockJS.prototype._transportMessage/<@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:22690:11
[14]</</SockJS.prototype._transportMessage@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:22688:9
[3]</EventEmitter.prototype.emit@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:21893:5
WebSocketTransport/this.ws.onmessage@http://192.168.178.44:88/static/webassets/packed_libs.js?eebed41b:24764:5
packed_core.js:11201:13

Going to bind 36 view models...
packed_core.js:11844:13
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js:11897:33
... binding done
packed_core.js:11936:13
Application startup complete
packed_core.js:11956:13
Terminal: Detected slow client (needed 656ms for processing new log data), disabling fancy terminal functionality
packed_core.js:5701:13
Terminal: Client speed recovered, enabling fancy terminal functionality
packed_core.js:5711:17
User tylon logged in
packed_core.js:2267:25
```

rongith, 2018-01-30:
@TylonHH the same for me.
If I load gcode through gcode viewer, I got quickly "Estimated total print time".
But Estimated layer height: 0.49mm is wrong as I have 0.15, 0.2 and 0.25 layer height in my model.
Moreover gcode viewer set my intro line at 100mm altitude.
[04500-r04g_-_15d-30-I3D1-(P)04500-vmax-CFXT-(F)ColorFabb_XT_-_chaud-(P)PMK2_-_Echelle-R0.8-1.38.5-prusa3d-win64-20180130-125131.gcode.zip](https://github.com/foosel/OctoPrint/files/1677638/04500-r04g_-_15d-30-I3D1-.P.04500-vmax-CFXT-.F.ColorFabb_XT_-_chaud-.P.PMK2_-_Echelle-R0.8-1.38.5-prusa3d-win64-20180130-125131.gcode.zip)
[image]

foosel, 2018-01-30:
Both of you please test in safe mode to rule out any plugin interfering here.

rongith, 2018-02-05:
Seeing how a reboot solve part of the issues, I thought that changing my way of uploading files, which is not conventional, would solve them.
Before, I exported directly from slic3r to Octoprint then move the file with ftp client from ~/.octoprint/uploads to a convenient subdirectory. I found that way quicker. But it seems it breaks file management of Octoprint.
Then I decided to move the file using FileManager addon (https://github.com/Salandora/OctoPrint-FileManager) instead of a ftp client. And I had no more issues since. Now I just export a gcode file from slic3r to a local directory and then upload that gcode to an active subdirectory with a browser. And that is just nice.

foosel, 2018-03-07:
Cannot reproduce this and can't see anything odd here... 

[image]

```
2018-03-07 17:05:58,440 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Pencils.gcode
2018-03-07 17:05:58,443 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Pencils.gcode
2018-03-07 17:06:56,231 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Pencils.gcode finished, needed 57.79s
```

Less than a minute for analysis of a 8.1MB GCODE file on a Pi3 doesn't strike me as unusual.

[image]

```
2018-03-07 17:09:48,798 - octoprint.filemanager.analysis - INFO - Starting analysis of local:04500-r04g_-_15d-30-I3D1-(P)04500-vmax-CFXT-(F)ColorFabb_XT_-_chaud-(P)PMK2_-_Echelle-R0.8-1.38.5-prusa3d-win64-20180130-125131.gcode
2018-03-07 17:09:48,801 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/04500-r04g_-_15d-30-I3D1-(P)04500-vmax-CFXT-(F)ColorFabb_XT_-_chaud-(P)PMK2_-_Echelle-R0.8-1.38.5-prusa3d-win64-20180130-125131.gcode
2018-03-07 17:11:23,749 - octoprint.filemanager.analysis - INFO - Analysis of entry local:04500-r04g_-_15d-30-I3D1-(P)04500-vmax-CFXT-(F)ColorFabb_XT_-_chaud-(P)PMK2_-_Echelle-R0.8-1.38.5-prusa3d-win64-20180130-125131.gcode finished, needed 94.95s
```

And neither does 90s for a 12.6MB GCODE file, on a Pi3 as well.

Both finish just fine and the data gets pushed to the frontend as you can see.

Remember, OctoPrint has to run through all of the lines in there, parse them, calculate the movement duration, extrusion length etc and all that together. That takes a while, it obviously can't be instantaneous doing that kind of work, and the more the machine has to do on top of things, the longer it can take (e.g. analysis triggered during startup while the whole application is still booting will prolly run slower).

69291954, 2018-06-05:
had possible the same issue: 

2018-06-04 22:06:12,215 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Ender3_PETG_Spider_Elbow.gcode
2018-06-04 22:06:12,221 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Ender3_PETG_Spider_Elbow.gcode
2018-06-04 22:06:12,230 - octoprint.plugins.autoselect - INFO - Selecting Ender3_PETG_Spider_Elbow.gcode on local that was just uploaded

started 2 (?) python processes running with 100% CPU, not finishing within 6 hours, bringing the PI to 80 degree. after removing all files and uploading Ender3_PETG_Spider_Elbow.gcode  again i could not reproduce the behaviour.

closed 2021-03-31 by LazeMSS