OctoPrint/OctoPrint #2384: [Request] Ability to add notes from the web interface on files printed

NemesisDude, 2018-01-25:
Ability to add notes from the web interface on files printed.
In a tab or within the file list dropdown?

It would be helpful for myself as well as others I think to have a central place for notes on prints, especially when they fail so it's easier to track changes to the gcode and such, also for the successful prints 

Thank you Thank You Thank you!!! I can't say it enough for the time and effort you have put in to this it is a work of art!!
I hope to contribute in some way someday.
Thanks Again
Glen

OctoPrint 1.3.6 running on OctoPi 0.14.0
I have read the FAQ.

ntoff, 2018-01-26:
more or less a duplicate of https://github.com/foosel/OctoPrint/issues/519

foosel, 2018-02-12:
Agreed, closing as duplicate.

closed 2018-02-12 by foosel