OctoPrint/OctoPrint #2386: Feed rate display not persistent across browser close/reopen

phisher1, 2018-01-26:
If you set a feed rate and close your browser, then reopen and go back to Octoprint, the current feed rate is not displayed and it displays 100% again.

phisher1, 2018-01-26:
Please remove this issue. I will open in OctoPi repo.

ntoff, 2018-01-27:
It's an octoprint "issue" so this tracker is the correct place for it. I started a brainstorming thread a while ago https://github.com/foosel/OctoPrint/issues/2103

btw you're able to close issues you create (in case you don't want them open anymore), you don't need to ask someone else to close them.

foosel, 2018-02-06:
This sadly works as intended. Problem is that the firmware doesn't provide any kind of way to query to currently set feed rate or flow rate modifiers (it's a write-only interface so to speak), and since those values also can be modified on the printer controller itself outside of OctoPrint's reach, there's absolutely no way to reliably determine the current value. Hence the fallback to 100%.

Can't change this unless firmwares introduce a means to query feed rate and flow rate modifiers (and also feed rate, fan speed and currently selected tool while we are at it).

closed 2018-02-06 by foosel