OctoPrint/OctoPrint #2387: UnicodeDecodeError accessing Octoprint from browser

mklhmnn, 2018-01-27:
After having not used my Raspberry-Pi with Octoprint 1.3.5 for a couple of weeks, I restarted it today, but it told me about "bad marshal data" somewhere in /home/pi/oprint/*Babel*. Google told me to clear .pyc files, so I invoked
```
sudo service octoprint stop
find ~/oprint -name '*.pyc' -delete
rm ~/.octoprint/logs/octoprint.log
```
Now, after running
```
sudo service octoprint restart
```
it starts up fine. Log output:
```
2018-01-27 09:59:21,864 - octoprint.server - INFO - ******************************************************************************
2018-01-27 09:59:21,868 - octoprint.server - INFO - Starting OctoPrint 1.3.5 (master branch)
2018-01-27 09:59:21,872 - octoprint.server - INFO - ******************************************************************************
2018-01-27 09:59:22,226 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-01-27 09:59:32,527 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 1 mixin implementations, 2 hook handlers
2018-01-27 09:59:32,701 - octoprint.server - INFO - Intermediary server started
2018-01-27 09:59:32,706 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-01-27 09:59:34,864 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 1 mixin implementations, 2 hook handlers
2018-01-27 09:59:35,125 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-01-27 09:59:35,139 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-01-27 09:59:50,262 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2018-01-27 10:00:05,248 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2018-01-27 10:00:05,258 - octoprint.plugin.core - INFO - Initialized 1 plugin implementation(s)
2018-01-27 10:00:05,270 - octoprint.plugin.core - INFO - 7 plugin(s) registered with the system:
| !Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/announcements
| !Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/cura
| !Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/pluginmanager
| !Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/virtual_printer
2018-01-27 10:00:05,314 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-01-27 10:00:05,322 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-01-27 10:00:06,682 - octoprint.server - INFO - Shutting down intermediary server...
2018-01-27 10:00:06,999 - octoprint.server - INFO - Intermediary server shut down
2018-01-27 10:00:07,049 - octoprint.events - INFO - Processing startup event, this is our first event
2018-01-27 10:00:07,057 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2018-01-27 10:00:07,070 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-01-27 10:00:07,083 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-01-27 10:00:07,706 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2018-01-27 10:00:07,753 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-01-27 10:00:07,797 - octoprint.plugins.pluginmanager - INFO - Looks like we are offline, can't fetch repository from network
2018-01-27 10:00:07,812 - octoprint.plugins.pluginmanager - INFO - Looks like we are offline, can't fetch notices from network
2018-01-27 10:00:08,599 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 113.52ms
2018-01-27 10:00:10,530 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 82.61ms
2018-01-27 10:00:12,540 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 81.15ms
2018-01-27 10:00:14,540 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 82.07ms
2018-01-27 10:00:16,552 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 82.86ms
2018-01-27 10:00:16,790 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-01-27 10:00:18,556 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 80.36ms
2018-01-27 10:00:19,325 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-01-27 10:00:19,407 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0-RC8 (Github)"
```
But as soon as I try to access it from my browser (Firefox, cookies deleted), I'm getting:
```
2018-01-27 10:01:21,514 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/server/views.py", line 413, in index
    response = default_view()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/server/views.py", line 356, in default_view
    now)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/server/views.py", line 424, in _get_render_kwargs
    locales = dict((l.language, dict(language=l.language, display=l.display_name, english=l.english_name)) for l in LOCALES)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/server/views.py", line 424, in <genexpr>
    locales = dict((l.language, dict(language=l.language, display=l.display_name, english=l.english_name)) for l in LOCALES)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Babel-2.4.0-py2.7.egg/babel/core.py", line 380, in get_display_name
    retval = locale.languages.get(self.language)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Babel-2.4.0-py2.7.egg/babel/core.py", line 481, in languages
    return self._data['languages']
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Babel-2.4.0-py2.7.egg/babel/core.py", line 363, in _data
    self.__data = localedata.LocaleDataDict(localedata.load(str(self)))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Babel-2.4.0-py2.7.egg/babel/localedata.py", line 111, in load
    merge(data, pickle.load(fileobj))
UnicodeDecodeError: 'utf8' codec can't decode byte 0xb8 in position 9: invalid start byte
2018-01-27 10:01:21,558 - tornado.access - ERROR - 500 GET / (127.0.0.1) 719.87ms
```
How to resolve this?

mklhmnn, 2018-02-02:
What information are missing?

foosel, 2018-02-05:
You didn't use the reporting template, but is *seems* like even though you didn't include the full log enough for a first analysis is there. In the future please just follow the template, that saves us all a lot of overhead in information collection.

From your description it looks like some data on your Pi's SD became corrupted and that's also now causing you issues. Always make sure to shutdown your Pi cleanly and do not just power it off. 

I'd suggest to first of all install the current version, because your 1.3.5 is already outdated by more than a month. I'm also not sure what instructions you followed there on how to fix your marshalling issues, but I think all you did is remove one symptom of a corrupted file system, not solve the actual problem. And you also nuked the old log that might have helped to get a better idea of what was the original problem - best not do this in the future either.

So, suggested steps: [Manually update to 1.3.6](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-update-the-octoprint-installation-on-my-octopi-image), see if that works. If it doesn't, do the following (via SSH):

```
rm -r ~/oprint
virtualenv ~/oprint
source ~/oprint/bin/activate
cd ~/OctoPrint
python setup.py clean
python setup.py install
```

That should hopefully get rid of any corrupted files in your `oprint` virtual environment and that might make things working again. But in general, a corrupted file system is a bad sign, and other stuff might break (or already have broken, just not noticeably). So just doing a [backup](https://github.com/foosel/OctoPrint/wiki/FAQ#i-want-to-flash-a-new-image-how-do-i-backup-my-octoprint-settings) and setting up a fresh install might be the safer option.

kantlivelong, 2018-03-09:
Closing due to lack of information and inactivity.

closed 2018-03-09 by kantlivelong