OctoPrint/OctoPrint #2389: [Request] ability to retrospectively pause at layer

BETLOG, 2018-01-28:
I often need a way to insert a pause AFTER a print is started.

Rather than realising i forgot, and then waiting for the right layer to manually push pause I really need a means of  automatically detecting a layer I specify.

Console interpretation and string substitution (issued from the console command line) seems like something that might already exist, but I have no idea how to even investigate that possibility or the required syntax.

The MultiColor plugin offers this basic functionality, but prevents any interaction after a print has started, so that's not very useful. No doubt such an interpreter is the basis for most of octo plugins, so maybe the core functionality of OctoPrint's console would benefit from this being built into the main or advanced interface.
ie: a console field set to filter/detect [user specified string] and respond with [user specified gcode or string substitution]. All of which can be added or edited while a print is in progress.

foosel, 2018-02-12:
> A "pause at layer/height" button next to the existing 'pause' button seems like one logical option.

One key issue with that: Won't work when printing from SD.

In any case, this sounds like it would be best suited to be implemented by a plugin.

jneilliii, 2024-08-29:
I released a plugin that covers this: <https://plugins.octoprint.org/plugins/pause_management/>

closed 2024-08-29 by jneilliii