OctoPrint/OctoPrint #2392: Copy all from terminal only works once

hoegge, 2018-01-29:
----

#### What were you doing?

1. Press `Copy all` below terminal window
2. Paste result to text editor
3. Wait for more lines to be added to terminal window (e.g. during print)
4. Press `Copy all` below terminal window
5. Past text to text editor
6. Compare results - they are the same. 

When you press the copy all button more than once, additional text is not copied. 

If you reload the web-page, the first copy is good and then the subsequent are the same. So must be a web-interface bug.
#### What did you expect to happen?

That the current full text in the terminal window was copied

#### What happened instead?
The same as last copy is copied. Have tried to clear clipboard before second copy - it does copy the same tihg again.

#### Did the same happen when running OctoPrint in safe mode?
n/a

#### Version of OctoPrint
1.3.6
[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint
Octopi 0.14

#### Printer model & used firmware incl. version
n/a
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Chrome newest version (63.0.3239.132 (Official Build) (64-bit)) on PC or Mac - same result

#### Link to octoprint.log

N/A but here it is: [https://www.dropbox.com/s/cg63ehzgjeoilkc/octoprint.log?dl=0](url)

#### Link to contents of terminal tab or serial.log
[https://www.dropbox.com/s/riixtqdbc7n6ak8/serial.log?dl=0](url)

#### Link to contents of Javascript console in the browser
[https://www.dropbox.com/s/8f2yrci6q0kbzbm/192.168.8.117-1517218431929.log?dl=0](url)

foosel, 2018-01-29:
Will look into this as soon as I can. Side note: don't remove any kind that are not encoded in [ ] from the issue template or the bot will complain.

foosel, 2018-02-08:
What a stupid bug :) Thanks for the heads-up!

Fixed on `maintenance`, soon `devel`, will be released with 1.3.7.

commit b79f6693 (2018-02-08): Fix "copy all" on Terminal tab

Closes #2392
closed 2018-04-09 by foosel