OctoPrint/OctoPrint #2395: Object doesn't fit print volume

scns, 2018-01-30:
#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. Upload sliced job from Cura
2. Print > error

[image]

3. Print > error
4. Print > OK

OR 

1. Upload sliced job from Cura
2. Print > OK
3 Finish print
4. Print again > Error

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

Just start the print de first time

#### What happened instead?

[image]

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Zonestar P802QS 

Frimware i have to check

#### Browser and Version of Browser, Operating System running Browser

MAC OS, IOS 
Browser Safar

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/1678882/octoprint.log)

#### Link to contents of terminal tab or serial.log

na
#### Link to contents of Javascript console in the browser
na
The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2018-01-30:
Do you have a nozzle priming routine or something else in your gcode that goes to X -20? If so this is completely correctly behaviour if you haven't told OctoPrint about it in your printer profile ("custom bounding box"). Otherwise please provide a gcode file with which to reproduce this and the printer profile settings you have configured.

scns, 2018-01-30:
This is my priming code:

> G21        ;metric values
> G90        ;absolute positioning
> M82        ;set extruder to absolute mode
> M107       ;start with the fan off
> G28
> G29
> G92 X-20
> G92 E0 ; zero the extruded length
> G1 X40 E25 F500 ; Extrude 25mm of filament in a 4cm line
> G92 E0 ; zero the extruded length
> G1 E-1 F500 ; Retract a little
> G1 X80 F4000 ; Quickly wipe away from the filament line
> G1 Z10.0 F{travel_speed} ;move the platform down 15mm
> G92 E0  

[ZP_PWM_Controller_bottom.gcode.txt](https://github.com/foosel/OctoPrint/files/1678997/ZP_PWM_Controller_bottom.gcode.txt)

scns, 2018-01-30:
I see it. Please close the issue. 

G92 X-20

foosel, 2018-01-30:
You can close issues you opened yourself ;)

closed 2018-01-30 by foosel