OctoPrint/OctoPrint #2396: stuttering during print

AnduriI, 2018-01-30:
#### What were you doing?

Simply printing... I did try in normal and safe mode. At the moment I print the file from SD-card and it works fine.
I first started in safe mode to test something else (at around 19:47, line 205 in octoprint.log) and later tryed in normal mode (around 20:25)

#### What did you expect to happen?

#### What happened instead?

In irregular intervalls the printer just 'paused' for about 0.5 to 2 seconds. During that time the terminal window in OctoPrint did not show updates, but continued as normal afterwards.

#### Did the same happen when running OctoPrint in safe mode?

Yep, same behaviour in safe mode.

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

System installed manually with rasbian and OctoPrint (hostname set to `OctoPi`, no OctoPi-image was used)
```
pi@OctoPi:~ $ uname -r
4.9.35-v7+
pi@OctoPi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
pi@OctoPi:~ $
```

#### Printer model & used firmware incl. version

CR-10mini with stock firmware

#### Browser and Version of Browser, Operating System running Browser

Windows 10 with Firefox 58.0.1

#### Link to octoprint.log

https://gist.github.com/AnduriI/7eae55386d4ee82f42641522da234ffa#file-octoprint-log

#### Link to contents of terminal tab or serial.log

https://gist.github.com/AnduriI/7eae55386d4ee82f42641522da234ffa#file-serial-log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

kantlivelong, 2018-01-31:
Hi @AnduriI,

1. What Pi model are you using?
2. Are you using a webcam? If so, which model?

AnduriI, 2018-01-31:
Hey,

I use a RPi3. It is not only running OctoPrint, but htop never shows more than 25% CPU (and these are generated by OctoPrint).
I also use a webcam, a Raspberry Cam (v2.1 I think). I enlarged the cable, but I don't think that should have an influence (just wanted to add for answering your question as completely as possible).

foosel, 2018-01-31:
From your serial log it looks like your firmware simply hesitates to send out an ok every couple of lines for 5s or so. And as long as your firmware doesn't tell OctoPrint it's ok to send the next line, it won't. And if the firmware continues to process its planned moves during that silence treatment, things may stutter. 

It's this something that started recently? If so, what changed since then on printer and pi?

AnduriI, 2018-01-31:
It started recently, but I can not state changes. I did had a few successful small prints (I don't remeber any stuttering) and started some larger things. During this print OctoPrint just completely stopped serial connection to my CR-10mini. (I saved the logs at that time for later inspection. I uploaded them just in case it will give you a hint for the stuttering https://gist.github.com/AnduriI/d5599ab37981ead3ecc0be7e71d26236. I copied the output of terminal log manually. Interesting part starts at line 220. The serial crash was at about 22:00, in octoprint.log line 351).
In the meantime I did some correct prints when they were small enough to be finished before serial disconnect and also some prints of the sd-card.
To track this issue down I just yesterday tried safe mode and first recognized the stuttering. But I did not change some major things as far as I know.
If you think the webcam could be a problem I could try to stop the streamer daemon or even completely deactivate/disconnect from the pi.

foosel, 2018-02-06:
I don't think that the webcam is at fault here per se, and if so only by pushing the power limits of the whole setup.

From your saved log, this looks like something really hiccuped hard:

```
Send: N428681 G1 X138.844 Y126.85 E3829.92979*85
Recv: E������
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N428682 M105*21
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N428683 M105*20
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N428684 M105*19
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N428685 M105*18
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N428686 M105*17
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from 'Printing' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
Connection closed, closing down monitor
```

Note the unprintable characters in there. I wonder what happened there, it certainly looks more like some kind of crash than the usual "printer suddenly disconnected" issue. 

This combined with your earlier log makes me think that either something's up with your firmware or the controller (too busy or something), or maybe some communication issues on the way although those usually manifest differently. Try [these steps against those](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-keeps-running-into-communication-errors-and-timeouts) just in case though.

You say printing from SD card works fine? How about printing from another host/another machine? It would be interesting if you see the same behaviour when printing from say a laptop with Pronterface installed vs the Pi with OctoPrint.

AnduriI, 2018-02-06:
yeah, I noticed these characters before...
As you supposed I tested with pronterface (version 18Nov2017) and it worked flawless. Just to mention I also changed USB-cable and power supply even before.
My next try will be to use a new Marlin firmware (1.1.8) and see if the error exists also there. But I will have to prepare for this step (go get one of my arduinos as ISP, burn bootloader, configure marlin correctly,...) and it might take a few days. I will report back after that.

AnduriI, 2018-02-07:
I did the upgrade to Marlin 1.1.8, but still my CR-10 stops from time to time without reason. So I would say it's probably not a mistake of the printer nor the general communication (pronterface works), but something with the pi. Maybe there really is something accessing the serial port, but I don't know how to track this down. Do you know a tool to check this?
I will also try to setup a OctoPI image on a fresh SD card to check for hardware issues on the pi. If you have further hints please let me know.
Just in case you need to check my marlin, I used the version 1.1.8 from the link in this tutorial: https://plastikjunkies.de/firmware-teil-2-creality3d-cr-10-firmware-flashen-inkl-autolevel/ (third link at the download section, used the one without ABL)

foosel, 2018-02-08:
Sorry, I currently don't have any additional ideas to try, but your current ones sound good.

AnduriI, 2018-02-09:
short update:  I tested with a fresh image of OctoPi and it works fine. So I think it's not a hardware issue or OctoPrint bug but rather a problem with my system. That's why I will close this as bug report.
Still I want to find out what causes this problem and find a way to fix it. Maybe I will find a way to monitor services accessing /dev/ttyUSB0.
Still it is unknown to me why it startet spontaniously. It was wokring for a few weeks before stuttering. Maybe I will restore one of my weekly Backups and test. Otherwise I will have to build the system from scratch (and hope for the best) or use a dedicaded pi for OctoPrint.

closed 2018-02-09 by AnduriI

foosel, 2018-02-09:
@AnduriI thanks for the update! In general I'd strongly recommend a dedicated Pi for OctoPrint. While the four core Pi2 and Pi3 aren't as prone to other stuff interfering as the single core Pi1, Pi0 and Pi0w, it can never be ruled out on a "shared" system.

EddyMI3d, 2018-04-06:
Hello Gina!
Just for info: I had the same issue as AnduriI, except I have an i3 MK2.
These random pauses left some nasty blobs behind.
After a new and clean installation (OctoPi 0.14.0 - update to OctoPrint 1.3.7rc4) it works fine again.
I think I've got into that trouble when I tried to get the USB camera to work and installed some drivers via SSH - what was not necessary at all. I assume this mixed the USB ports up that resulted in these random pauses.