OctoPrint/OctoPrint #2397: Thermal Runaway Issues

dkingsjr, 2018-01-31:
#### What were you doing?

printing different models. i selected the item i wanted to print. began printing. at approximately the 2nd layer, the print was halted due to thermal runaway. Temperatures were 200 hot end, 60 bed Attempted to reproduce the issue using only the SD card without the octoprint server connected, problem would NOT reproduce.

Ideally provide exact steps to follow in order to reproduce your problem:

1. ...i checked the printer firmware thermal runaway code. all fine.
2. ...Disconnected printer from octoprint server.
3. ...uploaded a model to an sd card.
3. ...printed the model without an issue.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?
for the file to print successfully
#### What happened instead?
thermal runaway at layer 2
#### Did the same happen when running OctoPrint in safe mode?
yes
[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Version of OctoPrint
1.3.6
[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint
octopi
[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version
Anet A8/ Marlin 1.1.8
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
chrome

#### Link to octoprint.log
Recv: Error:Thermal Runaway, system stopped! Heater_ID: bed
Changing monitoring state from 'Printing' to 'Error: Thermal Runaway, system stopped! Heater_ID: bed
'
Recv: Error:Printer halted. kill() called!
[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

gthanatos, 2018-02-08:
I've had the same issue. 

Preheated the extruder to 180 degrees via the webinterface.

Started print job.

Temperature rises to 420 (instead of 215 which is in the gcode)
I was able to capture the logging of octoprint aswell.

Gcode:

M107
M115 U3.0.11 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S215 ; set extruder temp
M140 S55 ; set bed temp
M190 S55 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside pritn area
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
; Filament gcode

Logging octoprint:

Recv: ok T:180.5 /180.0 B:60.0 /60.0 T0:180.5 /180.0 @:30 B@:28
Send: M105
Recv: ok T:181.8 /180.0 B:60.0 /60.0 T0:181.8 /180.0 @:0 B@:26
Send: M105
Recv: ok T:182.8 /180.0 B:60.2 /60.0 T0:182.8 /180.0 @:0 B@:0
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M107*36
Recv: ok
Send: N2 M115 U3.0.11*82
Recv: ok
Send: N3 M83*27
Recv: ok
Send: N4 M104 S420.000000*73
Recv: ok
Send: N5 M140 S55*80
Recv: ok
Send: N6 M190 S55*94
Recv: ok
Send: N7 M109 S420.000000*71
Recv: T:183.0 E:0 W:?
Recv: T:182.7 E:0 W:?
Recv: T:182.7 E:0 W:?
Recv: T:183.0 E:0 W:?
Recv: T:183.5 E:0 W:?
Recv: T:184.5 E:0 W:?
Recv: T:185.5 E:0 W:?
Recv: T:187.2 E:0 W:?
Recv: T:188.6 E:0 W:?
Recv: T:190.3 E:0 W:?
Recv: T:192.1 E:0 W:?
Recv: T:194.1 E:0 W:?
Recv: T:195.9 E:0 W:?
Recv: T:197.9 E:0 W:?
Recv: T:199.5 E:0 W:?
Recv: T:202.1 E:0 W:?
Recv: T:204.3 E:0 W:?
Recv: T:206.4 E:0 W:?
Recv: T:208.5 E:0 W:?
Recv: T:210.6 E:0 W:?
Recv: T:212.6 E:0 W:?
Recv: T:214.9 E:0 W:?
Recv: T:216.7 E:0 W:?
Recv: T:218.6 E:0 W:?
Recv: T:221.0 E:0 W:?
Recv: T:223.0 E:0 W:?
Recv: T:225.0 E:0 W:?
Recv: T:226.9 E:0 W:?
Recv: T:228.8 E:0 W:?
Recv: T:230.5 E:0 W:?
Recv: T:232.5 E:0 W:?
Changing monitoring state from 'Printing' to 'Operational'
Recv: T:234.8 E:0 W:?
Recv: T:236.3 E:0 W:?
Recv: T:238.7 E:0 W:?
Recv: T:240.0 E:0 W:?
Recv: T:242.3 E:0 W:?
Recv: T:244.5 E:0 W:?
Recv: T:246.7 E:0 W:?
Recv: T:248.3 E:0 W:?
Recv: T:250.0 E:0 W:?
Recv: T:252.4 E:0 W:?
Recv: T:253.5 E:0 W:?
Recv: T:255.1 E:0 W:?
Recv: T:257.0 E:0 W:?
Recv: T:258.5 E:0 W:?
Recv: T:260.2 E:0 W:?
Recv: T:262.5 E:0 W:?
Recv: T:264.5 E:0 W:?
Recv: T:265.9 E:0 W:?
Recv: T:267.7 E:0 W:?
Recv: T:270.3 E:0 W:?
Recv: T:271.6 E:0 W:?
Recv: T:272.8 E:0 W:?
Recv: T:275.0 E:0 W:?
Recv: T:276.9 E:0 W:?
Recv: T:278.1 E:0 W:?
Recv: T:279.8 E:0 W:?
Recv: T:281.7 E:0 W:?
Recv: T:283.3 E:0 W:?
Recv: T:284.2 E:0 W:?
Recv: T:286.7 E:0 W:?
Recv: T:288.1 E:0 W:?
Recv: T:290.2 E:0 W:?
Recv: T:291.5 E:0 W:?
Recv: T:293.3 E:0 W:?
Recv: T:294.8 E:0 W:?
Recv: T:295.8 E:0 W:?
Recv: T:296.9 E:0 W:?
Recv: T:298.5 E:0 W:?
Recv: T:299.8 E:0 W:?
Recv: Error:0
Recv: : Extruder switched off. MAXTEMP triggered !

foosel, 2018-02-08:
@gthanatos Please provide the exact file you attempted to print there, and verify you don't have a temperature offset set for your hotend. From your log it sure looks like there was a `M104 S420` in your GCODE file, *or* you had a temperature offset of 205°C set. See also ["Wrong temperatures are set"](https://github.com/foosel/OctoPrint/wiki/FAQ#wrong-temperatures-are-set) in the FAQ.

And I'm starting to feel like a broken record here but: *Both of you* provide a fully filled out ticket template. It's asked for for a reason and having to constantly run after information is incredibly annoying and outright unfair towards everyone using OctoPrint because it eats up time that could otherwise be used to *improve things or fix issues for everyone*.

foosel, 2018-02-16:
Closing due to lack of response.

closed 2018-02-16 by foosel

pjnovas, 2019-02-16:
For anyone who get here with same problem, for me this was the layer fan, my cooling setting was to start the fan at second layer at full speed which was causing the temp to drop 10° too quickly and Marlin were firing a Thermal Runaway.

Just changing the cooling to be full speed on 4 layers fixed the issue, hope it helps!