OctoPrint/OctoPrint #2399: OctoPrint unaware file is done for large gcode files

adammhaile, 2018-01-31:
#### What did you expect to happen?
Expected the time estimate to be at least somewhat close and for the progress bar in OctoPrint to say 100% when the print is completely done.

#### What happened instead?
With large gcode files (estimate >10MB) OctoPrint seems to be misreading the amount of data it has gone through for the gcode file. For example, [this gcode file](https://drive.google.com/file/d/1BWm4SpkPS9c7bqv5zDvDvFpyp7QFoYVc/view?usp=sharing) is 64MB but when the print was completely finished, OctoPrint showed that it had only read 34MB of the file and therefore seemed to think it wasn't actually complete.

[image]

As you can see in the image above, this is from when I checked on it in the morning and the print had been done for a couple of hours by my estimate. Either way, it was complete but OctoPrint seemed to think there was more to do.

As a side effect to this, when printing large objects the OctoPrint estimate is WAY off. Simplify3D estimated 11 hours for this print but at the lowest OctoPrint said about 25 hours. I'm guessing because it thinks it's not as far into the gcode as it really is. Obviously it can't *really* think it's only that far in as I'm assuming OctoPrint has to keep a pointer to the location in the file it's reading and it certainly wouldn't work if that pointer was wrong. But it's definitely reporting it wrong.

The above is just one example. My printer as an 80 liter volume so I print a lot of very large objects and it does this every time. Smaller things, <10MB as noted, seem to be ok. From what I've seen I would say the amount it's wrong is proportional to the gcode size. Would have to do more long prints to confirm has I haven't kept that exact data.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

FormBot T-Rex 2+ using Marlin ~v1.1.2 - Not 100% sure of version as FormBot has refused to comply with GPL and release their modified version. But based on feature set I've narrowed it down to around 1.1.2

#### Browser and Version of Browser, Operating System running Browser

Chrome v64 on Windows 10, latest with all updates/patches

#### Link to octoprint.log

https://gist.github.com/adammhaile/68e41e44974b3ef67f6a107b24db1b68

#### Link to contents of terminal tab or serial.log

As it had sit for hours doing nothing more all the terminal showed was temperature responses.

#### Link to contents of Javascript console in the browser

N/A - Not a UI issue. Note: This is also an issue in Printoid on my Android phone, so the incorrect data definitely seems to be coming from the server side.

#### Screenshot(s)/video(s) showing the problem:
Screenshot included in description above.

I have read the FAQ.

foosel, 2018-01-31:
That sounds suspiciously like #2124, and a look at your file confirms that just like the file in that ticket, the last 30MB of your file consist exclusively of zero bytes:

```
> hexdump Combined_Hilt.gcode | tail
22009e0 5020 616c 7473 6369 7620 6c6f 6d75 3a65
22009f0 3120 3238 3630 2e30 3335 6d20 5e6d 2033
2200a00 3128 3238 302e 2036 6363 0d29 3b0a 2020
2200a10 5020 616c 7473 6369 7720 6965 6867 3a74
2200a20 3220 3732 352e 2038 2067 3028 352e 2030
2200a30 626c 0d29 3b0a 2020 4d20 7461 7265 6169
2200a40 206c 6f63 7473 203a 3031 342e 0d37 000a
2200a50 0000 0000 0000 0000 0000 0000 0000 0000
*
4000000
```

My guess would be, if you strip that binary garbage off the end of the file, all would be well. Considering that completion percentage is based on position in the printed file and half of the file is zero-bytes, I'm not surprised that the estimates are off here.

That leaves why your print was actually done, eg nothing was moving anymore, but OctoPrint was simply sitting there doing nothing. For this I do need a serial.log though, to get an idea what happened at the end.

adammhaile, 2018-01-31:
Interesting. I will have to go back to Simplify3D and export the gcode
again and inspect it locally. I'm not sure why it would include that
garbage data at the end. Either S3D is adding it (in which case I'll talk
to them) or something in my upload to octopi is adding it... I have the
"watching" directory on the Pi shared over CIFS and just save from S3D to
that directly. Will be able to confirm tonight.

adammhaile, 2018-01-31:
Ah...more data: https://forum.simplify3d.com/viewtopic.php?f=9&t=8459
So it looks like it's an s3d bug and I just need to update... will confirm
tonight.

foosel, 2018-02-05:
Could you confirm that updating indeed solved the issue?

adammhaile, 2018-02-05:
It did. All fixed with S3D 4.0.1

foosel, 2018-02-05:
Thank, closing this then.

closed 2018-02-05 by foosel