OctoPrint/OctoPrint #2400: Respectfully requesting more information regarding the display of current coordinates

S1CAR1US, 2018-02-02:
It may just be me not knowing what I am talking about but in just looking at it from a logical standpoint, it seems like the explanation as to why the current position of the hotend can not be displayed has some holes in its reasoning.  Rather than just requesting this feature again which I understand has already been done, I am wanting to point out the issues in the explanation.  Also, all things considered, the bottom line which is explained basically revolves around the fact that the on-board controls of the printer are the main reason as to which this can not be implemented.  That being said, I know for sure that there are plenty users which use OctoPrint as an _alternative_ to having an on-board display / controls, so why could this not be implemented as an option for this type of user?  Here are my examples of explanations for which I have issues.

> The connection between OctoPrint and printer is limited to serial speeds, so 250000 baud at max.
...
>  Imagine what would happen 
...

> The transmission rate would half
...
> Also be aware that OctoPrint can't track that information on it's own either by just adding up any moves sent via it - what about moves not triggered by itself but the printer

First of all, the printer just doesn't call moves to itself, obviously I know this is not meant to imply that this is happening, but instead it is in reference to moves which are made through the display local to the printer.  But when do you ever execute a jog command _**during**_ a print?  You dont.
Next, it seems the slow speed of a serial connection is blamed, and the 250k baud rate being the limit as to which communication happens, but afaik, it is the same rate which any other host connects to your printer by which OctoPrint is connected.  Therefore the only difference in communication and relay of information back to the host would be the extra transmission of said data through the WiFi, however there is clearly no mention of the WiFi being a limiting factor.

I would not be in the least bit surprised if I was completely wrong in my argument and I know that there is quite possibly something that I am overlooking but I am just looking at it like I said from a common sense point of view.  I have also briefly looked for other solutions to this and have come across the possibility of adding a custom control for calling the M114 command.  To me this doesnt exactly qualify as an alternative.  Also, regarding the custom command, since I have added the plugin for a different theme... the button to add a custom command is no longer showing up.  I understand that this is an issue for at the very least a different thread and more likely for the developer of the plugin itself, but just wanted to mention why it is that I have been looking past that workaround.  Thanks.

kantlivelong, 2018-02-02:
@S1CAR1US which previous issue # are you referring to?

foosel, 2018-02-02:
People print from SD card. People utilize auto run code on the SD card. Custom firmware specific M codes in sliced files might have movement effects that are completely intransparent (I'm looking at you, printer vendors). People use baby stepping/live z adjustment during prints. So that's why you can't just rely on what is sent through OctoPrint to track position.

Next problem: the planner buffer. Your firmware doesn't execute move commands directly. Instead it puts them into a planner buffer. Commands therein are used to determine acceleration and such. Execution point in time is completely unknown to the host (unless advanced ok is enabled, no command expansion in the firmware happens, the host knows exactly what goes into the planner buffer and can track when a command makes it actually out of there again). 

And since things are constantly in motion, even if you were to send an M114 after every single command sent to the printer (which thanks to the ping pong protocol does in fact mean you will only get half of the commands through than you did before, and you couldn't even do for SD prints), you still wouldn't get The Truth because your print head will have changed position and your information becomes outdated almost immediately.

The problem here is simply that while it would be possible to regularly request the position data from the printer during printing - with the down side of increased traffic on the line - the information you'd get out of that would simply be wrong most of the time. So forget accurate position information. And that brings up the question - if it's not accurate, why do you even need it, and why should the print job be negatively impacted for something that is merely a more or less rough piece of information that is outdated faster than it can be displayed?

Note that on pause and cancel OctoPrint does fetch an accurate position from the firmware (by not only sending an M114 but only sending that after an M400 that makes things wait until all moves are finished), and that's even provided for pause/resume and cancel scripts. But that approach doesn't work during printing for obvious reasons (if you pause after each move, you don't get smooth movements, you get a stuttering mess).

closed 2018-02-12 by foosel