OctoPrint/OctoPrint #2401: Can't update Octoprint 1.3.5 to 1.3.6

LolloMazzy, 2018-02-02:
I'm running a version 1.3.5 (master branch) of OctoPrint on a Raspberry Pi 3 with Raspian.
Any week ago, me software alert me for a new version, the v1.3.6 in fact.

But the notify didn't have the bottom for update. 
So i search in the wiki and around the google bur i don't have update already well my OctoPrint.
I have follow this guide: https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#updating--switching-branches
But I have problema to clean e install file python.

Can you explain me to update my OctoPrint?

foosel, 2018-02-02:
See https://github.com/foosel/OctoPrint/wiki/Plugin:-Software-Update#octoprint--125

joris57, 2018-02-05:
Same problem here. When trying to update form 1.3.5 to 1.3.6, I get following error message 👍 

Updating, please wait.
+++++++++++++++++++++++++++++++
Now updating OctoPrint to 1.3.6
+++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --branch= --force=false "/home/pi/OctoPrint" 1.3.6
Traceback (most recent call last):
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 313, in <module>
main()
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 309, in main
update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 218, in update_source
raise RuntimeError("Could not update, \"git fetch\" failed with returncode {}".format(returncode))
RuntimeError: Could not update, "git fetch" failed with returncode 128
Python executable: '/home/pi/oprint/bin/python'
>>> Running: git diff --shortstat
> git diff --shortstat
>>> Running: git fetch
> git fetch
The update did not finish successfully. Please consult the log for details.

foosel, 2018-02-05:
@joris57 that is not the same problem that @LolloMazzy described. Your's sounds like a corrupted or otherwise broken git checkout. Do this (via SSH):

```
source ~/oprint/bin/activate
pip install https://github.com/foosel/OctoPrint/archive/1.3.6.zip
sudo service octoprint restart
```

joris57, 2018-02-06:
Apologies. From the description, it looked like the same issue.

Anyway, I followed the advise above, which only seemed to make the corruption worse, so I went for a full new install as described in the FAQ and all is ok now. Thanks.

closed 2018-02-07 by foosel