OctoPrint/OctoPrint #2403: [Request] Allow a greater range for "waiting for printer to stabilize"

nixomose, 2018-02-04:
For whatever reason at higher temperatures my printer can't maintain the exact temperature. It will fluctuate betweet 230 and 250 never hitting 235 or 240 so octoprint will wait forever for the printer to stabilize, and thus I can't print anything at high temperatures.

Is there an easy way to adjust this tolerance of what's considered stablilizing?

ntoff, 2018-02-04:
The stabilisation range is usually controlled by your firmware and outside of octoprint's control. Is there a specific error message you're encountering? Providing all the info requested in the issue template is the only way to diagnose any sort of issue with octoprint. Typically the printer itself will be the one to say "ok, I'm hot now, give me the rest of the gcode" and not something octoprint has any say over (though some firmwares may allow changing of how that's handled, that's definitely outside the scope of this bug tracker).

See here for the information required: https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report

But as I suspect this is more of a general help request than an actual bug with octoprint (or really anything to do with octoprint) I'd suggest asking this in a more general 3d printer related forum such as the 3d printing community on google+ or on reddit: https://www.reddit.com/r/3Dprinting/new/

foosel, 2018-02-04:
Heatup (and waiting for target temperature) is fully in the hands of the firmware. Nothing OctoPrint can do about that. You might be able to adjust your firmware, but a PID tune might make more sense here. In any case, not a bug, and also not a request implementable in OctoPrint, thus closing.

closed 2018-02-04 by foosel

nixomose, 2018-02-06:
Didn't know that it was a printer thing, my apologies, sorry about that. But thanks for the explanation.