OctoPrint/OctoPrint #2408: Problem with webcam and picamera. Not working.

fr3n0z, 2018-02-08:
Hi guys, i am having big trouble. It's weeks i'm trying all...i've got 3 webcamera 1 of wich was properly working with ocroprint and 1 picamera.

With all of them the webcam server always says: The webcam server is currently not running (that is my link http://192.168.0.220/webcam/?action=stream)

If i ssh in to pi get in to /mjpg-streamer folder and send ./mjpg_streamer -i "input_uvc.so $camera_usb_options" -o "output_http.so -w ./www" they all works correctly, as soon i close the ssh session they stop working and i can't get them to run on octopi UI. 

I've tried all setting, different resolution, frame rate, -y, etc. I installed latest version, older version (until 0.9),new nightly build, tried jackson setup, foosel setup, installed script for run/stop stream nothing :(

Can't sort it out. Please, could someone help me troubleshoot? 

Pi 3
OctoPrint 1.3.4 (master branch) freshly downloaded from octoprint site. Only modified octopi.txt and network
Anet A8
Marlin 1.1.8

pi@octopi:/var/log $ lsusb
Bus 001 Device 004: ID 05a3:9210 ARC International
Bus 001 Device 013: ID 0603:8612 Novatek Microelectronics Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

webcamd.log:

/boot/octopi.txt: line 51: $'\r': command not found
Starting up webcamDaemon...

--- Configuration: ----------------------------
camera:        auto
usb options:   -r 640x480
raspi options: -fps 10
http options:  -w ./www-octopi -n
-----------------------------------------------

No camera detected, trying again in two minutes
No camera detected, trying again in two minutes

fr3n0z, 2018-02-08:
what is missing? All info requested a part of octprint.log that i can't find are there i hope.

foosel, 2018-02-08:
> **Please do not abuse the bug tracker as a support forum** - if you have a question or otherwise need some kind of help or support refer to the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ Community](https://plus.google.com/communities/102771308349328485741) instead of here.

Your problem is getting your webcam to work. That's a typical support question and this *bug* tracker is not the right location for that as also explained in the linked contribution guidelines (which btw also contain *all* information that is required to submit a fully filled out ticket template for actual bug reports).

closed 2018-02-08 by foosel