OctoPrint/OctoPrint #2412: [Request] Label near any slider to know current value

MasterPIC, 2018-02-09:
By smartphone while moving any slider it's not possible to read the current value.
Maybe adding a label is a good workaround.

Thanks
Armando

ntoff, 2018-02-09:
What do you mean? There's already labels on the buttons built into octoprint's control page. Are you talking about custom controls? Plugins?

Have you thought about using the TouchUI plugin? It's optimised for touch interfaces.

MasterPIC, 2018-02-10:
By adding a slider in config.yaml (*) no numeric value is shown when on smartphone..
When on PC I see the current numeric value while moving the slider.
I think it's worth making the slider working as expected also on a smartphone.

@ntoff Thank you for your tip: I didn't know TouchUI plugin!

( * ) This is the kind of code added in config.yaml:
```
- input:
  - default: 150
    name: Size
    parameter: dim
    slider:
      max: 210
      min: 50
      step: 5
  name: Calibration
  script: Calibrate.gco

```

foosel, 2018-02-14:
Looks like this is actually a shortcoming of the library used for the sliders in the version bundled with OctoPrint (which is somewhat ancient). There appears to have been a [a lot of development on that one](https://github.com/seiyria/bootstrap-slider) and the tooltip shows up on the current version even on mobile, but it breaks the sliders in other ways (can't seem to drag them anymore on mobile at all, layout's wonky too probably due to some customizations in OctoPrint's own CSS), so it's not a simple fix and will need some more work than just dropping in the newer version. I'm also not sure yet how updating it might break plugin UIs utilizing the current version.

PRs heartily welcome - this honestly looks like too much investigative work needed for a minimum gain that I could justify to do it myself for the foreseeable future. But I might also be mistaken and it's a quick fix after all - just doesn't look like it at all on first glance.

MasterPIC, 2018-02-14:
Hi Gina,
frankly if I had known TouchUI plugin I wouldn't have opened this issue.
So please feel free to close it if you think it's not worth it...
And in any case thank you for your time!
Armando

ntoff, 2018-02-14:
> I'm also not sure yet how updating it might break plugin UIs utilizing the current version.

Definitely breaks those too, which is probably fixable in css:
[image]

But there's also apparently a missing method:
[image]

(which I'm sure you're aware of, just putting it here for reference)

foosel, 2018-02-15:
Actually, I didn't take a look at the JS console 😅 But that looks like there were some API changes as well, and the definitely will cause problems with backwards compatibility. As I said, definitely not a drop-in replacement. Leaving this open however, would make sense to update (it usually always makes sense to update), it's just more work than *I* can justify in the moment (hence I've also marked it as up-for-grabs).

ntoff, 2018-02-15:
Got It working on android, but is intermittent. The tool tip comes and goes and I have no idea why. Plugging a keyboard into my tablet works fine (because mouse/trackpad on keyboard) but touching the screen with my finger will only occasionally bring up the tooltip (but at the same time, sliding *does* work).

Well, at least I've got everything lined up and I *think* fixed the javascript errors (although the fix may potentially break plugins because of the library change, it's either risk having to re-write plugins or fork the slider library and re-add setMax).

I'd leave the "up for grabs" tag for now though, unless I magically figure out why the tooltip only shows some of the time.

LazeMSS, 2021-03-31:
Mobile support is improved/fixed by adding a plugin:
https://plugins.octoprint.org/plugins/touchui/
https://plugins.octoprint.org/plugins/screensquish/
https://plugins.octoprint.org/plugins/uicustomizer/

Closing.

closed 2021-03-31 by LazeMSS