OctoPrint/OctoPrint #2418: Solution for Tevo printers

Skystriker, 2018-02-11:
Hello,

i found a solution to get Octoprint running with Tevo printers.
Problem: communication with the RAMPS like mentioned  in #2374 #2353.

Solution: Flash a custom Marlin to the board. The baudrate should be set to 115200  NOT the Tevo default of 250000. Additionally the line dwc_otg.speed=1 has to be inserted into /boot/cmdline.txt just as described in the FAQ.

That solved the problem for me.
I added the Tornado under supported printers.
Maybe this will be of use to someone.

Cheers
Sky

foosel, 2018-02-12:
Thanks for the feedback. I saw you already added that information to the wiki as well, otherwise I'd have done that now. Closing this ticket then :)

closed 2018-02-12 by foosel

rrehders, 2018-04-26:
Thank you for investigating this Sky,

I just got a Tornado myself but am much less knowledgeable. I believe you list 3 changes:
1) dwg_otg.speed=1 (change made, performing test prints now...)
2) baud rate to 115200 (where should this be made ? a config file in marlin ?)
3) Customize Marlin (or was that what you meant by the item above ?)

Any clarity you can provide would be appreciated !