OctoPrint/OctoPrint #2424: Random crashing of the octoprint server

stidf, 2018-02-14:
---

#### What were you doing?
When running a variety of prints, my installation of the OctoPrint server will crash. If you use a web browser you are greeted with a message of please SSH into the pi and restart the octopi service and to check out the logs. In the logs the only things logged are some connection issues with one of my plugin's (octoprint anywhere) ability to reach the internet, then the registering of the rebooting of the Pi.  
 I usually just power cycle the PI to reset everything rather than SSH in.

This is often accompanied by a crashing of the MPEG steamer service in advance of the Octoprint service crash, but not always. I am using a systemD script to start and stop the MPEG steamer.

My printer (a MBI replicator 2) will just hang waiting for more commands over the serial connection, that is now lost due to the octoprint service crash. I have GPX installed.

I have been dealing with this problem almost since I got the octoprint and the GPX plugin stable close to a year ago. 

The first crash happens several hours after I started a print and left for the evening. This is the last message logged before the log shows me starting the Pi the next day when I find it crashed: 2018-02-11 21:20:24,394 - anywhere - WARNING - Not connected to server ws or connection lost

I started another print and had the same crash occur 2 hours later as the print is progressing. The crash stamp is 2018-02-12 12:52:20,990.

Plugins enabled:
GPX
Cura Engine
Print History Plugin
Octoprint Anywhere

#### What did you expect to happen?
The print to finish

#### What happened instead?
The printer stopped receiving commands from Octoprint and sat there until a human canceled the print and rebooted octoprint.

#### Did the same happen when running OctoPrint in safe mode?
I have not been able to reproduce the bug in safe mode as this has happened probably 1-4 times a month and I don't want to always be running is safe mode.

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.13.0

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

MBI replicator 2 with HBP running sailfish 7.7

#### Browser and Version of Browser, Operating System running Browser

Windows 10 running Chrome Version 64.0.3282.167 (Official Build) (64-bit)

#### Link to octoprint.log

[plugin_GPX.log](https://github.com/foosel/OctoPrint/files/1722660/plugin_GPX.log)
[octoprint.2018-02-12.log](https://github.com/foosel/OctoPrint/files/1722663/octoprint.2018-02-12.log)

#### Link to contents of terminal tab or serial.log

Serial.log not enabled, contents of terminal tab don't show anything of value as this crash occurred 3 days and several prints ago. Will update ticket if I can get them.

#### Link to contents of Javascript console in the browser

Not a UI or web interface issue.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2018-02-14:
> I usually just power cycle the PI to reset everything rather than SSH in.

You definitely should stop doing that *immediately*, you risk file system corruption that way and that can cause a lot of issues including crashes. Honestly, since you've been doing that for a while I'd strongly recommend to reflash with a clean image since there are probably a lot of things now broken on that image messing with its general operation. As a rule of thumb, never just cut power to a running Linux system, it's bad news, and it's especially bad news with the Raspberry Pi.

When it crashes and you *do* SSH into it, I take it the OctoPrint process is no longer running when you check that via `ps -ef | grep -i octoprint`? Since your webcam server also at least sometimes crashes, it sounds less than an application specific issue to me and more like a general stability issue. The question is why. 

I'd recommend to secure the output of `dmesg` and the contents of `/var/log/syslog` the next time it happens, maybe that will give us a hint. Right now there's sadly not much to go on in order to analyse this.

It might also be worth a shot to have OctoPrint run directly from within a `screen` instance instead of as a service:

```
sudo apt-get update && sudo apt-get install screen
sudo service octoprint stop
screen
/home/pi/oprint/bin/octoprint serve
```

That way everything printed to stdout (even a segfault of the underlying python process) will be recoverable by simply reattaching to the screen session after a crash (`screen -x`).

This huge block of `\0` bytes in your log here:

```
00083190: 2E 31 36 38 2E 32 2E 31 39 36 0A 32 30 31 38 2D    .168.2.196.2018-
000831a0: 30 32 2D 31 32 20 31 30 3A 33 37 3A 33 38 2C 33    02-12.10:37:38,3
000831b0: 35 35 20 2D 20 6F 63 74 6F 70 72 69 6E 74 2E 75    55.-.octoprint.u
000831c0: 74 69 6C 2E 63 6F 6D 6D 20 2D 20 49 4E 46 4F 20    til.comm.-.INFO.
000831d0: 2D 20 4D 31 31 30 20 64 65 74 65 63 74 65 64 2C    -.M110.detected,
000831e0: 20 73 65 74 74 69 6E 67 20 63 75 72 72 65 6E 74    .setting.current
000831f0: 20 6C 69 6E 65 20 6E 75 6D 62 65 72 20 74 6F 20    .line.number.to.
00083200: 30 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00    0...............
00083210: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083220: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083230: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083240: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083250: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083260: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083270: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083280: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083290: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000832a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000832b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000832c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000832d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000832e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000832f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083300: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083310: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083330: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083340: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083350: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083360: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083370: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083380: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083390: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000833a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000833b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000833c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000833d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000833e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
000833f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083400: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083410: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083420: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083430: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00083440: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32    ...............2
00083450: 30 31 38 2D 30 32 2D 31 32 20 31 30 3A 33 36 3A    018-02-12.10:36:
00083460: 32 35 2C 32 34 31 20 2D 20 6F 63 74 6F 70 72 69    25,241.-.octopri
00083470: 6E 74 2E 73 74 61 72 74 75 70 20 2D 20 49 4E 46    nt.startup.-.INF
00083480: 4F 20 2D 20 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A    O.-.************
00083490: 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A    ****************
000834a0: 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A    ****************
[... cut ...]
00083590: 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A 2A    ****************
```

indicates log corruption, so just cutting power definitely let to data loss at least once here.

I also see that your instance isn't connected to the internet, but you still have the OctoPrint Anywhere plugin installed, which strikes me as a bit odd considering that that obviously needs a working net connection, hence the ton of errors in the log.

And something's up with your webcam configuration as well, hence the huge numbers of capture failures.

stidf, 2018-02-14:
Well this is why I am here.

The Pi is definitely connected to the internet. The office router sees both its WiFi connection and ethernet connections. Plus the router software shows that the Pi was definitely an active device on the day in question.

Sounds like the good old nuke and pave approach is the best way forward here. Not only for the power cycling issues. It will also likely fix some of the lingering silly things I did while trying to setup the web cam, before going to the systemD script. Blindly following setup guides on the internet is not the best of ideas, especially when your command line fu is weak. These things may or may not be affecting connections and the managing of port access.

Is there any logs of value that can/should be acquired now 3 days and several prints later before I re-image?

foosel, 2018-02-14:
I don't think so. Since we are basically looking at a possibly corrupted system, it would make more sense in any case to reimage and see if that makes the problems go away. And if not we have new logs.

About your net connectivity, OctoPrint determines that by trying to connect to a server on the net. By default that's Google's DNS at 8.8.8.8. Maybe it can't reach that for whatever reason but otherwise has perfect connectivity, in which case you should be able to simply configure a different IP on the net. Or alternatively just disable connectivity check, it's under Settings > Server.

stidf, 2018-02-14:
Thanks! I'll reopen the ticket if the problem comes back.

stidf, 2018-02-23:
So an update: It happened again. I reflashed and re-configured the octoprint install after downloading the latest version of octopi. Printer and print server were running fine since Saturday. The Pi and printer haven't lost power and been running more or less continuously since I reflashed the Pi. Last night I started a 24 hr print and when I checked on it this afternoon, it had crashed. Was working fine in the morning. 
Attached are the logs. 
[Octoprint.log.2018-02-20.txt](https://github.com/foosel/OctoPrint/files/1750407/Octoprint.log.2018-02-20.txt)
[Octoprint.log.2018-02-21.txt](https://github.com/foosel/OctoPrint/files/1750408/Octoprint.log.2018-02-21.txt)
[Octoprint.log.current.txt](https://github.com/foosel/OctoPrint/files/1750409/Octoprint.log.current.txt)

Output of dmesg: pi@mulePi:~/.octoprint/logs $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.28-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #998 SMP Mon May 15 16:55:39 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 80c6edc0, node_mem_map b603a000
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b5ff4000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=538ad6ae-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 871340K/901120K available (7168K kernel code, 483K rwdata, 2012K rodata, 1024K init, 778K bss, 21588K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
    lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c78f94   ( 484 kB)
       .bss : 0x80c7a000 - 0x80d3ca64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000327] Console: colour dummy device 80x30
[    0.001219] console [tty1] enabled
[    0.001268] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001339] pid_max: default: 32768 minimum: 301
[    0.001706] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001749] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002775] Disabling cpuset control group subsystem
[    0.002967] CPU: Testing write buffer coherency: ok
[    0.003035] ftrace: allocating 22386 entries in 66 pages
[    0.055973] CPU0: update cpu_capacity 1024
[    0.056038] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.056134] Setting up static identity map for 0x100000 - 0x100034
[    0.058358] CPU1: update cpu_capacity 1024
[    0.058365] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.059080] CPU2: update cpu_capacity 1024
[    0.059087] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.059751] CPU3: update cpu_capacity 1024
[    0.059757] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.059840] Brought up 4 CPUs
[    0.060013] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.060043] CPU: All CPU(s) started in HYP mode.
[    0.060070] CPU: Virtualization extensions available.
[    0.060883] devtmpfs: initialized
[... 204 lines cut ...]
Feb 22 13:17:01 mulePi CRON[12022]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Feb 22 14:17:01 mulePi CRON[20035]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Feb 22 15:17:01 mulePi CRON[28057]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
[... cut ...]
Feb 23 03:17:01 mulePi CRON[31545]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)

At this point I have restarted the octoprint service and gone back to printing. 

Hoping the logs have something this time. 

Also there shouldn't be any reason why the pi loses connectivity to the internet. The office didn't seem to lose internet, plus the router's logged device up time for both the WIFI connection and the ethernet connection on the pi doesn't seem to indicate a loss of connectivity. 

Please help!

foosel, 2018-02-23:
Your logs are incomplete since you used `cat` and just copy-pasted the contents of your terminal, which means anything to the right of the width of your window got cut off. What is there though doesn't hint at anything at all other than it still being offline. Same goes for your `dmesg` and `syslog`. Did you reboot the Pi in between? Because that would make `dmesg` sadly pretty unhelpful.

What I see is that it's still offline (or at least "OctoPrint Anywhere" keeps complaining about that). I can't tell you why it's offline. But just because the network you are in in principle has a internet connection doesn't mean your Pi does. If its default gateway or nameserver isn't set correctly, you can count on it being offline. Or if your office's firewall blocks it.

In any case, this sounds more and more like an issue with your environment rather than anything else, so the bug tracker isn't actually the right location for this. I suggest you try the `screen` approach I suggested [in this comment](https://github.com/foosel/OctoPrint/issues/2424#issuecomment-365589467) and the next time it happens provide the output of that plus `/var/log/syslog` and for good measure also `/var/log/syslog.1`. And please don't provide that stuff by `cat`ing and copy-paste, use an SCP client to get the actual log files off the Pi (e.g. WinSCP if you are on Windows) and provide them. It doesn't look like whatever causes the crash also causes a log entry to be written, so the only chance to get information on that is to actually capture everything on stdout while running from console. My money would be on some kind of segfault.

stidf, 2018-02-24:
Good thing I didn't reboot! Getting stronger with my terminal fu.
```
pi@mulePi:/var/log $ uptime
 03:03:26 up 6 days,  2:20,  1 user,  load average: 0.17, 0.27, 0.31
pi@mulePi:/var/log $
```
[Octoprint.log.2018-02-21.txt](https://github.com/foosel/OctoPrint/files/1753986/Octoprint.log.2018-02-21.txt)
[Octoprint.log.current.txt](https://github.com/foosel/OctoPrint/files/1753987/Octoprint.log.current.txt)
[Octoprint.log.2018-02-20.txt](https://github.com/foosel/OctoPrint/files/1753988/Octoprint.log.2018-02-20.txt)
[dmesg.log](https://github.com/foosel/OctoPrint/files/1753966/dmesg.log)
[messages.log](https://github.com/foosel/OctoPrint/files/1753977/messages.log)
[syslog.log](https://github.com/foosel/OctoPrint/files/1753978/syslog.log)

stidf, 2018-02-24:
Got the screen logging business all setup, so the next crash should be caught by the terminal.

stidf, 2018-03-02:
Output from screen when I got another crash. It ends with a segmentation fault.

```
Starting the server via "octoprint" is deprecated, please use "octoprint serve" from now on.
2018-03-01 03:04:46,852 - octoprint.startup - INFO - ******************************************************************************
2018-03-01 03:04:46,853 - octoprint.startup - INFO - Starting OctoPrint 1.3.6
2018-03-01 03:04:46,853 - octoprint.startup - INFO - ******************************************************************************
2018-03-01 03:04:46,897 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2018-03-01 03:04:47,740 - octoprint.startup - INFO - Blacklist processing done
2018-03-01 03:04:47,818 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /
home/pi/.octoprint/plugins and installed plugin packages...
2018-03-01 03:04:49,284 - octoprint.plugin.core - INFO - Found 11 plugin(s) providing 11 mixin implementations, 12 hook handlers
2018-03-01 03:04:49,407 - octoprint.server - INFO - Intermediary server started
2018-03-01 03:04:49,408 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /
home/pi/.octoprint/plugins and installed plugin packages...
2018-03-01 03:04:49,696 - octoprint.plugin.core - INFO - Found 11 plugin(s) providing 11 mixin implementations, 12 hook handlers
2018-03-01 03:04:49,727 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-03-01 03:04:50,794 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-03-01 03:04:50,897 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-03-01 03:04:53,282 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2018-03-01 03:04:55,620 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2018-03-01 03:04:55,623 - octoprint.plugin.core - INFO - Initialized 11 plugin implementation(s)
2018-03-01 03:04:55,631 - octoprint.plugin.core - INFO - 11 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
|  GPX (2.6.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX
|  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/octopi_support
|  OctoPrint Anywhere (0.6.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_anywhere
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
|  Print History Plugin (1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
2018-03-01 03:04:57,213 - octoprint.environment - INFO - Detected environment is Python 2.7.9 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 901689344
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: 3B
|      revision: a22082
|      version: 0.14.0
|  python:
|    pip: 9.0.1
|    version: 2.7.9
|    virtualenv: /home/pi/oprint
2018-03-01 03:04:57,222 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-03-01 03:04:57,246 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-03-01 03:04:57,517 - octoprint.server - INFO - Shutting down intermediary server...
2018-03-01 03:04:57,544 - octoprint.server - INFO - Intermediary server shut down
2018-03-01 03:04:57,546 - octoprint.events - INFO - Processing startup event, this is our first event
2018-03-01 03:04:57,547 - octoprint.events - INFO - Adding 1 events to queue that were held back before startup event
2018-03-01 03:04:57,575 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-03-01 03:04:57,650 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-03-01 03:04:57,651 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2018-03-01 03:04:57,711 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2018-03-01 03:04:58,089 - octoprint.plugins.GPX - INFO - Connecting through x3g.
2018-03-01 03:04:58,090 - octoprint.plugins.GPX - INFO - GPXPrinter created, port: /dev/ttyACM0, baudrate: 115200
2018-03-01 03:05:00,127 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "MulePi" for _http._tcp
[... cut ...]
2018-03-01 17:07:34,910 - backoff - ERROR - Giving up __forward_ws__(...) after 8 tries (WebSocketConnectionClosedException: Connection is already closed.)
2018-03-01 17:07:34,914 - anywhere - WARNING - Reached max backoff in waiting for server ws connection
2018-03-01 17:10:12,050 - anywhere - WARNING - Not connected to server ws or connection lost
2018-03-01 17:10:12,828 - anywhere - WARNING - Not connected to server ws or connection lost
2018-03-01 17:10:13,465 - anywhere - WARNING - Not connected to server ws or connection lost
2018-03-01 17:10:13,470 - anywhere - WARNING - Not connected to server ws or connection lost
Segmentation fault

foosel, 2018-03-02:
This is what I feared. So the python process is just crashing away. This could have multiple reasons, one of which being unlimited recursion (others being faulty RAM or bugs in C modules - note that OctoPrint itself doesn't have any C modules but depends on libraries that DO have them).

It might be worth testing on another RPi if you have one on hand, especially since you stated that your webcam server on the same device also crashes. It might be faulty RAM causing this.

If this is not it, you should rule out any of your plugins causing this. Considering that your printer depends on the GPX module to work if I understood correctly, I understand that running productively in safe mode is out of the question, but I would like to ask you to disable all other third party plugins that you have installed (based on your log up there that should only be OctoPrint Anywhere and Print History) and then see if you can still reproduce the issue.

@markwal the GPX plugin does have a native C extension, right? Are you aware of any issues like this?

Your's is the only case of such a segfault that I'm currently aware, so for now I'm more leaning towards faulty RAM or a plugin being at fault here than some crazy difficult to trigger segfault condition caused by OctoPrint itself.

kisersosay, 2018-03-12:
I am currently experiencing the exact same issues so it may not be a hardware related problem.

foosel, 2018-03-12:
@kisersosay to quote the [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report):

> **Always use the following template** (please remove what's within [...], that's only provided here as some additional information for you), **even if only adding a "me too" to an existing ticket**

Also include the output of the `screen` test as mentioned above.

markwal, 2018-03-12:
And, by the way, yes GPX is largely C code compiled to native and can cause a segfault though I haven't seen one myself in quite some time.

foosel, 2018-03-27:
@stidf any updates here with regards to testing with all plugins but GPX disabled, on another Pi and such?

stidf, 2018-03-28:
Unfortunately no. Printer doesn't function without the GPX plugin and I just haven't had the time to do the hardware swap yet.

stidf, 2018-05-15:
So a couple of updates: I got a new raspberryPi to run the print server and have since updated both the octoprint and GPX plugins. The failure rate seems to have gone down, but I am still getting the same segment fault crashes.

markwal, 2018-05-15:
Are you also using a new micro SD card?  And/or fresh image?  FWIW I haven't seen a seg fault in quite some time.

stidf, 2018-05-15:
The only new hardware was a new raspberry pi. I moved the SD card from the old pi to the new pi and kept running with the same image. I haven't changed the mirco SD card out yet, but I have reflashed the image in an earlier attempt at troubleshooting. 

The I applied the new updates for both Octoprint and the GPX plugin when they were recently published. I ran crash free for most of the month of April then had 3 crashes in the past week. Hence me reopening the ticket.

foosel, 2018-05-16:
Did you try running this *only* with stock OctoPrint and the GPX plugin, so without any further 3rd party plugins, as I suggested earlier? I just want to rule out that we are barking up the wrong tree here looking at the OctoPrint/GPX combo.

stidf, 2018-05-16:
Yes, prior to submitting the bug report. in 2017, I disabled all plugins
but the GPX plugin and ran for about 3 weeks before I had a similar crash,
after which I turn everything back on. I have not tried to it again
recently.

foosel, 2018-05-22:
So, I can't reproduce this, my instances are all rock solid since months. However, #2647 shows similar symptoms (still waiting for the `screen` test results) and *also* had the OctoPrint Anywhere plugin installed. Might be coincidence, might not be, but it's an interesting data point nonetheless.

stidf, 2018-05-25:
At this point I am going to try a new SD card and new image on the new pi. Only install Octoprint and the GPX plugin. Then get the web cam running and get printing. Then see what happens.

As for #2647, I believe that we are experiencing the same issue. I don't understand why the printer doesn't cancel the print after it loses the serial connection. I am assuming that the printer is just waiting for more serial commands that the pi is no longer giving because of the seg fault. I would assume that at some point the printer would cancel the print after missing the serial talkback with the print server, but this clearly isn't the case.

foosel, 2018-05-25:
> I would assume that at some point the printer would cancel the print after missing the serial talkback with the print server, but this clearly isn't the case.

You are expecting way more intelligence and thoughtfulness from that printer controller than they have. Your printer doesn't even know there's a print job going on (because pretty much all firmware doesn't have any kind of command to indicate that being the case), all it sees is commands from serial which it executes and then waits for more. Some printers have some kind of timeout (no new commands within in a certain timeframe = shut off heaters), but most don't.

There's no well defined, protocol, there's no consistency. That all this works at all surprises me all the time to be honest (and takes a ton of workarounds)

LDHall77, 2019-01-22:
New at Octoprint. First long print. Crashed mid-print. Don't think this beta garbage is worth it.
Never had an SD card crash in years!

foosel, 2022-06-20:
Still unreproduced 4 years later, no more information that would help to figure out what happened. Closing this as not actionable, if someone finds this having a similar issue, please open a new bug report.