OctoPrint/OctoPrint #2426: UnicodeDecodeError: 'ascii' codec can't decode byte 0x94 in position 48: ordinal not in range(128)

gordonBimig, 2018-02-15:
#### What were you doing?
Printing via Octoprint 

#### What did you expect to happen?
Finish the print

#### What happened instead?
The print stoped (nozzle remained on print and heating elements were switched off. Not error message on the printers LCD (Marlin LCD)

#### Did the same happen when running OctoPrint in safe mode?
Haven't tried/used safe mode so far

Hello,

I'm trying to print several items in one print, but the print is failing (simply stops, it seems always on the same layer height). I changed the items arrangement in the slicer and also removed some other parts, but the problem remains. I found the following lines in the log:

2018-02-14 19:56:55,308 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 1249, in _monitor
line = self._handle_errors(line)
File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 2003, in _handle_errors
level=logging.WARN)
File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 529, in _to_logfile_with_terminal
log = message + "\n| " + log
UnicodeDecodeError: 'ascii' codec can't decode byte 0x94 in position 48: ordinal not in range(128)

Hopefully you can explain that to me. Is this a bug?
Version of OctoPrint

1.3.6
Operating System running OctoPrint

OctoPi version : 0.14.0
Printer model & used firmware incl. version

UM Clone with Marlin1.1.5
Browser and Version of Browser, Operating System running Browser

Google Chrome 63.0.3239.132 on Windows 10 x64 1709 16299.125
Link to octoprint.log

https://pastebin.com/X0T2kM7Q
G-Code with which the error occured:

https://www.file-upload.net/download-12979754/spheres_emboitees.7z.html

Hope this is enough information?

foosel, 2018-02-15:
That looks like your printer returned something non ASCII during a resend request. Which should actually have been filtered (making this error a bug), however it would be interesting what your printer was doing there. Do you have it set to a non-english menu by any chance?

gordonBimig, 2018-02-15:
I'm seeing "weird" symbols when opening the serial log in notepad++. Not sure if this is meant to be this way.

serial.log
https://pastebin.com/hds6iYT9

serial.log.2018-02-15_09-09-46
https://pastebin.com/Lp5DG72r

and here is the serial.log of my last print from today which was successfull.
serial.log.2018-02-15_09-09-46
https://www.file-upload.net/download-12980089/serial.log.2018-02-15_09-09-46.html

Hope this helps.

foosel, 2018-02-16:
Those "weird symbols" are OctoPrint talking to your printer at the wrong baudrate during baudrate detection. That's normal (as a side note, you can greatly speed up your connection and reduce the likelihood of connection errors if you set your connection to the correct baudrate for your printer - apparently 250000 - instead of "AUTO").

Thanks for the `serial.log`, however I need a log of a *reproduction of the issue*, as in, when it failed/fails. A successful print won't help me determine what kind of output triggered the issue you reported here :)

gordonBimig, 2018-02-17:
That's what i've been expecting ;-)
I leave the serial logging on and set the baudrate to 250000. Will do some prints on the weekend. When the problem appears again I'll post a new serial log.

gordonBimig, 2018-02-17:
Well, that happened faster then I thought ... :-(

This seems to be the problem:
2018-02-17 07:28:05,199 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.

Attached the full serial log:
https://www.file-upload.net/download-12983147/serial.log.html

gordonBimig, 2018-02-17:
Just realized this plugin is not for the OctoPrint version I'm running. So I assume something is faulty with the serial connection?

gordonBimig, 2018-02-17:
So this seems to be the problem:
2018-02-17 10:43:43,262 - Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
2018-02-17 10:43:43,265 - Changing monitoring state from 'Printing' to 'Operational'

Is this a hardware related issue? I replaced the arduino and did another print and the print failed as well, but this time almost at the end of the print.

Attached the serial.log but probably nothing new to find here:
https://www.file-upload.net/download-12983294/serial.log.html

foosel, 2018-02-19:
Sounds like your printer is rebooting right in the middle of things. Nothing to be done about that from OctoPrint's side. Check firmware and hardware.

You are mixing problems here though. Your initially problem that caused the exception your reported to be logged doesn't seem to reproduce at all so far :/

PS: Please use something else than file-upload.net in the future, they keep to trying to redirect me to fishy pages when I try to download your logs.

foosel, 2018-02-19:
The above commit should fix the original error that was triggered, even though I still don't know how that could be triggered. Available on `maintenance`, soon on `devel`, will be part of 1.3.7.

gordonBimig, 2018-02-20:
Sounds like your printer is rebooting right in the middle of things. Nothing to be done about that from OctoPrint's side. Check firmware and hardware.

>> You are right => it preasumably was die PSU causing the ramps/arduino to reboot due to low voltage. After replacing the PSU this problem was gone.

You are mixing problems here though. Your initially problem that caused the exception your reported to be logged doesn't seem to reproduce at all so far :/

>> You are right again, sorry for that :-(

>> PS: Please use something else than file-upload.net in the future, they keep trying to redirect me to 
>> fishy pages when I try to download your logs.

Acknowledged ;-)

>>  Available on maintenance, soon on devel, will be part of 1.3.7.

Great! OctoPrint is awesome and your support too. Thanks!

commit 33147b03 (2018-02-19): Make terminal logging to log unicode safe

Closes #2426
closed 2018-04-09 by foosel