OctoPrint/OctoPrint #2427: [Request] Restricted areas

ysoldak, 2018-02-16:
#### What were you doing?

Printing several objects.

#### What did you expect to happen?

Have them all printed or, if any one of them detaches or fails in any other way, the rest survives.

#### What happened instead?

One fails and ruins the rest. Sad story.

#### Version of OctoPrint

1.3.6

#### Proposed solution

The idea in a nutshell.
In UI have a way to restrict the printer from doing anything in some specific area[s], dynamically, on-the-fly, to save at least part of the print.

From my very limited knowledge of G-code:
Can be implemented either by ignoring G0/G1 commands over that area entirely, or converting G1s to G0s, zeroing E.

Have no any good proposal for UI really, just sort of marking with "red" squares on GCode Viewer tab.

foosel, 2018-02-16:
This sounds very much like a duplicate of #2126. I suggest following that :)

closed 2018-02-16 by foosel

ysoldak, 2018-02-16:
@foosel You are right, thank you, my github-search-foo obviously was not enough to find that.