OctoPrint/OctoPrint #2430: Losing serial comm

leofaoro, 2018-02-17:
#### What were you doing?

Just printing. I connect to the printer just fine. In the middle of the print (randomly) I get this error.

#### What did you expect to happen?

Just print until completion

#### What happened instead?

I got a serial communication error and the print stopped

#### Did the same happen when running OctoPrint in safe mode?

I can't as my printer requires GPX

#### Version of OctoPrint
1.3.6
#### Operating System running OctoPrint
OctoPi 0.14

#### Printer model & used firmware incl. version

Qidi Tech I (FFP clone). Sailfish 7.8 reported.

#### Browser and Version of Browser, Operating System running Browser

Chome 64.0.3282.167. Windows 10

#### Link to octoprint.log

https://pastebin.com/biPDzqLE

#### Link to contents of terminal tab or serial.log

https://pastebin.com/1wj16GFy
(I clipped all the G1 in the middle of the file it got too long)

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Here's also the GPX log just in case

Thanks in advance!

foosel, 2018-02-17:
@markwal that exception looks like it might rather be originating in the GPX bits of the comm stack (at least I've so far never seen it before in regular communication). Could you take a look?

markwal, 2018-02-17:
Yes, this is indicative of the similar serial cable noise problems that plague serial over USB communications. Standard troubleshooting tips apply:

* Use a thick, short USB cable running as far away and perpendicular to the motor cable runs as possible
* Make sure your pi has a power supply with sufficient current rating and a good, short cable.

If you need more followup, you should open an issue on the OctoPrint-GPX bug tracker since there isn't really anything Octoprint can do about this since all communication is handled by GPX for your printer.

closed 2018-02-17 by markwal

leofaoro, 2018-02-18:
Wonderful, thank you both for the quick response. I will use a different USB cable. At this time the cable is connected to a powered USB hub so hopefully that is not the issue. Will test some more.

thanks again
Leo

leofaoro, 2018-02-18:
Just to close the loop. Replaced USB cable, power supply, checked for ground, no good. I had to replace the Rpi, all is good. I suspect the USB connector got loose and was losing connection sometimes...

thanks!
Leo