OctoPrint/OctoPrint #2431: mk2 mmu resets itself during print

patrick3463, 2018-02-17:
#### What were you doing?

3D printing

Ideally provide exact steps to follow in order to reproduce your problem:

Slice
upload
print
wait.......

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?
Finish printing my model for the second time

#### What happened instead?
wasted 24h print

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint

OctoPrint: 1.3.6 

#### Operating System running OctoPrint

Linux

#### Printer model & used firmware incl. version

prusa mk2 MMU 

#### Browser and Version of Browser, Operating System running Browser

NA

#### Link to octoprint.log

[octoprint (2).log](https://github.com/foosel/OctoPrint/files/1734123/octoprint.2.log)

#### Link to contents of terminal tab or serial.log

Recv: Full RX Buffer
Recv: Error:Line Number is not Last Line Number+1, Last Line: 175742
Recv: Resend: 175743
Send: N175743 G1 X128.969 Y80.129 E0.02116*106
Recv: ok
Send: N175744 G1 X137.719 Y80.129 E0.29618*106
Recv: start
Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
Changing monitoring state from 'Printing' to 'Operational'

#### Link to contents of Javascript console in the browser

NA

#### Screenshot(s)/video(s) showing the problem:

NA

I have read the FAQ.

patrick3463, 2018-02-17:
Forgot to mention I have "create additional OK for resend requests" on or something similar"

foosel, 2018-02-19:
> Forgot to mention I have "create additional OK for resend requests" on or something similar"

What firmware version are you running? From your terminal output it looks like this setting might not be necessary here and/or even causing the issue in some way as the firmware complains about a full rx buffer just prior to restarting. Try printing without it. 

And if you then run into resend requests with a missing `ok` I suggest to report in [this issue on the Prusa Firmware tracker](https://github.com/prusa3d/Prusa-Firmware/issues/331) to maybe make them provide a new firmware build for MK2 printers without the resend bug.

patrick3463, 2018-02-20:
Hi, Im running 3.1.0, the latest mmu firmware from prusa.