OctoPrint/OctoPrint #2432: [REQUEST] G29 results as 3D Heightmap

aimartin, 2018-02-18:
Hi!

I think that implementing something to see easily the G29 result as a 3D heightmap would be really nice.

Something like Duet has would be awesome. I know that they're plotting it from a CSV, but as M420 V returns the values from the EEPROM, it shouldn't be that different.

Regards

[image]

kantlivelong, 2018-02-20:
Pretty cool idea! Would be best implemented as a plugin.

sbts, 2018-02-20:
I'll vote for this one too

modem7, 2018-04-13:
Definitely got my cote

Redferne, 2018-04-13:
Please someone whom has spare time, make this happen 😉

jneilliii, 2018-04-14:
I'll take this one on.  I have an idea of how to make this work.

jneilliii, 2018-04-14:
Quick question to you guys.  To retrieve the current mesh does the command 

> G29 T

return something like the following for you as well in the terminal?  

> Send: G29 T
> Recv: echo:Home XYZ first
> Recv: 
> Recv: Bed Topography Report:
> Recv: 
> Recv: (0,9)                                                                   (9,9)
> Recv: (30,270)                                                                (270,270)
> Recv:   0.425    0.295    0.175   -0.147    0.056    0.071   -0.037    0.085    0.063    0.063
> Recv: 
> Recv:   0.450    0.325    0.056    0.166    0.050   -0.006   -0.069   -0.069    0.096    0.261
> Recv: 
> Recv:   0.358    0.343    0.107   -0.058    0.061    0.055   -0.021   -0.017    0.024    0.064
> Recv: 
> Recv:   0.368    0.223    0.150    0.030    0.015   -0.066   -0.071   -0.017   -0.031   -0.031
> Recv: 
> Recv:   0.359    0.235    0.180    0.005   -0.024   -0.123   -0.071   -0.089   -0.089   -0.089
> Recv: 
> Recv:   0.544    0.301    0.178    0.053 T:23.44 /0.00 B:23.66 /0.00 @:0 B@:0
> Recv:    -0.015   -0.108   -0.187   -0.210   -0.192   -0.173
> Recv: 
> Recv:   0.462    0.315    0.164    0.020   -0.246   -0.183   -0.174   -0.135   -0.227   -0.227
> Recv: 
> Recv:   0.871    0.324    0.104   -0.415   -0.039   -0.024   -0.095   -0.128   -0.149   -0.149
> Recv: 
> Recv:   0.696    0.172    0.168    0.112    0.148   -0.207   -0.072   -0.324   -0.177   -0.030
> Recv: 
> Recv: [ 0.535]   0.526    0.286    0.109    0.034   -0.067   -0.168   -0.359   -0.309   -0.258
> Recv: (30,30)                                                                    (270,30)
> Recv: (0,0)                                                                     (9,0)
> Recv: ok P15 B3

I'm using UBL and I'm not sure if the responses are the same between that and other methods.

koenkooi, 2018-04-14:
A plugin for the format the prusa marlin fork uses can be found here: https://github.com/ff8jake/OctoPrint-PrusaMeshMap

jneilliii, 2018-04-14:
Here's my stab at a Marlin version.  As long as you have a firmware version capable of auto-reporting temperature and G29 T1 responds in the same expected manner I programmed for it should work.  I'm investigating how to pause default temperature polling to bypass some programming knowledge shortcomings.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer

foosel, 2018-05-03:
Closing since we now have a plugin.

closed 2018-05-03 by foosel