OctoPrint/OctoPrint #2434: Cannot Login to OctoPrint with new computer

dmathias1974, 2018-02-19:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
aka the Contribution Guidelines that are linked ^-- just up there
in the big yellow box.

Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ...", "I have a
question ..."), that belongs on the mailing list or the G+ community
(both linked in the "guidelines for contributing" linked above, read
them!), NOT here.

Mark requests with a "[Request]" prefix in the title please. For bug
reports fully fill out the bug reporting template (if you don't know
where to find some information - it's all described in the Contribution
Guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete ANY lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to report is still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the Contribution
Guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. ...Connected fine with laptop to my Octoprint as always.
2. ...Started using a new desktop computer on the same network and tried to login to Octoprint web page.
3. ...Username/password are accepted but will not log in.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen? 
Login as usual.

#### What happened instead? 
No login accepted (with no message)

#### Did the same happen when running OctoPrint in safe mode? 
How do I do this?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.] 
Where is contribution guidelines?

#### Version of OctoPrint 
v1.3.6

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint 
Windows Vista 64-bit

[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version] 
OctoPi v0.14.0

#### Printer model & used firmware incl. version 
Creality CR-10 (stock firmware)

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Internet Explorer v11
Google Chrome v64

[If applicable, always include if unsure.]

#### Link to octoprint.log
??

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser
packed_libs.js?eebed41b:9176 POST http://10.0.1.53/api/login 500 (INTERNAL SERVER ERROR)

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2018-02-19:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

foosel, 2018-02-23:
Well, it's impossible to tell for certain since you still haven't provided the information necessary to analyse this, but if you had the Astroprint plugin install and disabled it, this would be an explanation. See #2442.

Closing this for now, feel free to request a reopen once you can provide all the requested information.

closed 2018-02-23 by foosel