OctoPrint/OctoPrint #2438: MultiMaterial Printing Issue - Erratic Printing

bwright86, 2018-02-20:
#### What were you doing?

I recently upgraded to OctoPrint v1.3.6, and was attempting to print a multicolor object.
In the same timeframe, I also enabled timelapse configuration on printer, and was attempting to use that feature.
(RasPi camera module was already installed and used frequently during previous prints with no issues.)

I have been able to print single material objects with OctoPrint and the Prusa i3 MK2S printer, since the update. But multimaterial objects seem to be an issue.

#### What did you expect to happen?

1. Slice the multicolor object in Slic3r, and upload the resulting gcode to OctoPrint.
2. Load the 3 filaments into the 3 extruders, and start the print from the OctoPrint console.
3. Printer heats up and prints the 3 test color strips and then beeps/pauses and raises the extruder.
4. I resume the job in OctoPrint, and the extruder head lowers and it starts printing the first layer for the waste tower and object.
5. Successfully finish printing the part and the resulting waste tower.

#### What happened instead?

1. Slice the multicolor object in Slic3r, and upload the resulting gcode to OctoPrint.
2. Load the 3 filaments into the 3 extruders, and start the print from the OctoPrint console.
3. Printer heats up and prints the 3 test color strips and then beeps/pauses and raises the extruder.
4. I resume the job in OctoPrint, and the extruder head lowers and proceeds to the right edge of print bed, and prints a stripe to the back-right corner, then proceeds to extrude large amounts of filament while occasionally raising the Z-Axis.
5. The serial.log sh

#### Did the same happen when running OctoPrint in safe mode?

I have printed the same object in safe mode, and still had the same issue as above.

#### Version of OctoPrint

OctoPrint v1.3.6

#### Operating System running OctoPrint

Raspberrypi 3 (Raspbian GNU/Linux 8 (jessie))

#### Printer model & used firmware incl. version

Prusa i3 MK2S w/ MMU (Firmware: )

#### Browser and Version of Browser, Operating System running Browser

Google Chrome Version 63.0.3239.132 (Official Build) (64-bit)

#### Link to octoprint.log

https://gist.github.com/bwright86/3c2eb2753685c9933d419650a756f3a3

#### Link to contents of terminal tab or serial.log

https://gist.github.com/bwright86/3c2eb2753685c9933d419650a756f3a3

#### Link to contents of Javascript console in the browser

https://gist.github.com/bwright86/3c2eb2753685c9933d419650a756f3a3

#### Screenshot(s)/video(s) showing the problem:

MulitMaterial print result: https://pasteboard.co/H8vRuWN.jpg
Zoomed in: https://pasteboard.co/H8vRa98.png

I have read the FAQ.

bwright86, 2018-02-26:
I found the cause of the issue. I had created a simple gcode script during pauses and unpausing that caused this issue to happen.

# Pause Script
G91             ; Use relative coordinates
G1 E-4 F200     ; Retract filament
G1 Z10.0 F7200	; Lift Z and retract

# Unpause Script
G91             ; Use relative coordinates
G1 E4 F200     ; Restore filament

I was building up a new image of Octoprint on a new Raspberry Pi, and was copying configurations over, when i noticed this script.

After removing these scripts, and running a multimaterial job, it completed perfectly.

Not sure what in these commands are causing Octoprint/Printer to glitch and quickly run through the print job. But these scripts caused it to be triggered.

foosel, 2018-02-26:
Apologies, this ticket somehow fell through the gaps.

I have an idea what happened here. You set your movements to relative coordinates in your pause and resume script, but never returned them to absolute. Meaning that on resume the firmware interprets what should be absolute coordinates as relative and all hell breaks loose:

```
2018-02-13 08:42:02,198 - Send: N173 M300 S800 P500*43
2018-02-13 08:42:02,203 - Recv: ok T:218.3 /215.0 B:96.9 /97.0 T0:218.3 /215.0 @:0 B@:99
2018-02-13 08:42:02,205 - Changing monitoring state from 'Printing' to 'Paused'
2018-02-13 08:42:02,221 - Send: M400
2018-02-13 08:42:02,222 - Recv: ok
2018-02-13 08:42:02,224 - Send: M114
2018-02-13 08:42:02,226 - Recv: ok
2018-02-13 08:42:02,227 - Recv: ok
2018-02-13 08:42:02,709 - Recv: ok
2018-02-13 08:42:02,711 - Recv: ok
2018-02-13 08:42:02,715 - Recv: X:187.57 Y:3.70 Z:0.50 E:-4.00 Count X: 187.81 Y:4.36 Z:-0.01
2018-02-13 08:42:02,733 - Send: G91
2018-02-13 08:42:02,735 - Recv: ok
2018-02-13 08:42:02,737 - Send: G1 E-4 F200
2018-02-13 08:42:02,738 - Recv: ok
2018-02-13 08:42:02,740 - Send: G1 Z10.0 F7200
2018-02-13 08:42:02,745 - Recv: ok
2018-02-13 08:42:02,749 - Recv: ok
2018-02-13 08:42:03,643 - Send: M105
2018-02-13 08:42:03,654 - Recv: ok T:217.5 /215.0 B:96.9 /97.0 T0:217.5 /215.0 @:1 B@:97
2018-02-13 08:42:08,645 - Send: M105
2018-02-13 08:42:08,718 - Recv: ok T:214.6 /215.0 B:96.9 /97.0 T0:214.6 /215.0 @:69 B@:92
2018-02-13 08:42:13,653 - Send: M105
2018-02-13 08:42:13,775 - Recv: ok T:215.0 /215.0 B:96.9 /97.0 T0:215.0 /215.0 @:45 B@:95
2018-02-13 08:42:18,654 - Send: M105
2018-02-13 08:42:18,665 - Recv: ok T:217.5 /215.0 B:97.1 /97.0 T0:217.5 /215.0 @:0 B@:73
2018-02-13 08:42:19,360 - Changing monitoring state from 'Paused' to 'Printing'
2018-02-13 08:42:19,379 - Send: N174 G91*19
2018-02-13 08:42:19,393 - Recv: ok
2018-02-13 08:42:19,396 - Send: N175 G1 E4 F200*46
2018-02-13 08:42:19,405 - Recv: ok
2018-02-13 08:42:19,408 - Send: N176 G1 Z-10.0 F7200*3
2018-02-13 08:42:19,416 - Recv: ok
2018-02-13 08:42:19,418 - Send: N177 G1 X160.269 Y111.859 F7200.000*90
2018-02-13 08:42:19,441 - Recv: ok
2018-02-13 08:42:19,444 - Send: N178 G1 Z0.200 F7200.000*13
2018-02-13 08:42:19,453 - Recv: ok
2018-02-13 08:42:19,455 - Send: N179 G1 E4.00000 F3000.00000*19
2018-02-13 08:42:19,466 - Recv: ok
2018-02-13 08:42:19,467 - Send: N180 G1 Y82.109 E1.2941 F2400*80
```

Try 

```
G91 ; Use relative coordinates
G1 E-4 F200 ; Retract filament
G1 Z10.0 F7200	; Lift Z and retract
G90
```

and 

```
G91 ; Use relative coordinates
G1 E4 F200 ; Restore filament
G90
```

instead. In a nutshell, doesn't look like a bug in OctoPrint, rather in your scripts ;)

closed 2018-02-26 by foosel