OctoPrint/OctoPrint #2441: Accept Bitcoin Cash donations

omersiar, 2018-02-22:
So i can and many more can tip. @chaintip 10 days

chaintip, 2018-02-22:
`0.00110242 BCH`| `~   1.32 USD` has [been tipped](https://explorer.bitcoin.com/bch/address/bitcoincash:qqh0ld4t4zhwlnjk7hfqc4nl7qvy60mwpy72dd5ucf) to this issue by @omersiar.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/17), get a pull request merged with ```@chaintip fixes #2441``` in the creation comment.

foosel, 2018-02-22:
Bitcoin or crypto currencies in general are a taxation nightmare and to be quite frank they are also way too volatile to feel comfortable with getting part of my livelihood through them. Sorry.

closed 2018-02-22 by foosel

chaintip, 2018-02-22:
@omersiar, I appreciate that you want to let people know about this bot, but please don't create issues asking maintainers to accept Bitcoin Cash donations. 

Tipping an existing issue or an individual user on an existing issue is reasonable, but posting a new one asking the maintainer to accept Bitcoin Cash is very spammy and will be unwanted most of the time.

@foosel apologies in this instance.