OctoPrint/OctoPrint #2442: Suddenly cannot login to Octoprint

hoegge, 2018-02-22:
#### What were you doing?
1. I go to the octoprint page and enter my name and password (correctly - if I enter the wrong ones I get an error message)
2. I press login (with remember me checked or not)
3. Nothing happens

#### What did you expect to happen?
Login as usual

#### What happened instead?
Nothing - stays at same screen
Have cleared browser cache (with closed tab) and tried both Chrome, Brave and Edge (on MS Windows 10)

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint

OctoPrint 1.3.6 running 

#### Operating System running OctoPrint

on OctoPi 0.14.0

#### Printer model & used firmware incl. version

CR-10 N/A

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 64.0.3282.167 (Official Build) (64-bit)
Same on Brave and Edge

#### Link to octoprint.log
[https://gist.github.com/hoegge/3e327e31dd6619262e36f644699b94b1](url)

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
[https://gist.github.com/hoegge/78c09977dd4c4259c675473601b29b63](url)

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-02-22:
I don't know how but it looks like the Astroprint plugin found a way around safe mode and is the cause of this since it's causing an exception on login. My guess is what triggered this is disabling that plugin. I suggest you uninstall it manually for now (see the FAQ on how to do this) and report this to the plugin authors.

foosel, 2018-02-22:
I just opened AstroPrint/OctoPrint-AstroPrint#26 as seen above and will also look into making OctoPrint more resilient against such situations.

hoegge, 2018-02-23:
Thanks a lot - will try manual uninstall. Also wrote to Astroprint. If they don't fix it maybe their plug-in should warned against. Let's see

hoegge, 2018-02-23:
Yes - manual uninstalling Astroprint worked. Everything works fine again. Thanks a log

ntoff, 2018-02-23:
@foosel do you suppose that's the same issue as here: https://github.com/foosel/OctoPrint/issues/2434 (in which case this would be a duplicate)

foosel, 2018-02-23:
@ntoff Hm, could be. Hard to say considering that #2434 contains pretty much no information.

foosel, 2018-02-23:
This will no longer be possible to happen starting with 1.3.7. If plugins are disabled they will be handled just like blacklisted plugins are handled today and not even be imported. That should prevent any shenanigans like this.

Available on `maintenance`, soon `devel`, to be released with 1.3.7.

hoegge, 2018-02-24:
Sounds good. For months I've had the issue that the Octopi/Octoprint froze after the print was done. No way to access the web interface nor logging in with SSH. After removing Astroprint, this does not happen anymore so could be related. I will report to them. Just in case you should stumble upon similar reports from others blaming Octoprint

ntoff, 2018-02-24:
>No way to access the web interface nor logging in with SSH.

If you can't log in via SSH that most likely has absolutely nothing to do with OctoPrint. SSH is a system service running on your (presumably) raspberry pi and is controlled by the operating system. It's more likely your pi has just dropped off the network if the print continues, but network access fails. Mine does that constantly, so much so that I run a script that checks and reconnects to the wifi every minute.

hoegge, 2018-02-24:
Well could be. Just wondered that Astroprint might be the culprit, since it does it own network / cloud communication and it never happens during a print - always after and never if it is just sitting idle online (connected with ethernet cable to LAN and WiFi disabled). But will investigate, check if it is accesible via keyboard and display and try to build in a watch dog that could log to a file if it is still active, so I can see after reboot, if and when it died. Don't think it is Octoprint's doing either.

CoDanny, 2018-03-17:
@hoegge, I’d like to point out that we have 100s of people actively connected to AstroPrint and using the plugin without problems. So while it’s a problem for some it’s not a widespread problem. 

This leads to believe that the problem is more to do with something in the user’s environment than the plugin itself.

The challenge is to find what it is. 

I also can’t think of what we could be doing do prevent an SSH connection. This again looks like something happening outside of the entirety OctoPrint environment, including the plugin

commit 40082456 (2018-02-23): Don't even import disabled plugins

We want to avoid disabled plugins interfering with our application
in any way at all.

Hence disabled plugins are now treated the same way as blacklisted ones.
The system will only create a dummy entry for them and not import them
at all. As a consequence, a restart of the platform will be necessary
for disabling plugins that were disabled.

The Plugin Manager has been adjusted accordingly.

Note that by no longer even importing the base file, plugin name,
version, description and other such metadata defined through control
properties won't be accessible.

C [... cut]
closed 2018-04-09 by foosel