OctoPrint/OctoPrint #2446: Logitech QuickCam Messenger not detected by mjpg-streamer

Skystriker, 2018-02-24:
#### What were you doing?

I shutdown octoprint.
Then I plugged the USB-Webcam in
and restarted Octoprint

#### What did you expect to happen?
For Octoprint to display an image

#### What happened instead?
The mjpg-streamer did not start.
The Webcam is seen by the OctoPi via lsusb as Bus 001 Device 005: ID 046d:08f0 Logitech, Inc. QuickCam Messenger

However it is not detected as a Camera as the /var/log/webcamd.log indicates

Starting up webcamDaemon...

Configuration: 
camera:        auto
usb options:   -r 640x480 -f 10
raspi options: -fps 10
http options:  -w ./www-octopi -n

No camera detected, trying again in two minutes
No camera detected, trying again in two minutes

I have tried fiddeling with the settings in the octopi.txt, but that didnt solve the problem either.
I have tried editig -r -f and -y.
I noticed that they do not influence the settings in the webcamd.log. Is this normal?
Aditionally I tried to get an image with dwc_otg.speed=1 enabled and disabled in /boot/cmdline.txt.
Now I am out of options.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi
#### Printer model & used firmware incl. version

Tevo Tornado

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 63.0.3239.132

#### Link to octoprint.log

https://pastebin.com/QXkf4fUe

#### Link to contents of terminal tab or serial.log

https://pastebin.com/5w7WGyW3

#### Link to contents of Javascript console in the browser

https://pastebin.com/Eg5zEfWk

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2018-02-24:
For questions or other kind of help or support, please get in touch on the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741). The bug tracker is reserved for bugs, feature requests and brainstorming tickets *in OctoPrint*. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2018-02-24 by foosel

Skystriker, 2018-02-24:
Sorry, you are right. This is definitly not a bug of OctoPrint, rather the OctoPi.

foosel, 2018-02-24:
Actually, it sounds more like the underlying Linux is not even recognizing
your camera as a camera, so this is even outside of the control of OctoPi.

Skystriker, 2018-02-24:
You are right. My image was somehow corrupted.