OctoPrint/OctoPrint #2447: [Request] Restart with M999 support

Depaulicious, 2018-02-24:
Repetier (for sure since 1.0.0, probably even before) and other firmwares support recovering from fatal errors by issuing M999 command:

* http://reprap.org/wiki/G-code#M999:_Restart_after_being_stopped_by_error
* https://github.com/repetier/Repetier-Firmware/blob/1.0.0/src/ArduinoAVR/Repetier/Commands.cpp#L2599
* https://github.com/repetier/Repetier-Firmware/blob/1.0.0/src/ArduinoAVR/Repetier/gcode.cpp#L1301

At the moment, when a fatal error occurs, OctoPrint simply reports it as a notification. To recover from it one must reset the printer and reconnect it.

Repetier (ab)uses fatal errors even for trivial things, i.e. "Z-probe failed", which occurs when the Z-probe doesn't fire and the Z endstop is hit. Recovering from it is just a matter of sending M999 and homing Z.

I think OctoPrint should detect a recoverable fatal error and show a dialog that allows one to fix the issue and send the M999 code, like Repetier-Host does.

I don't know about the other firmwares, but Repetier sends this message when such errors occur:

```
 - Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
```

ntoff, 2018-02-25:
Just turn off octoprint's disconnect on errors (it's somewhere in the serial settings I think, haven't got octoprint in front of me to check exactly) and add a custom control to the controls page that sends M999. Then the firmware can error out all it wants and resetting is but a click away (or a couple of clicks)

I, like you, run repetier firmware and had to disable the disconnect on errors feature for just that reason (abuses it for trivial things).

Depaulicious, 2018-02-25:
That may work, but it would be nice to have it as an actual feature.

A red dialog like on Repetier Host

speendo, 2018-07-20:
This solution doesn't work for me (with Smoothieware).

I receive a `!!` error for example when I ask the printer to "heat up" to a temperature lower than the actual temperature. This is a security feature of Smoothieware that I might need to configure better.

However, although I activated the setting to ignore firmware errors in `Settings > Behavior` I am not able to send a command `M999` in the terminal as the terminal seems to be locked after a `!!` error