OctoPrint/OctoPrint #2449: Cancelling and reprinting results in temperature setting not being applied

Nurgak, 2018-02-25:
#### What were you doing?

I uploaded a .gcode and clicked print, I remembered that I forgot to add supports because my part had overhangs. Before the nozzle temperature even reached the printing temperature I performed the following actions:

1. Regenerated the Gcode with supports in Cura
2. Canceled the print from the OctoPrint web interface
3. Deleted the old file in OctoPrint
4. Uploaded the new file (with the same name) in the interface
5. Clicked print
6. The nozzle temperature reached the printing temperature for the first, canceled, print
7. The new print started
8. The nozzle cooled down *while printing*
9. I noticed the problem in the temperature graph and quickly canceled the print

I think it might've happened because the printer does not accept new instructions before the nozzle temperature reaches printing temperature (it seems most 3D printers work this way). I canceled and launched a new print, OctoPrint sent a new initialisation routine including the temperature setting, but the printer did not process it, as it was still heating up for the previous print. This is only a theory, I don't know if that's actually the case.

#### What did you expect to happen?

I expected the new print to properly set the temperature. Actually I'd like the printer to stop heating after I click "cancel", but that's how the 3D printer firmware is made I guess...

#### What happened instead?

The nozzle started cooling while the printing.

#### Did the same happen when running OctoPrint in safe mode?

Did not try.

#### Version of OctoPrint

OctoPrint 1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Printrbot Simple MK 1405, unsure about the firmware version

ntoff, 2018-02-26:
>This is only a theory, I don't know if that's actually the case.

Neither do we without logs, please provide **all** the information the template requests. See here: https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report

Nurgak, 2018-02-26:
I updated with more information, it didn't really seem relevant, but who am I to judge.

Perhaps the `serial.log` file can show some insight: the temperature reaches 208C as expected, but in the meantime the `Changing monitoring state from 'Printing' to 'Operational'` might've reset the temperature setting, only applied retroactively once the nozzle  has reached the printing temperature.

Octoprint seems to detect that the temperature setpoint has been reset as one can see from the temperature graph, but still continues printing.

foosel, 2018-02-26:
See, this is why a serial log is part of the requested info for bug reports, because now I actually can see what's happening and also have an idea how to fix it. Timing of the cancel script interfering with the start of the next print.

foosel, 2018-03-05:
Closing this since it's actually a duplicate of #2321 and we don't need two tickets for the same issue. Currently working on the fix though.

closed 2018-03-05 by foosel

dancorrigan1, 2018-12-03:
I may be seeing this in 1.3.9

My starting Gcode has
G4 S45
M117 Printing

See terminal output after a start of print.
Send: N17 G4 S45*105
Recv:  T:215.31 /0.00 B:59.91 /0.00 @:0 B@:0