OctoPrint/OctoPrint #2450: [Request] API call to start timelapse separate from Start Job

DanalEstes, 2018-02-26:
### REQUEST 
Please consider adding an internal API to allow starting a video camera capture that will ultimately become a timelapse. 

* This would be separate from starting a Print Job. 
* This would use the current (at the moment the API is called) default settings for time lapse capture.
  * Possible exception: 
  * "On Z Change" might not be useful, since there would be no job to provide the Z events.  Were this the default configuration at the moment the API is triggered, the API could return an error, or could substitute a default time based set of parameters. 
* Naming would need to be based on a fixed prefix and date/time, as there is no job from which to derive a name.  Something like "API_Triggered_20180213065521.mpg" or similar. 

### REASON / INTENT
The ultimate intent of the requested feature is to enable Octoprint to be able to capture from camera(s) aimed at printer(s) that Octoprint is not controlling via USB.  This would allow consistency in video capture across multiple printers. 

### MULTIPLE CAMERAS
I don't know enough about the overhead of Video Capture to comment on the prudence of allowing or disallowing a Print Job start to occur if a capture is active via a camera that is NOT the camera for the printer that is USB controlled by Octoprint.  And/or what the limitations would be regarding multiple cameras on different Pi hardware platforms.  

Perhaps an initial implementation would simply make Job and Capture exclusive, for "the" camera controlled by this instance.  This would allow the known techniques for installing multiple instances of Octoprint on a single Pi, where some were used for USB (and camera) on a given printer, and some instances for camera only.

### PLUGIN
A plugin would need to be written to accept external events, such as GPIO or network messages, and call the "Start Capture" API.  Such a plugin would potentially be very short, and could be modeled after the "FilamentSensorReloaded" plugin that currently accepts GPIO and calls APIs to pause and resume print jobs.  Once the "StartVideoCapture" API is available, the plugin should be very simple. 

I volunteer to write the plugin, once the API is available.  The plugin would be configurable as to which GPIO pin and, if the API supports multiple cameras, which camera should be started based on that pin. 

Normal end of the capture would also be an API call, driven by the plugin when the pin goes back low.  

Cancellation could be provided by existing abilities to configure UI elements to take actions. 

### CONCLUSION

Thank you for your consideration of this feature.

foosel, 2018-02-26:
This sounds like a duplicate of #452 and #1116, so I'll close this. But thank you for this very extensive and thoughtful feature request!

closed 2018-02-26 by foosel

DanalEstes, 2018-02-26:
Thank you.  I searched, but did not find, the others. 

Is there any feedback on the likelihood and/or timing of this feature request being added?

foosel, 2018-02-27:
Likelihood: High. ETA: None yet

amercer86, 2018-04-15:
Just checking on this! Looking forward to this functionality!