OctoPrint/OctoPrint #2451: [Request] Recursive directory search in watched folder

LHelge, 2018-02-27:
I really like to organize my uploads in folders based on project, slicing profile, material and such. Recently I've started to used the watched folder over a samba share which works great and have really improved my workflow. But...

I can't get subfolders to work in the watched folder. I would like the file from the Watched folder to be moved to correct subfolder in the Upload-section.

I could probably make the change myself, maybe event submit my first ever pull request to GitHub, but I cannot find the logic for the watched folder in the repository.

foosel, 2018-02-27:
A PR for this certainly would be awesome! 👍 

The place you are looking for is [`src/octoprint/server/util/watchdog.py`](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/util/watchdog.py) (as a side note, I feel like this should be moved 🤔). It's currently always "uploading" into root. What you'd probably need to do is extract any sub folders from the path, make sure those are created server side through the file managers folder management methods, then upload in that target folder.

leoheck, 2020-04-29:
Hi, I am also interested in this. I would like to send a whole folder with files to the watch folder and keep the folder hierarchy. Is it possible?

foosel, 2020-05-12:
Implemented through @3557 by @fieldOfView, ready for 1.4.1 :)

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel