OctoPrint/OctoPrint #2452: 500 Internal Server Error

CrazyCitystars, 2018-02-28:
#### What were you doing?

1. Doing nothing at the moment, printer was offline

#### What did you expect to happen?

nothing really

#### What happened instead?

Octoprint works, when idle for the night the next day i get the error 500 Internal Server Error.
Had to do a hard reset ( pulling power cord and insert again) and works again fine.

#### Did the same happen when running OctoPrint in safe mode?

Didn't try that yet.

#### Version of OctoPrint

OctoPi 0.14.0

#### Operating System running OctoPrint

OctoPrint 1.3.6 

#### Printer model & used firmware incl. version

Tevo Tarantula with JimBrown EasyConfig-1.1.x

#### Browser and Version of Browser, Operating System running Browser

Google Chrome Version 64.0.3282.186 (Official Build) (64-bit)
Windows 10 PRO Version 1709 OS Build 16299.248

#### Link to octoprint.log

https://gist.github.com/CrazyCitystars/ed6608d4a6b4b7f8e2e8d21f56398cf0

#### Link to contents of terminal tab or serial.log

https://gist.github.com/CrazyCitystars/d1eaabacf6147dc31f211447f30bf85c

#### Link to contents of Javascript console in the browser

https://gist.github.com/CrazyCitystars/b06cf40f50ecff7045c5f6ddd96d5ed1

#### Screenshot(s)/video(s) showing the problem:

Didn't have one.

I have read the FAQ.

kantlivelong, 2018-02-28:
Hi @CrazyCitystars 

Can you replicate the issue in [Safe Mode](http://docs.octoprint.org/en/master/features/safemode.html)?

CrazyCitystars, 2018-02-28:
Will try, will report back the next day.

CrazyCitystars, 2018-03-01:
Well did try to connect to octoprint.
Got the error back...

foosel, 2018-03-01:
Can you please provide a screenshot of exactly what you are seeing? There's nothing in the log, and that's weird because something causing a "Internal Server Error" page should *definitely* produce a log entry, so either the error isn't coming from OctoPrint, or the logs you provided were not ones from when the error occurred.

CrazyCitystars, 2018-03-01:
https://snag.gy/i6D2kG.jpg
Here is the screenshot.
Is it possible to get the log when octoprint is in error? if so how?

kantlivelong, 2018-03-01:
I see you are conecting to port 1990 while OctoPrint is on 5000. What is
listening on 1990?

CrazyCitystars, 2018-03-01:
Yes, changed the port so i can access it from outside.
That port is open on the router

kantlivelong, 2018-03-01:
So .148 is the router listening on 1990?

What port is it forwarding to?
Can you try without going through your routers forwarding?

CrazyCitystars, 2018-03-01:
yes, i changed the port to 1990 instead of 80.
with my router cant forward a port to another port only open a port.
So i changed the port in the config file from 80 to 1990.

kantlivelong, 2018-03-01:
Alright so you changed the listening port of HAProxy then.

Is there anything useful in either OctoPrint or HAProxy logs when the issue
occurs?

/home/pi/.octoprint/logs/octoprint.log*

/var/log/haproxy.log*

kantlivelong, 2018-03-01:
...and one more thing. When the issue occurs try logging into the device via SSH and run `curl http://127.0.0.1:5000` what does it return?

CrazyCitystars, 2018-03-01:
where can i find those logs? to where do i need to go?
when i try ssh i get : network error: software caused connection abort

kantlivelong, 2018-03-01:
Logs are located on the device running OctoPriny/HAProxy at the paths from
the previous comment. You can try WinSCP on Windows but based on your error
it seems like there may be something causing system instability.

CrazyCitystars, 2018-03-01:
tried winscp didn't work.
Did a hard reboot ( plugging power ) still error.
Have to do this a couple of times then it boots up.

kantlivelong, 2018-03-01:
It definitely sounds like you have either disk corruption or a failing SD card. I would suggest the following:

1. Make a backup of your OctoPrint configuration (Located in /home/pi/.octoprint) and any other important files. Can also backup the entire SD if desired.
2. Test the SD card for read/write errors.
3. Once you have confirmed the SD is good you can re-install OctoPi and retest. 

You may be able to restore your backup of the .octoprint configuration directory but will have to re-install the plugins.

foosel, 2018-03-01:
Also please try to be more specific. "Didn't work" can mean pretty much everything: you couldn't connect, you couldn't find the file, you couldn't download the file, the file didn't contain anything, and so on and so on.

CrazyCitystars, 2018-03-02:
I wiped the SD card, tested it with windows-h2testw-fakeflashtest and no errors.
Downloaded the new image from the website and installed again on the SDcard with win32diskimager.
Works now, going to see in the morning if it works or not.

CrazyCitystars, 2018-03-03:
Octoprint is still running after the night. 
Whats the other thing i can do?
Reinstall the plugins and see if it happens again?

kantlivelong, 2018-03-03:
@CrazyCitystars Yep that seems like the next best step.

CrazyCitystars, 2018-03-05:
Still no crash, have 5 plugins now: PSU control, Cura plugin, navbar temperature, preheat button, themify.
Also i used it to print yesterday.
Will install more plugins later on.
I think its possible it crashed because i changed the port 80 to 1990.

kantlivelong, 2018-03-05:
Changing the port shouldn't cause any issue. Give it a shot.

kantlivelong, 2018-03-09:
Hey @CrazyCitystars,

Just wanted to see how things were going?

CrazyCitystars, 2018-03-10:
Hi,

Strange but everything keeps on working.
Installed different plugins and its fine...

foosel, 2018-03-22:
Considering this solved then.

closed 2018-03-22 by foosel