OctoPrint/OctoPrint #2454: Missing ok's causing printer to hang - serial buffer issue?

cesarvandevelde, 2018-03-01:
#### What were you doing?

OctoPrint occasionally stalls the GCode stream during prints. The issue is
relatively rare, and occurs roughly every 8-10 hours of printing. I have reason
to believe this may be caused by a serial buffer overrun on the host side,
please refer to the sections below for my preliminary analysis.

Here are the relevant settings on the OctoPrint side:
- Port: `/dev/ttyAMA0` (RPi's built-in serial port). The RPi's serial pins are
  connected via an active level-shifter to the RAMPS UART port.
- Baud rate: `250000`
- Temperature interval (autoreport): `2s`

And the relevant Marlin settings:
- Configuration.h:
  - `#define SERIAL_PORT 0`
  - `#define BAUDRATE 250000`
  - `#define HOST_KEEPALIVE_FEATURE`
  - `#define DEFAULT_KEEPALIVE_INTERVAL 2`
  - `#define BUSY_WHILE_HEATING`
- Configuration_adv.h:
  - `#define MAX_CMD_SIZE 96`
  - `#define BUFSIZE 4`
  - `#define TX_BUFFER_SIZE 4`
  - `//#define RX_BUFFER_SIZE 1024`
  - `//#define NO_TIMEOUTS 1000`
  - `//#define ADVANCED_OK`

#### What did you expect to happen?

I expected OctoPrint to reliably stream GCode files to the printer.

#### What happened instead?

Occasionally, OctoPrint gets stuck while midway through a print. The issue
occurs roughy every 8-10 hours of print time, making the issue very hard to
diagnose.

My analysis shows that this issue caused by a garbled "ok" message from the
printer, causing OctoPrint to pause sending new commands until a fake
acknowledgment is issued. Communication still works in this state, and OctoPrint
continues to receive temperature status messages.

I kept my eye on the situation and logged a bunch of corrupted messages, shown
below. Note that each line is from a separate print job:
```
Recv:  T:210.26 /210.00 B:60.03 /60.00o\x00k
Recv:  T:209.87 /210.00 B:60.01 /60.00 @:100 Bo\x00k
Recv:  T:210.00 /210.0o\x00k
Recv:  T:210.05 /210.0o\x00k
Recv:  T:210.10 /210.0o\x00k
Recv:  T:209.82 /210.0o\x00k
Recv:  T:210.26 /210.00 B:60.03 /60.00o\x00k
Recv:  T:209.87 /210.00 B:60.01 /60.00 @:100 Bo\x00k
Recv:  T:210.00 /210.0o\x00k
Recv:  T:210.05 /210.0o\x00k
Recv:  T:210.10 /210.0o\x00k
Recv:  T:209.82 /210.0o\x00k
Recv:  T:209.96 /210.0o\x00k
```

All these lines show the same pattern: an `ok` message is interjected into an
auto temp report message. The pattern is predictable, and it seems to me that
these errors cannot be caused by noise on the serial communication line
(e.g. a bit-flip on the RX line). For good measure, I also probed TX/RX data
lines, described below in section "Printer model & used firmware incl. version".
My measurements show a clean, crisp signal with minimal noise.

Some further digging led me to the following issue on the Marlin GitHub
repository: MarlinFirmware/Marlin#9111. My takeaway from that discussion is that
it is impossible for marlin to send an `ok` during a auto temperature report
message, leading me to believe that the host side is at fault. My initial guess
is that the RX serial buffer gets overrun, causing the two lines to be put
together and introducing null characters (`\x00`) into the stream. However, I am
not familiar with the OctoPrint serial port API.

To alleviate the issue, I hacked together a small plugin that looks for the
string `o\x00k` in incoming lines of data, and returns `ok` if detected. Since
using this plugin, i've the print stream freeze only once. I suspect this is
because the garble was in a different spot.

https://gist.github.com/cesarvandevelde/6e4b54ce1977cb89ffa875d095c550af

The plugin also logs instances of garbled messages, though null characters are
curiously dropped from the log. Below are two log files from the plugin:

```
2018-02-02 18:13:45,907 Garbled OK detected:
2018-02-02 18:13:45,909 --------------------------------------------------------------------------------
2018-02-02 18:13:45,910 Send: M110 N0
2018-02-02 18:13:45,910 Recv: o k

2018-02-02 18:13:45,911 --------------------------------------------------------------------------------
2018-02-02 20:19:59,496 Garbled OK detected:
2018-02-02 20:19:59,497 --------------------------------------------------------------------------------
2018-02-02 20:19:59,498 Send: G1 X101.15 Y51.325 E1069.44284
2018-02-02 20:19:59,499 Recv: ok

2018-02-02 20:19:59,499 Send: G1 X100.387 Y51.183 E1069.46865
2018-02-02 20:19:59,500 Recv: ok

2018-02-02 20:19:59,500 Send: G1 X99.613 Y51.183 E1069.4944
2018-02-02 20:19:59,501 Recv: ok

2018-02-02 20:19:59,502 Send: G0 F7200 X99.417 Y51.219
2018-02-02 20:19:59,502 Recv: ok

2018-02-02 20:19:59,503 Send: G1 F3000 E1068.9944
2018-02-02 20:19:59,503 Recv:  T:209.82 /210.00 B:59.96 /60.00 @:102 Bo k

2018-02-02 20:19:59,504 --------------------------------------------------------------------------------
```

```
2018-02-10 16:18:37,453 Potential garble detected:
2018-02-10 16:18:37,456 --------------------------------------------------------------------------------
2018-02-10 16:18:37,456 Send: G1 X88.289 Y53.384 E241.48638
2018-02-10 16:18:37,457 Recv: ok

2018-02-10 16:18:37,458 Send: G1 X87.536 Y53.156 E241.51254
2018-02-10 16:18:37,458 Recv: ok

2018-02-10 16:18:37,459 Send: G1 X86.854 Y52.79 E241.53829
2018-02-10 16:18:37,459 Recv: ok

2018-02-10 16:18:37,460 Send: G1 X86.182 Y52.238 E241.56721
2018-02-10 16:18:37,461 Recv: ok

2018-02-10 16:18:37,461 Send: G1 X74.726 Y40.779 E242.10614
2018-02-10 16:18:37,462 Recv:  T:210.10 /210.09 7 B@:41

2018-02-10 16:18:37,463 --------------------------------------------------------------------------------
2018-02-10 19:20:29,004 Potential garble detected:
2018-02-10 19:20:29,005 --------------------------------------------------------------------------------
2018-02-10 19:20:29,006 Send: G1 X111.867 Y145.848 E4089.55515
2018-02-10 19:20:29,007 Recv: ok

2018-02-10 19:20:29,007 Send: G1 X112.766 Y146.12 E4089.58639
2018-02-10 19:20:29,008 Recv: ok

2018-02-10 19:20:29,008 Send: G1 X113.592 Y146.562 E4089.61755
2018-02-10 19:20:29,009 Recv: ok

2018-02-10 19:20:29,009 Send: G1 X114.337 Y147.174 E4089.64962
2018-02-10 19:20:29,010 Recv: ok

2018-02-10 19:20:29,011 Send: G1 X125.799 Y158.636 E4090.18876
2018-02-10 19:20:29,011 Recv:  T:209.33 /210.0/ 60.00 @:110 B@:49

2018-02-10 19:20:29,012 --------------------------------------------------------------------------------
2018-02-10 20:48:43,800 Potential garble detected:
2018-02-10 20:48:43,804 --------------------------------------------------------------------------------
2018-02-10 20:48:43,805 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,806 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,807 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,808 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,808 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,809 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,809 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,810 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,811 Recv:  T:19.33 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,811 Recv:  T:19.29 /0.00 B:19.37 /0.00 @:0  T:19.29 /0.00 B:19.37 /0.00 @:0 B@:0

2018-02-10 20:48:43,812 --------------------------------------------------------------------------------
```

#### Did the same happen when running OctoPrint in safe mode?

I have not tried to reproduce this issue in safe mode.

#### Version of OctoPrint

OctoPrint 1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0 running on a Raspberry Pi 2

#### Printer model & used firmware incl. version

Self-sourced Prusa i3 running Marlin 1.1.8 on RAMPS1.4/Mega2560. Serial port is
connected via GPIO pins, not over USB. The Pi's TXD/RXD pins are connected via
an active level shifter to the RXD/TXD pins of the RAMPS board. A quick probing
with my Logic analyzer shows a clean signal with crisp edges, negligible noise
and minimal cross-talk.

[image]

[image]

#### Browser and Version of Browser, Operating System running Browser

Not applicable.

#### Link to octoprint.log

https://gist.github.com/cesarvandevelde/d51439ece12b35bb9faa4e87ec229eb9

#### Link to contents of terminal tab or serial.log

Please see above.

#### Link to contents of Javascript console in the browser

Not applicable

#### Screenshot(s)/video(s) showing the problem:

Not applicable.

I have read the FAQ.

sbts, 2018-03-01:
@cesarvandevelde you may be interested to take a look at
https://github.com/KevinOConnor/klipper/issues/113 : printer falter/stumble continuously cause of full host buffer

I don't believe it's directly related but it may provide some food for thought

cesarvandevelde, 2018-03-02:
@sbts Interesting read, thanks. I agree with you, the two issues bear similarities.

I've done some further digging, and found that hardware serial ports are usually configured with a 16-byte buffer, whereas USB serial ports tend to have larger buffers (e.g. 64 bytes). I notice that many of the errors occur at 16 bytes or at 32 bytes, maybe indicating that the ISR can't empty the buffer quickly enough? 

From what I could dig up online, there is another parameter in play as well: the serial port's receive trigger level (RTL). This parameter determines when an interrupt is fired to empty the buffer. For instance, a serial port with RTL of 14 would fire an interrupt when 14 bytes are ready in the buffer. With a 16-byte buffer, this setup would have 2 extra bytes to handle incoming data during the ISR. I've been looking for a way to either increase the RX buffer size or decrease the RTL, but I haven't found anything useful yet...

*Further reading:*
https://www.logic-control.com/datasheets/4/Tech%20Note/The%20Secrets%20of%20UART%20FIFO.pdf
https://raspberrypi.stackexchange.com/questions/76954/size-of-serial-port-read-size-in-c

foosel, 2018-03-02:
Thank you for the really exhaustive analysis that you did there! It sounds like what you are seeing is fairly similar to what was already reported by @b-morgan in #2370, just with slightly different initial setup (no dual extrusion in play here if I understand correctly).

The second link is very interesting. OctoPrint internally uses pyserial's `readline` method for receiving data from the serial port, and I just took a look at the implementation of that and apparently it's only ever reading one byte at a time. So it might be worth a shot to look into an alternative, more buffered approach to do the reads. What I can't do though is adjust any read buffer size parameters as far as I can see.

cesarvandevelde, 2018-03-02:
I've been looking around for info on serial ports in linux, and came to the same conclusion: the buffer size cannot be adjusted. Seems that this is driver-dependent or even hardware-dependent...

To me, this feels like a "perfect storm" type of deal, where a number of different factors work together to create a situation where a buffer overrun can occur:
- High baud rate
- Small buffer due to hardware serial (16 bytes vs 64 bytes on USB)
- Large RTL resulting in only 2 bytes of safety margin
- Slower processor (I'm using a RPi 2)
- Slow single-byte transfers inside PySerial
- Auto temp enabled, meaning that temperature reports can arrive at any given time
- GCode with lots of short line segments, increasing the probability of an `ok` collision. 

I can see a number of different solutions:
1. As you mentioned, perhaps using bulk reads will solve the problem.
2. Disable auto temp reports. This does not feel like a good solution to me as it as it doesn't address the underlying issue.
3. Incorporate a more advanced version of my hacky plugin into the OctoPrint comm layer. It should be doable to detect incoming \x00 characters and use that to information to trigger a fake ack when the stream gets stalled.
4. Find a way to either increase the buffer or decrease the RTL. This solution would be hardware-specific, though I imagine a substantial segment of users run OctoPrint on a Raspberry Pi.

Right now, I'm exploring option 4. I think I've found a way to decrease the RTL by manually modifying one of the control registers of the Pi's serial port. [This PDF](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) lists all the UART registers of the Pi's Broadcom SoC. Of particular interest is `TXIFLSEL`, listed on p188. 

I think it should be possible to modify this register using the [raspi-direct](https://github.com/offis/raspi-directhw) library, though I'll need to write a custom executable for that. Furthermore, there is a difference between the different RPi models. RPi models with wifi/bt use the high-performance /dev/ttyAMA0 port to communicate with the bluetooth module. As a substitute, the less performant mini-UART port is routed to the GPIO pins instead...

b-morgan, 2018-03-02:
@cesarvandevelde : I don't believe solution 3 will capture all the possible causes.  Case in point, my issue  #2370,  doesn't have any \x00 characters in the input stream.

I agree that solution 2 doesn't address the underlying issue, but again it did fix my issue and IMO is a reasonable workaround.

MarlinFirmware/Marlin#9111 also has a workaround which slows down Marlin's autoreport by introducing a delay.

b-morgan, 2018-03-02:
There hasn't been any discussion on using CTS/RTS hardware flow control. I wonder if that is a possible solution?

foosel, 2018-03-02:
A substantial segment of users are running on a Pi, but most of them are connecting through USB and not by wiring up the UART pins. Which - unless I'm mistaken and please tell me if I am - would mean having to find similar solutions for basically all possible serial chipsets out there, inviting detecting them in the first place, making option 4 somewhat unfeasible. Also, the routing on Pi3/Pi0W can (and should) be swapped through an overlay. In any case, it would be more of a topic for OctoPi than OctoPrint.

cesarvandevelde, 2018-03-03:
@b-morgan I took a look at your `serial.log` and you're right, could not find any null characters. 
Which RPi model are you using? Are you streaming over USB or through the UART pins?

Hardware flow control is an interesting idea, but I don't know whether Marlin supports that feature.
Software flow control (XON/XOFF) would probably also solve the issue, but is not yet implemented in OctoPrint. I noticed that a PR (#2346) was submitted to implement software flow control, but the code does not seem ready for general use.

@foosel That's a very good point. Option 4 would only solve the issue for /dev/ttyAMA0 on Raspberry Pi boards. At best, such a fix could be included with OctoPi, but maybe it should just be distributed as a board-specific plugin.

adriancuzman, 2018-03-03:
I also implemented a direct UART connection using level shifters, but there are more communication errors, this should not be an issue, every communication can have errors, the problem is that the 3D printer sending protocol doesn't have any error corrections (for example sometimes a 'resend 66879' would arrive as a 'resend 66') and Octoprint has no way to check if it's correct ... :/ 
Maybe we need to send data from Marlin to Octoprint (or any host) with checksum, like it is sent from host to printer? Should we implement something like this?

foosel, 2018-03-05:
In any case, forget about depending on firmware capabilities to solve this. There are thousands of Marlin forks and forks of other firmware types out there that rarely if ever get updated, and if so usually not fully merged but just cherry picked some functionality here and there. A lot of current generation printer owners don't even know how to update firmware and depend on their printer vendor to maintain it for them. Due to this things like XONXOFF will probably take years to become available in the long tail (= forks that are not the main repo), so depending on that for solving this issue would simply not be feasible.

As a side note, this kind of market fragmentation is also why interfacing with 3d printers from the view point of the host is such a horribly hot mess and why even small changes in how things are done can cause stuff to break for some printer models in new and exciting ways.

foosel, 2018-03-08:
I've tackled option 1 in the above commit. Would be interesting to see if that improves things for you. The commit is available on the `maintenance` branch.

foosel, 2018-03-09:
Short update on that, c6dfdf8 still had a compatibility issue with pyserial 2.7 (lower bound of current installs), a2ccf414e62e74f6330a73df3accc522131da3e8 fixes that. Also available on the `maintenance` branch.

cesarvandevelde, 2018-03-12:
Thank you for your work, @foosel! I've switched my install to the maintenance branch. I have some printing to do this week; I'll keep my eyes open to see if this fixes the issue... Fingers crossed!

cesarvandevelde, 2018-03-14:
Bad news, it seems that the problem is still present:

```
2018-03-14 13:29:17,345 --------------------------------------------------------------------------------
2018-03-14 14:29:37,721 Potential garble detected:
2018-03-14 14:29:37,722 --------------------------------------------------------------------------------
2018-03-14 14:29:37,723 Send: G0 X76.322 Y71.512
2018-03-14 14:29:37,723 Recv: ok

2018-03-14 14:29:37,724 Send: G0 X75.865 Y71.34
2018-03-14 14:29:37,725 Recv: ok

2018-03-14 14:29:37,725 Send: G1 F2700 X74.246 Y69.721 E1634.75418
2018-03-14 14:29:37,726 Recv: ok

2018-03-14 14:29:37,726 Send: G1 X74.176 Y69.651
2018-03-14 14:29:37,727 Recv: ok

2018-03-14 14:29:37,727 Send: G0 F7200 X73.602 Y71.34
2018-03-14 14:29:37,728 Recv:  T:210.00 /210.07  B@:37

2018-03-14 14:29:37,729 --------------------------------------------------------------------------------
```

foosel, 2018-03-14:
🤬😭

cesarvandevelde, 2018-03-14:
Yeah, couldn't agree more...

For sake of completeness, I did some more work on option 4. I've created a small executable to poke and prod at the Raspberry Pi's UART registers:

https://gist.github.com/cesarvandevelde/e21af54f4bf07ee5bd39013d32e189e0

I've found out that:
- By default, the RX interrupt is set to fire when the 16 byte hardware buffer is 1/2 full. AKA, the default value for `RXIFLSEL` is `010`. I was kind of expecting to find it set to 7/8...
- The value of `RXIFLSEL` is reset (by the driver?) every time the serial port is opened.

I have yet to do experiments with lower RTL values (e.g. 1/4). I think it's worth a shot, if only to rule it out as an underlying cause. However, the utility works by memory mapping to `/dev/mem`, and consequently requires root access. I am not sure if I can launch a sudo command from within an OctoPrint plugin. I imagine OctoPrint does not run with root privileges...

token47, 2018-05-01:
Guys, this problem may not be exclusive to the direct connection. I was having somethng VERY similar on USB too, a few months ago, but at that time it seemed like it was only something in my setup, since no one else could reproduce it.

There are a bunch of info im my previous (admittedly incomplete) bug report, and also one opened against marlin:

https://github.com/foosel/OctoPrint/issues/2380

and

https://github.com/MarlinFirmware/Marlin/issues/9179

As you can see in the end of 2380, I concluded that if I did not use the USB for the OS boot at the same time, it stopped. Just today I saw another user on a 3d printing group on Telegram saying that the printer would freeze (same behaviour) if it was READING the GCODE from a USB flash drive while printing, even if theos was installed on SD card. This reminded of my problem.

foosel, 2018-10-29:
I just noticed this still had the "needs testing" flag. As we confirmed that the mentioned commit(s) didn't solve this I've removed that tag. 

I'm a bit at a loss though as to how to continue here to be honest. It's basically the reverse situation of the reason why OctoPrint strictly adheres to the ok-ping-pong-protocol in the first place - to avoid overflows at the recipient. And as mentioned above, waiting for a solution for this issue in firmware is not going to cut it. I honestly can't start implementing my own custom serial drivers though, so... yeah, as I said, a bit at a loss here on what to do about this at all.

token47, 2018-10-30:
@foosel unfortunately I don't have that setup anymore. I think if no one else is having this problem then let's close it and if ever come back in any new setup I'll investigate it a little further and help open it again. Thank you.

cesarvandevelde, 2018-10-31:
I took a look in the log files of my printer, and the problem is still there, unfortunately. Roughly every 40m of active printing, I get a `0x00` character in the serial stream from the printer. However, since implementing my hacky plugin, I haven't had a single printer freeze so the issue does not really bother me so much anymore. 

A while ago, I did some testing with manual UART registry poking. I managed to get the RX interrupt to fire when the internal buffer is 1/4 full so that there'd be more headroom in the FIFO buffer. However, that did not really impact the issue. This leads me to believe the issue is somewhere in the in-between layers, e.g. linux drivers or the PySerial module.

I'll do some upgrading today on my system to see if that fixes anything. Haven't run `apt-get` in a while, and I'll also try installing the latest PySerial from GitHub. It's a longshot though, I'm not really holding my breath... :)

101jallen, 2018-12-13:
I have discovered this issue as well. The ackackack plugin does not catch it.
Marlin 1.1.9 +Linear Advance + Scurve 

Log lines
2018-12-12 17:49:09,569 - Send: N11084 G1 X99.314 Y86.842 E0.0310 F1170*59
2018-12-12 17:49:09,596 - Recv: X:100.24 Y:86.81 Z:5.86 E:0.00 Count X:7532 Y:7043 Z:234N1108ok
2018-12-12 17:49:10,948 - Recv:  T:207.19 /207.00 B:60.06 /60.00 @:71 B@:24
2018-12-12 17:49:12,950 - Recv:  T:207.03 /207.00 B:60.08 /60.00 @:74 B@:22
2018-12-12 17:49:12,955 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout
 if that happens regularly on specific commands or long moves.

The "ok" is intermixed in the previous lineending in Z:234N1108ok

DanielJoyce, 2019-02-05:
If PySerial readline reads a character at a time, that seems seriously slow and dumb...  Between Python, the Raspberry PI, and other things running on it yes it seems likely a serial overrun could occur.

We should use a custome readline ( in _readline in comm.py ), that reads into a fixed buffer ( say 4k ) and returns lines when \n is found. Done this several times in many different languages. Fast and never had issues.

foosel, 2019-03-12:
@DanielJoyce That is basically what we are doing since c6dfdf8 and a2ccf41 - reading *everything* that is reported as being there, then looking for a newline in that. See

https://github.com/foosel/OctoPrint/blob/8ce2290d63ed395ef1b1caa74537a517eac9efb2/src/octoprint/util/comm.py#L4653-L4690

which is wrapped around the serial object (at least if you use the default serial transport).

Didn't help.

Insane88, 2019-04-20:
Same thing at the first auto-connect 
`Connecting to: /dev/ttyAMA0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0xb0cd6050, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: o\x00k
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor`

eightheads, 2019-05-02:
I started running into the same issue as the OP when moving to a SKR v1.3 control board and running direct serial via Aux1 pin to Octoprint running on a Pi3b. Baud currently set to 250k, going to change the direct serial out for USB and see if I continue to run into issues.

eightheads, 2019-05-02:
I've recently started running into this issue with Octoprint on the Pi after moving to this control board. I did not have these issues with a MKS Gen L 8-bit board. I did try moving to connecting via a usb cable and continued to run into issues.

It looks like each time there is a line without the 'ok' and then the temp reporting starts.  

Hardware:
- Bigtreetech SKR v1.3 32-bit board running Marlin 2.0-bugfix
- TMC 2208s over UART
- Raspberry Pi 3b connected via direct serial-serial on Aux 1 of board
- Baud rate set to 250k
- Running Linear Advance and S-Curve Accel currently

Things Tested(these did not help):
- Changed GCode to verify it was not corrupt
- Moved Pi away from other components to reduce electronic noise
- Changed cables to new shielded cables and added ferrite chokes
- Changed out direct serial cabling for USB

Planned Testing:
- Turn off Linear Advance
- Upgrade to latest Marlin 2.0-bugfix(this pull was only from a couple weeks ago)

```
2019-05-01 01:02:23,675 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-05-01 01:02:23,698 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 63178, current line = 63181
| Last lines in terminal:
| Recv: ok
| Send: N63174 G1 X155.765 Y184.399 E3.1541*92
| Recv: ok
| Send: N63175 G1 X156.798 Y183.366 E3.1978*93
| Recv: ok
| Send: N63176 G1 X157.363 Y183.366 E3.2148*87
| Recv: ok
| Send: N63177 G1 X156.330 Y184.399 E3.2585*83
| Recv: ok
| Send: N63178 G1 X156.896 Y184.399 E3.2754*85
| Recv:  T:194.81 /195.00 B:59.58 /60.00 @:78 B@:127
| Recv:  T:194.96 /195.00 B:59.58 /60.00 @:76 B@:127
| Recv:  T:195.27 /195.00 B:59.58 /60.00 @:71 B@:127
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
| Send: N63179 M105*45
| Recv: echo:Unknown command: "63178 G1 X156.8N63179 M105"
| Recv: ok
| Send: N63180 M117 ETA 10 31 48 Day 01*42
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 63177
| Recv: Resend: 63178
```

```
2019-05-01 10:02:20,984 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-05-01 10:02:21,009 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 979421, current line = 979424
| Last lines in terminal:
| Send: N979416 G1 X109.741 Y164.217 E5.3399*110
| Recv: ok
| Send: N979417 G1 X110.079 Y165.141 E5.3693*101
| Recv: ok
| Send: N979418 G1 X110.654 Y166.577 E5.4156*104
| Recv: ok
| Send: N979419 G1 X111.238 Y167.899 E5.4589*108
| Recv: ok
| Send: N979420 G1 X111.640 Y168.747 E5.4870*101
| Recv: ok
| Send: N979421 G1 X112.368 Y170.168 E5.5347*100
| Recv:  T:194.81 /195.00 B:60.00 /60.00 @:77 B@:0
| Recv:  T:194.81 /195.00 B:59.79 /60.00 @:75 B@:0
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
| Send: N979422 M105*20
| Recv: echo:Unknown command: "979421 G1 X112.N979422 M105"
| Recv: ok
| Send: N979423 G1 X112.909 Y171.146 E5.5682*106
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 979420
| Recv: Resend: 979421
```

```
2019-05-01 17:34:41,361 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-05-01 17:34:41,399 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 8, current line = 11
| Last lines in terminal:
| Send: N3 M106 S255*102
| Recv: ok
| Send: N4 G4 P500*108
| Recv: ok
| Send: N5 M106 S0*98
| Recv: ok
| Send: N6 M140 S60*85
| Recv: ok
| Send: N7 M190 S60*89
| Recv: ok
| Send: N8 M104 S195 T0*36
| Recv:  T:195.27 /195.00 B:60.00 /60.00 @:47 B@:0
| Recv:  T:195.27 /195.00 B:60.00 /60.00 @:47 B@:0
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
| Send: N9 M105*46
| Recv: echo:Unknown command: "8 M104 S195 T0"
| Recv: ok
| Send: N10 M109 S195 T0*16
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 7
| Recv: Resend: 8
```

Here is my full [octoprint.log](https://pastebin.com/Z69A6dGZ). Here is my [serial.log](https://www.dropbox.com/s/zxob150615wra24/serial.log?dl=0).

corporategoth, 2020-05-25:
So I re-wrote the readline function as follows:

```
        def readline(self, terminator=serial.LF):
                termlen = len(terminator)
                timeout = serial.Timeout(self._timeout)

                while not timeout.expired():
                        termpos = self._buffered.find(terminator)
                        if termpos >= 0:
                                line = self._buffered[0:termpos + termlen]
                                del self._buffered[0:termpos + termlen:]
                                return line

                        # Can this return a None?  Of will it always return bytes
                        # (including b"") or throw an exception?  Functionally I
                        # guess it doesn't matter too much.
                        readsz = self.in_waiting or 1
                        data = self.read(readsz)
                        if not data:
                                break

                        nullpos = data.find(b'\x00')
                        if nullpos >= 0:
                                try:
                                        str = data.decode('utf-8')
                                except UnicodeDecodeError:
                                        str = data.decode('latin1')
                                _logger.error("NULL detected in read of {} bytes [{} / {}]: {}".format(readsz, nullpos, len(data), sanitize_ascii(str)))

                        self._buffered += data

                return b''
```

I find it a little cleaner than the implementation that is there.  More importantly, it can detect and error on any nulls it sees - and tell me where in the string they come from.

For example:
```
2020-05-25 16:08:46,551 - octoprint.util.comm - ERROR - NULL detected in read of 114 bytes [81 / 114]: NORMAL MODE: Percent done: 0; print time remaining in mins: 444
2020-05-25 16:10:47,457 - octoprint.util.comm - ERROR - NULL detected in read of 114 bytes [49 / 114]: NORMAL MODE: Percent done: 0; print time remainin  mins: 443
2020-05-25 16:12:45,676 - octoprint.util.comm - ERROR - NULL detected in read of 4 bytes [0 / 4]:  k
2020-05-25 16:12:45,828 - octoprint.util.comm - ERROR - NULL detected in read of 65 bytes [33 / 65]: SILENT MODE: Percent done: 0; pr  time remaining in mins: 446
2020-05-25 16:15:54,660 - octoprint.util.comm - ERROR - NULL detected in read of 68 bytes [65 / 68]: ok T:200.0 /200.0 B:55.6 /55.0 T0:200.0 /200.0 @:28 B@:0 P:43.7 6
2020-05-25 16:17:04,756 - octoprint.util.comm - ERROR - NULL detected in read of 4 bytes [0 / 4]:  k
2020-05-25 16:22:38,446 - octoprint.util.comm - ERROR - NULL detected in read of 72 bytes [49 / 72]: ok T:200.1 /200.0 B:55.1 /55.0 T0:200.1 /200.0 @7  B@:17 P:43.7 A:39.8
2020-05-25 16:25:51,671 - octoprint.util.comm - ERROR - NULL detected in read of 59 bytes [1 / 59]: E : Percent done: 3; print time remaining in mins: 433
2020-05-25 16:26:49,655 - octoprint.util.comm - ERROR - NULL detected in read of 97 bytes [33 / 97]: NORMAL MODE: Percent done: 3; prm aining in mins: 427
```

Since it is dumping that log after a single read (ie. this is not a combination of the end of one read and the beginning of the next or something), we can see that the nulls are coming in the middle of single reads of data from pyserial.

This then would indicate a pyserial problem (if it is client side).  If it was always the last or first character in the read, or I always showed the null when trying to read 1 byte, that might show a different problem.

Also of note is that each time it happens, it seems that multiple bytes are being replaced with the null character.

The other potential issues would be either the printer is sending random nulls and dropping characters (unlikely since this has been shown on many different printers), or a wiring issue (which also seems unlikely as it seems too specific - it is not flipping bits changing one character into another, it's specifically generating NULL and dropping whole bytes).

foosel, 2020-06-17:
So, I still haven't been able to observe this myself, and after going through this report and a bunch of the linked ones it seems to be fairly restricted to direct serial connections (vs USB). We already established that there might be something up with the driver situation for that further up.

As long as there are only extra null bytes getting injected into the stream, that can be recovered from at least in the case of mangled `ok`s by the AckAckAck plugin, which I've also forked and marked as Python 3 compatible here: https://gist.github.com/foosel/23b55541299b072e3f6dd399115f70ba

Anything else, and I fear my hands are bound. What OctoPrint can do is communicate via pyserial. It relies on the data it sends and the data it receives being the data that actually goes over the line. If something in between causes data corruption, there's no way to do anything about that that I can see. From OctoPrint to the printer we have line numbers and checksums to help us a bit, on the return path there's sadly no sanity checking built into the protocol, so that has to rely on stopgap solutions like the AckAckAck and similar plugins.

I've added some null byte warning logging into `maintenance` (to be pushed out as part of OctoPrint 1.4.1), and I've also adjusted the `readline` wrapper based on your suggestions, @corporategoth, but that's about all that I can do here. I'm marking this as `not octoprint` and am closing this now as all evidence so far points at it happening "underneath" OctoPrint (just like a yanked out serial cable would) and it doesn't make sense to keep tickets open forever that based on everything we've so far seen here are not actionable.

I wish I had better news.

closed 2020-06-17 by foosel

cesarvandevelde, 2020-06-18:
Your decision is definitely reasonable. Thank you for the effort for getting to the bottom of this! For what it's worth, I have been using the AckAckAck plugin for two years now and have not had my printer freeze since. It does not fix the root cause, but as long as it works, who cares? :)

corporategoth, 2020-06-19:
FYI, I commented on the commit you tagged in this ticket.  But I too can understand not keeping open a ticket that you can't resolve (ie. that has been proven to be something NOT inside OctoPrint).