OctoPrint/OctoPrint #2455: Webcam snapshot fails if https self signed url it's used, same cert with stream url it's ok.

hashashin, 2018-03-03:
#### What were you doing?

I'm using apache as reverse proxy to provide https to octoprint and the webcam. Test an https url in the stream box works as expected but the snapshot url test fails if I use https. Try to create a timelapse using self signed https url also fails.

#### What did you expect to happen?

Perhaps a warning if the webcam it's using self signed cert for the snapshot url and an accept dialog?

#### What happened instead?

Octoprint complains the url  it's not valid and generates an error in octoprint.log.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

maintenance branch at b8e4ac2

#### Operating System running OctoPrint

Armbian 5.38

#### Printer model & used firmware incl. version

Anet A8, repetier 1.0.1

#### Browser and Version of Browser, Operating System running Browser

Chrome 64.0.3282.186 32bits. Windows 10.

#### Link to octoprint.log

http://paste.debian.net/1012919/
With plugins, safe mode, and a try to use snapshot feature.

#### Link to contents of Javascript console in the browser

http://paste.debian.net/1012916/

I have read the FAQ.

Thanks, salud.

PD: Sorry for my poor english.

hashashin, 2018-03-03:
Just a note, as ugly workaround I added "verify=False" to requests calls in "server/api/__init.py" and "octoprint/timelapse.py". Snapshot url test passes now and timelapse feature also works again.

Salud.

foosel, 2018-03-07:
SSL validation can now be disabled for the snapshot URL per the above commit, available on `maintenance` and soon `devel`. To be released with 1.3.7.

foosel, 2018-04-09:
Closing as 1.3.7 was just released.

closed 2018-04-09 by foosel