OctoPrint/OctoPrint #2457: [Request] user metadata in files

DongerZone, 2018-03-04:
Tagging files with the name of the user who uploaded them would be pretty great in terms of file management. Restricting people from printing files they didn't upload or preventing them from deleting someone elses files have both been issues that I've had come up.
How difficult would that be for me to implement on my own if needed?

ntoff, 2018-03-04:
I believe a more granular permission system is on the list of things to do https://github.com/foosel/OctoPrint/pull/2198 , couldn't say when it would be ready though.

foosel, 2018-03-05:
This could be seen as a add-on to #1110 once that gets implemented through #2198. First things first though.