OctoPrint/OctoPrint #2459: [Bug] Printing from LCD and SD does not update State information in OctoPrint

Slavulj, 2018-03-04:
OctoPi-0.14.0 with OctoPrint-1.3.6 on Raspberry Pi Zero W
Marlin-1.1.8 on Mega2560 wirh Ramps1.4
Pi connected over USB
(tested in safe mode)

When i select print from LCD to print from SD, OctoPrint do not update informations in state tab
(there is no "SD printing byte xx/xx" in terminal)
[image]

But when i select same gcode(SD) from web interface in octoprint its showing all informations in state tab
(now it show "SD printing byte xx/xx" in terminal)
[image]

and when i start print from octroprint, cancel button does not work, it put printer in pause mode while in octoprint show its canceled and i have to cancel it on LCD manually

because of this telegram plugin does not show /status and not showing when print start or finish

marcboivin, 2018-03-04:
It's in the work #2335

foosel, 2018-03-05:
I'm closing this since it's pretty much a duplicate of the aforementioned ticket, comments are still possible though.

closed 2018-03-05 by foosel