OctoPrint/OctoPrint #2461: Octoprint disconnects from the printer due to a "Error: Failed to enable Bed Leveling"but the console keeps receiving information about the print

valerie16901, 2018-03-05:
What were you doing?

Started a print from the printer SD card and then octoprint disconnected from the printer with an "Error: Failed to enable Bed Leveling", but it keeps receiving information from the printer.

Steps to reproduce the issue:

1. Load a file into the printer SD card.
2. Start the print from the file located on the SD, choosing it from the octoprint interface.
3. After the heat and nozzle is heated and the print starts the first layer octoprint disconnects from the printer, but the console keeps receiving information from the printer, this didn't happened on the previous version.

What did you expect to happen?

Octoprint should continue working and connected to the printer

What happened instead?

Octoprint disconnected with the following error:

"Unhandled communication error
There was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: Failed to enable Bed Leveling"

Did the same happen when running OctoPrint in safe mode?

Yes

Version of OctoPrint

1.3.6

Operating System running OctoPrint

OctoPi

Printer model & used firmware incl. version

Marlin 1.1.8

Browser and Version of Browser, Operating System running Browser

Safari on Mac OS X

Link to octoprint.log

There is none since I enabled the serial.log

Link to contents of terminal tab or serial.log

https://gist.githubusercontent.com/valerie16901/2633ddd266da1eda2489bb19a8c21e9d/raw/d060379125c11e93bb20d7991fc666929bf471ff/octoprint_log

Link to contents of Javascript console in the browser

https://gist.githubusercontent.com/valerie16901/58f90fb9847788d94b841b5f2d169f91/raw/957df15dfc57d3335039cba5387ae939dbe99820/gistfile1.txt

Screenshot(s)/video(s) showing the problem:

https://youtu.be/V02Ni-mP0tI

I have read the FAQ.

Yes

foosel, 2018-03-06:
That OctoPrint disconnects on an unhandled error sent by the printer is expected behaviour. You can change this behaviour in Settings > Serial Connection:

[image]

That you say that it continues to display stuff data in the terminal sounds like a bug however, it should disconnect fully. To look further into this though I need a `serial.log` or at the very least a copy of your terminal contents which are still missing from the above report. Please provide them.

valerie16901, 2018-03-06:
OK, here are the logs:

Serial.log
https://gist.githubusercontent.com/valerie16901/c5ba6ea3e65548134da5bab7eb537b20/raw/313eaf821e173474000da2f8f310cd2da957df77/serial.log

Octoprint.log
https://gist.githubusercontent.com/valerie16901/e241ad29b1ef78b9e63a15463d62ac61/raw/27b2e5055102afdee9edd27f4ac275315eb1b890/octoprint.log

foosel, 2018-03-06:
What do you have configured in your serial settings for the two marked options in my screenshot above?

valerie16901, 2018-03-06:
I had the first one checked, but I will make a test unchecking that one, to see what's happens.

foosel, 2018-03-06:
Never mind, I was just able to reproduce.

valerie16901, 2018-03-06:
Ok, but now that I unchecked the first box, the printer just home X and Y and then stops, it never starts the print. it's that a normal behaviour?

I have the logs again here, updated with this last problem:

serial.log:
https://gist.githubusercontent.com/valerie16901/93fe4ab87054e2232217c1baf893a2e4/raw/34159b422c69fccab930fe22b105a42446ff8d9e/serial.log_2

octoprint.log
https://gist.githubusercontent.com/valerie16901/3eb9499f38c80758a49bd6185b794a48/raw/c1272893eef72c0a062b88de53f85249243174a4/octoprint.log_2

foosel, 2018-03-06:
Yes... unhandled errors reported by the firmware by default lead to a disconnect. If you uncheck that box, they will only cancel the print. If you also check the checkbox below with the fat warning, they'll be ignored. The latter is not recommended.

During normal operation, your printer should *never* report an error. The fact that it does means either you need to fix something on your printer or you need to look into your firmware config and disable it sending catastrophic errors for issues that can be safely ignored.

foosel, 2018-03-06:
Fixed by the above commit on `maintenance`, soon `devel`, to be released with 1.3.7

closed 2018-04-09 by foosel

lily257, 2018-10-15:
Hello. Thank you for the interesting post. I want to say that I have been programming for a long time. I think, to solve this problem, you should look here https://amrama.com/technoexpert/best-printers-for-mac A bug in the firmware is really common.

commit de0660f6 (2018-03-06): Force disconnect on error state

Fixes #2461