OctoPrint/OctoPrint #2464: Uploaded STL file doesn't seem to be HTTP compressed (gzip, etc)

eyal0, 2018-03-06:
#### What were you doing?

Uploading larges file.

#### What did you expect to happen?

I expected the uploaded files to be sent compressed 

#### What happened instead?

The uploaded files were sent without HTTP compression

#### Version of OctoPrint
OctoPrint 1.3.5.20171121 (master branch)

#### Operating System running OctoPrint

OctoPi

#### Browser and Version of Browser, Operating System running Browser

Chrome

I have read the FAQ.

---
Maybe it's just in my setup.  Are files compressed in transit?  I sent two STL files, one in binary form, one in text.  Both are large but the text file is about 4 times larger.  When gzipped, both files are about the same size.  For that reason, I would expect that if the upload were compressed, the upload time would be about the same for each.  However, the upload time is very slow for the text STL file.

Does the server have HTTP compression enabled?

foosel, 2018-03-07:
Gzip compression for uploads is rather unusual and as far as I know not natively supported on upload elements by browser or the used file upload widget, so things would have to be zipped up completely in the browser and the unzipped inside Tornado on the server. At least the unzipping would cause additional load, meaning it would be something that needs to be investigated whether it causes issues during printing, in which case uploads while printing would need to be disabled. Finally I don't see the traffic here being the most pressing concern (the time you'd save in upload would prolly be taken up by first zipping client side, then unzipping server side). 

All this considered I don't see that it would be worth looking into at the present time, at least not from my side (hands already more than full). Feel free though to experiment with this and if it looks worthwhile opening a PR.

eyal0, 2018-03-11:
Yup, looks like it's difficult and uncommon.  Never mind!

Maybe it would be possible to have the server accept .zip or .gz files and decompress them?  That's a bigger project!

Compression savings on binary STL files are pretty good and compression savings on text STL are of course, very very good.

closed 2018-03-11 by eyal0

mdshw5, 2019-08-31:
I've just found this issue while checking if PKZIP archive uploads have ever been discussed. I could imagine allowing .zip archive upload, using python's `ZipFile` to extract the contents verbatim, including any directories. This could be especially convenient for Thingiverse archives.

eyal0, 2019-08-31:
Would be nicer to make it a plugin, I think.

mdshw5, 2019-08-31:
Oh nice idea. I’ll look into it.