OctoPrint/OctoPrint #2465: My new printer (FLSUN QQ) will not print from Octoprint unless in safe mode

Meridian64, 2018-03-07:
----

#### What were you doing?

Octoprint connects to the printer but when I upload a gcode file and hit Print the heatbed will heat up and when it reaches the correct temperature the nozzle will not heat up and the heatbed temperature will drop again and nothing happens. 

Also I cannot upload the file to sd card as I used to with my previous  printed. ( I cannot access the sd card at all through Octoprint)

If I manually copy the same file to sd card and transfer it to printer the files prints without problem

#### Did the same happen when running OctoPrint in safe mode?

In safe mode it will start printing but after a few minutes it will stop.

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0 

#### Operating System running OctoPrint

Windows 10

#### Printer model & used firmware incl. version

FLSUN QQ

#### Browser and Version of Browser, Operating System running Browser

Firefox 58.0.2 64bit

#### Link to octoprint.log

https://gist.github.com/Meridian64/37f20c96098e8aa5554abf48901b4d51

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2018-03-07:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

kantlivelong, 2018-03-07:
To add onto that. The octoprint.log appears to be funky version of a serial(or terminal?) log. Please correct this.

Meridian64, 2018-03-07:
Working on it now….

foosel, 2018-03-27:
@Meridian64 The serial output you provided is a bit... limited. There's basically nothing in there except two state changes and a ton of temperature responses from your firmware. It doesn't even show your issue.

Please enable safe mode, then [enable `serial.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299), connect to your printer, reproduce the problem and provide the resulting `serial.log`.

closed 2018-05-03 by foosel