OctoPrint/OctoPrint #2467: job_on_hold allows next file command to queue

FormerLurker, 2018-03-08:
----

#### What were you doing?

First, I hope I provided enough information.  I DID read the FAQ and templates, but it's a whole lot to digest, so I apologize in advance if I forgot anything.

I used job_on_hold in queueing phase in a thread.  Here is some pseudocode that should give you the gist:

`
gcode_queuing(self, cmd, cmd_type, gcode,tags):

    def task_to_run(cmd):

        with self._printer..job_on_hold(blocking=False OR True):

            do_something_that_takes_a_while()

            # send the command we postponed earlier

            _printer.commands(cmd)

    thread = threading.Thread(target=task_to_run, args=[cmd])

    thread.daemon = True

    thread.start()

    return None, # This successfully suppresses the command that was just queued

`

I thought at first this was due to the home command being intercepted by Octoprint, but I'm pretty sure this is not the case because I'm seeing the same issue later in the file.  I also thought that maybe the command is queued BEFORE reaching the job_on_hold call, but that's not the case either, as is confirmed by the log (see this entry, attached via git gist later in the ticket: A position altering command has been detected.  Fetching and updating position.  Position Command: G80).

If you want to see the plugin code, let me know and I'll push it up.  It's kind of a mess right now since I'm ripping out the guts so I can use the job_on_hold.

#### What did you expect to happen?
I expected the above code to do the following:

1.  Start the thread
2.  Lock the current job
3.  At some point after step 2, suppress the orignal cmd via return (None , ) 
4.  Execute the call do_something_that_takes_a_while()
5.  Eventually send the the original cmd via the _printer.commands(cmd) call
6.  Release the lock to continue processing the gcode file

#### What happened instead?
What actually happens:
1.  The thread starts
2.  The lock is acquired
3.  The original cmd that was queuing is suppressed
4.  The call to do_something_that_takes_a_while() starts 
**5.  The next line of the gcode file is queued!  I've never seen more than one slip through here.  It looks like it works perfectly aside from this.
6.  The original cmd is sent via the _printer.commands(cmd)
7.  The lock is released and the rest of the file is processed

Here is example excerpt from my gcode file.
G28 W
G80
G92 E0.0
M203 E100

Here's what I see as this gcode gets executed:

1.  (G28 W)  gets queued and sent
2.  (G80) gets queued, but enters our locked section (the pseudo code above)
3.  I get a log entry that the thread has started and is inside the lock section.  I can verify that it is the G80 command, which is what I expected
4.  Somewhere around here G80 is suppressed, not sure exactly when yet, but I'll find out if you want.
5.  The G92 E0.0 line gets queued, but definitely after I'm in the the job_on_hold section.  This is the problem
6.  M400 + M114 is sent via _printer.commands
7.  M400 queues
8.  The G92 E0.0 line gets sent to the printer.
9.  M114 is queued
10. M400 is send the virtual printer
11. M114 is sent to the virtual printer 
12. I receive the position from the M114 command
13. G80 is queued via a call to _printer.commands
14. The job_on_hold section is exited
15. G80 is sent to the printer and the file continues to process normally

#### Did the same happen when running OctoPrint in safe mode?

I cannot run OctoPrint in safe mode because I am testing out a plugin.  However, there are no other plugins installed.

#### Version of OctoPrint

OctoPrint 1.3.7.dev106+g6a3f1d28

#### Operating System running OctoPrint

Windows 10

#### Printer model & used firmware incl. version

Virtual Printer (haven't had the guts to try it on my actual printer)

#### Browser and Version of Browser, Operating System running Browser

NA - no browser issues at all

#### Link to octoprint.log

The octoprint log is very large and unhelpful (doesn't contain details about the plugin), so I attached the complete console.log for the debug session, which includes the octoprint logs for the session:

https://gist.github.com/FormerLurker/62b41d541d3c0712abe75aa57c19ddfb

However, that is pretty large too.  I also cut out the relevant section illustrating the gcode sequence I explained above from the console.log, which can be found here:  

https://gist.github.com/FormerLurker/2e80d17920523956ff797c3fcfe6b7c0

You'll notice that the tagging is working great!  Thanks for that!

#### Link to contents of terminal tab or serial.log

https://gist.github.com/FormerLurker/175baeb224b89540f452d43ebf1caacb

#### Link to contents of Javascript console in the browser

NA - no UI issues

#### Screenshot(s)/video(s) showing the problem:

NA - nothing to screenshow

I have read the FAQ.

foosel, 2018-03-08:
Yep, there was still a bug in there causing one more command to be sent. I've fixed that. While experimenting with my implementation of your pseudo code I then noticed though that it could happen that the thread was still only started after the next line was already sent, simply due to execution order. So I changed things around a bit.

Additionally to the `printer.job_on_hold` context manager that you already know there's now also a `printer.set_job_on_hold(value)` that you can use to explicitly set or unset the `job_on_hold` flag. 

Example `test_job_on_hold.py`:

``` python
# coding=utf-8
from __future__ import absolute_import

import octoprint.plugin

import time
import threading

class TestJobOnHoldPlugin(octoprint.plugin.OctoPrintPlugin):

	def handle(self, comm_instance, phase, cmd, cmd_type, gcode, tags=None, *args, **kwargs):
		if gcode != "G80" or (tags and "plugin:test_job_on_hold" in tags):
			return

		def process():
			try:
				self._logger.info("Going to sleep for a bit")
				time.sleep(20)
				self._logger.info("... and back!")

				# now send the command
				self._printer.commands(cmd, tags=tags)
			finally:
				self._printer.set_job_on_hold(False)

		if self._printer.set_job_on_hold(True):
			thread = threading.Thread(target=process)
			thread.daemon = True
			thread.start()
			return None,

__plugin_implementation__ = TestJobOnHoldPlugin()
__plugin_hooks__ = {
	"octoprint.comm.protocol.gcode.queuing": __plugin_implementation__.handle
}
```

Combined with a GCODE file consisting of the lines you provided that produces this output:

```
2018-03-08 15:18:27,204 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing"
2018-03-08 15:18:27,210 - octoprint.util.comm.command_phases - DEBUG - phase: queuing | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.reset_line_numbers, trigger:comm.start_print ]
2018-03-08 15:18:27,210 - octoprint.util.comm.command_phases - DEBUG - phase: queued | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.reset_line_numbers, trigger:comm.start_print ]
2018-03-08 15:18:27,213 - octoprint.util.comm.command_phases - DEBUG - phase: sending | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.reset_line_numbers, trigger:comm.start_print ]
2018-03-08 15:18:27,213 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-03-08 15:18:27,219 - octoprint.util.comm.command_phases - DEBUG - phase: sent | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.reset_line_numbers, trigger:comm.start_print ]
2018-03-08 15:18:27,242 - octoprint.util.comm.command_phases - DEBUG - phase: queuing | command: M117 Starting... | gcode: M117 | tags: [ script:beforePrintStarted, source:script, trigger:comm.send_gcode_script, trigger:printer.script ]
2018-03-08 15:18:27,242 - octoprint.util.comm.command_phases - DEBUG - phase: queued | command: M117 Starting... | gcode: M117 | tags: [ script:beforePrintStarted, source:script, trigger:comm.send_gcode_script, trigger:printer.script ]
2018-03-08 15:18:27,243 - octoprint.util.comm.command_phases - DEBUG - phase: sending | command: M117 Starting... | gcode: M117 | tags: [ script:beforePrintStarted, source:script, trigger:comm.send_gcode_script, trigger:printer.script ]
2018-03-08 15:18:27,243 - octoprint.util.comm.command_phases - DEBUG - phase: sent | command: M117 Starting... | gcode: M117 | tags: [ script:beforePrintStarted, source:script, trigger:comm.send_gcode_script, trigger:printer.script ]
2018-03-08 15:18:27,249 - octoprint.util.comm.command_phases - DEBUG - phase: queuing | command: G28 W | gcode: G28 | tags: [ api:job, fileline:1, filepos:7, source:api, source:file, trigger:comm.start_print, trigger:printer.start_print ]
2018-03-08 15:18:27,250 - octoprint.util.comm.command_phases - DEBUG - phase: queued | command: G28 W | gcode: G28 | tags: [ api:job, fileline:1, filepos:7, source:api, source:file, trigger:comm.start_print, trigger:printer.start_print ]
2018-03-08 15:18:27,250 - octoprint.util.comm.command_phases - DEBUG - phase: sending | command: G28 W | gcode: G28 | tags: [ api:job, fileline:1, filepos:7, source:api, source:file, trigger:comm.start_print, trigger:printer.start_print ]
2018-03-08 15:18:27,255 - octoprint.util.comm.command_phases - DEBUG - phase: sent | command: G28 W | gcode: G28 | tags: [ api:job, fileline:1, filepos:7, source:api, source:file, trigger:comm.start_print, trigger:printer.start_print ]
2018-03-08 15:18:27,259 - octoprint.util.comm.command_phases - DEBUG - phase: queuing | command: G80 | gcode: G80 | tags: [ fileline:2, filepos:12, source:file ]
2018-03-08 15:18:27,259 - octoprint.plugins.test_job_on_hold - INFO - Going to sleep for a bit
2018-03-08 15:18:30,938 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-08 15:18:35,269 - octoprint.util.comm.command_phases - DEBUG - phase: queuing | command: M117 I'm sent in the middle through the terminal tab | gcode: M117 | tags: [ api:printer.command, source:api, trigger:printer.commands ]
2018-03-08 15:18:35,269 - octoprint.util.comm.command_phases - DEBUG - phase: queued | command: M117 I'm sent in the middle through the terminal tab | gcode: M117 | tags: [ api:printer.command, source:api, trigger:printer.commands ]
2018-03-08 15:18:35,272 - octoprint.util.comm.command_phases - DEBUG - phase: sending | command: M117 I'm sent in the middle through the terminal tab | gcode: M117 | tags: [ api:printer.command, source:api, trigger:printer.commands ]
[... cut ...]
2018-03-08 15:18:47,312 - octoprint.util.comm.command_phases - DEBUG - phase: sending | command: M117 Done! | gcode: M117 | tags: [ script:afterPrintDone, source:script, trigger:comm.send_gcode_script, trigger:printer.script ]
2018-03-08 15:18:47,316 - octoprint.util.comm.command_phases - DEBUG - phase: sent | command: M117 Done! | gcode: M117 | tags: [ script:afterPrintDone, source:script, trigger:comm.send_gcode_script, trigger:printer.script ]
2018-03-08 15:18:47,331 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Operational"
```

closed 2018-03-08 by foosel

FormerLurker, 2018-03-08:
That's great, I'll test it out over lunch.  Thank you so much for reviewing this so quickly, and thanks again for the new features!

I thought about the same thing regarding locking the job inside of the new thread.  It looks like your solution is simple fix for it.

BTW, the time it takes to test a print in the virtual printer with this new implementation has decreased by a significant amount.  I'm not exactly sure why.  It seems to have something to do with calling _printer.pause, though.  I am very grateful for this side effect since it's normal for me to run dozens of tests!  Also, my timelapse.py file is getting skinny, and I love it!  Real-time gcode processing here we come!

foosel, 2018-03-09:
> I figured you'd want to know about this

Oh definitely, thanks for spotting this! Too much going back and forth between approaches while testing and I got confused.

Should be fixed by 4c3e44d50e6f34c20053569c9f64d6a110807320

commit 10af4328 (2018-03-08): comm: fix job_on_hold & add further options for using it

  * next command could be queued even if job was already on hold due to
    a check missing
  * context manager was too limiting, could lead to race conditions
    where the job_on_hold flag wasn't set in time when moved to a
    handler thread, introducing a dedicated set_job_on_hold method
    for that solves this (at the downside of requiring more care from
    plugin developers utilizing this)

Fixes #2467