OctoPrint/OctoPrint #2468: 'module' object has no attribute 'Timeout'

FormerLurker, 2018-03-08:
#### What were you doing?

I installed the latest maintenance branch to my raspberry pi from instructions I found in the wiki.  The installation went well, and Octoprint came back up.  However, I get an error when trying to connect to my Prusa MK2.

Exact steps to reproduce:

1. Install OctoPrint 1.3.7.dev130+g02c42ec on OctoPi 0.13.0
2. Attempt to connect to a printer (Prusa MK2 in this case)
3. An error is generated.

#### What did you expect to happen?

An error free connection

#### What happened instead?

I got the following error:  

2018-03-08 22:04:03,418 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.7.dev130_g02c42ec-py2.7.egg/octoprint/util/comm.py", line 2358, in _readline
    ret = self._serial.readline()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.7.dev130_g02c42ec-py2.7.egg/octoprint/util/comm.py", line 4186, in readline
    timeout = serial.Timeout(self._timeout)
AttributeError: 'module' object has no attribute 'Timeout'

#### Did the same happen when running OctoPrint in safe mode?

Yes, I got the same error in safe mode.

#### Version of OctoPrint

OctoPrint 1.3.7.dev130+g02c42ec running on OctoPi 0.13.0

#### Operating System running OctoPrint

OctoPi 0.13.0

I updated the distro and all modules.  Everything appears to be running the latest version.

#### Printer model & used firmware incl. version

Prusa MK2 Latest Firmware

#### Browser and Version of Browser, Operating System running Browser

NA - happens in all browsers I've tried

#### Link to octoprint.log

https://gist.github.com/FormerLurker/ad5d29a8dddc20f939217848925b24f1

#### Link to contents of terminal tab or serial.log

https://gist.github.com/FormerLurker/4a13166e0f689da36de28ebaa05bc6cb

#### Link to contents of Javascript console in the browser

NA

#### Screenshot(s)/video(s) showing the problem:

NA

FormerLurker, 2018-03-08:
I'm not sure what I missed.  Bad bot!

foosel, 2018-03-09:
You missed the last line ;)

Thanks for reporting this... Current maintenance allows two versions of pyserial, and I tested against the wrong one. Fixed by the above commit, sorry for the hassle.

closed 2018-03-09 by foosel

FormerLurker, 2018-03-09:
This solved my issue!  My first test is printing with all the new functionality, and so far it's going great!  Thank you so much!

FormerLurker, 2018-03-09:
It worked perfectly!  I printed 3 calibration towers:  

1. with the old plugin
2. without the plugin
3. with the new plugin version that doesn't pause

Version 3 is way closer to version 2 than it is to version 1!  Even with a (poorly configured) bowden setup.  I am super happy.  I think this has fixed a bunch of issues for me (need to test all to confirm), which means V1.0 is much closer than I'd thought!

Maybe this belongs somewhere else, but Is there a way that I can submit this plugin to the repository so that it's only offered if the user has your next RC version installed?  If so, I think I will do this.  It will make the instructions on how to install much better.

Thanks again!  Today is a good day!

foosel, 2018-03-09:
I haven't tested it with RCs in a long while, but you can define a minimum version in the `.md` file. Setting that to `1.3.7` or `1.3.7rc1` might already do the trick. Remind me on monday and I can give it a quick test.

FormerLurker, 2018-03-09:
Will do.  I'll make sure I have the .md changed and everything up-to-date by Monday.  

I'm sure you're used to this by now since you've been working on Octoprint for so long, but I'm having a hard time containing my excitement about a V1 release.  Actually, having it limited to RC1 will probably make the roll-out a lot smoother.  Everything seems to be falling into place.

commit a2ccf414 (2018-03-09): comm: compat fix for pyserial 2.7 vs 3.4

Closes #2468