OctoPrint/OctoPrint #2474: WIFI fine, but no internet connection to install plugins

dseebauer, 2018-03-10:
#### What were you doing?

Attempting to install plugins via the plugin manager.   

#### What did you expect to happen?

#### What happened instead?

Octoprint reports no internet connection.    I was able to verify via SSH that Octopi/Raspberian(?) does in fact have an internet connection using PING, I can get a connection to any external site via both IP and and name resolution (i.e. google.com).  WIFI connection is seemingly fine as I can run the printer without a problem via any browser on the local network.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0 (running on Pi Zero W)

#### Printer model & used firmware incl. version

Rostock Max V2, Repetier 0.91

#### Browser and Version of Browser, Operating System running Browser

Chrome 64.64.0.3282.186 

#### Link to octoprint.log

https://pastebin.com/g6YBrEuZ

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

foosel, 2018-03-12:
What do you have configured under Settings > Server > Connectivity Check? Because it looks like whatever you have configured there isn't reachable by OctoPrint and hence it has to assume it's offline. Could be caused by some firewall configuration. If push comes to shove, disable the connectivity check by unchecking the box that says "enable regular connectivity check" and save. OctoPrint will then assume it's always online.

dseebauer, 2018-03-12:
Thanks you very much for the response.   Disabling the connectivity check did fix the repository access.   The check was configure to use google (8.8.8.8) on port various ports (22, 53, 5000, etc.   Not sure the default).   I do not have any firewall set up outgoing, unless it is buried in the Raspberian install that I am not aware of.  I saw a couple others with same issue out there and a re-flash was suggested, which may indicate something is setup/changed incorrectly in Raspberian.

foosel, 2018-03-12:
It could be a firewall at ISP level. Rasbpian/OctoPi doesn't come with anything like that included and in fact the connectivity check on stock OctoPi works flawlessly under normal circumstances. My money would definitely be on some kind of filtering on your network being the cause here. As I said, that could also be outside of your control at ISP level.

closed 2018-03-12 by foosel

IntegersOfK, 2018-05-26:
I stumbled across this issue with the same problem, and I fixed the connectivity check by changing the IP to OpenDNS with 1.1.1.1 instead of the Google DNS 8.8.8.8 set by default.

tubbers, 2018-07-02:
I have the same issue https://discourse.octoprint.org/t/octoprint-claims-to-not-have-a-connection-to-the-internet-but-it-does/177 
I turrned off the check and instantly I got notified of 3 months of updates. During the time of the issue the pi was able to access the internet (Astroprint plugin for Octo worked fine)

CoyoteProd, 2018-07-11:
For me this issue was resolved when I update time on the raspberry...

AndrewDarkicorn, 2019-07-17:
> 
> 
> I stumbled across this issue with the same problem, and I fixed the connectivity check by changing the IP to OpenDNS with 1.1.1.1 instead of the Google DNS 8.8.8.8 set by default.

For the record, 1.1.1.1 (and 1.0.0.1) is operated by CloudFlare, not OpenDNS (208.67.222.222 and 208.67.220.220).