OctoPrint/OctoPrint #2475: [Request] Enable microstepping controls during print

dseebauer, 2018-03-11:
One of the features found in some other print controllers is microstepping during a print, particularly with the Z axis.    This allow fine tuning the print especially during the initial layer to improve layer quality and bed adhesion.    In the case of my printer, the calibration drifts every time it goes idles and the temperatures cycle (off then on after a period)  

Enabling the controls for the Z-axis during a print would allow a user to "adjust" the Z height live as the print starts.  Effectively creating a "Z-offset"  value, which would need to be displayed somewhere visible (status panel?)

To implement, it seems this could be as simple as keeping track of the "Z-offset" as it is produced from the use of the controls, and then intercepting the g-code line just prior to it being sent to the printer, parsing out the z-height, doing the math and replacing the z-axis with the offset incorporated.  The "Z-offset" could be retained for future prints, with a reset function for when it is no longer needed.  I suppose an alternative implementation could be a text field that could allow direct entry of the value. 

To be most effective this would require that the general controls include a finer adjustment value vs. what is currently in place.   Suggest adding  .02 at a minimum.

e.g. pressing the Z-down control arrow twice with the .02 distance enabled would produce a -0.040 offset.  This would then effect all subsequent gcode going to printer (with a Z value), e.g. : 
 
Original clip:    G1 Z0.500 F12000.000
Modified line:  G1 Z0.460 F12000.000

ntoff, 2018-03-11:
>To implement, it seems this could be as simple as keeping track of the "Z-offset" as it is produced from the use of the controls

Not if the file was sliced with relative movements, in which case you could microstep up, and then just not care anymore because all the movements are relative to the current position. OctoPrint would have to detect whether or not the file was sliced with absolute or relative positions. That's something that used to give me no end of issues with most host softwares as I used relative positioning and things like gcode viewers and movement would always assume absolute, thus screwing things up in unpredictable ways.

>To be most effective this would require that the general controls include a finer adjustment value vs. what is currently in place. Suggest adding .02 at a minimum.

Check out the extra distance plugin https://plugins.octoprint.org/plugins/extradistance/ which I made for just that reason :)

dseebauer, 2018-03-12:
Thanks for the feedback,   I just added your plug in and it would be perfect for that purpose as you said.  I was not familiar with slicers using relative movements.   Could an explicit check for a G90/G91 in the file be sufficient to base a decision on?  Do slicers switch back and forth throughout?   I believe my firmware (repetier 0.91)  assumes absolute, but my personal start gcode explicitly sets it as well.    If a print is using relative movements throughout (i.e. a single explicit G91), than the feature would be simply throwing out a z movement in the increment/direction requested.  Otherwise, it would work as described for absolute.   Enabling or disabling the feature could be dependent on the explicit presence of a single  G91/G90 at the top of the gcode file when it loads, but maybe that is getting too specific for general purpose use.

foosel, 2018-03-12:
The problem is that actually rewriting all coordinates send to the printer will be a lot of processing overhead. For each and every line containing a rewritten coordinate you'd have to parse that value out, calculate the new one, then create a new command based on that. If you only have to do that for Z that should be manageable since you usually don't constantly change that (vase mode prints excluded). However with X and Y I fear it would cause so much calculation overhead that you'd seriously impact print performance.

Maybe `M290` or `M206` would be an option here, but I don't know if Repetier Firmware implements those.

In any case, a feature like this makes way more sense to implement through coordinate shift on the firmware side instead of rewriting every single `G0` and `G1` line live during printing.

ntoff, 2018-03-12:
> I don't know if Repetier Firmware implements those.

http://reprap.org/wiki/G-code#M290:_Babystepping
http://reprap.org/wiki/G-code#M206:_Offset_axes

no, and no. M206 in repetier sets EEPROM values so you definitely don't want to go messing with that :)

dseebauer, 2018-03-18:
Hi ntoff,
I have been looking into the plugin approach, but given I am pretty far from being a programmer, it is pretty confusing to me.   Am I wrong that an approach very similar to you FanSpeedContol plugin could be adapted to this task.  It looks like in that plugin your are essentially doing what I had thought could work in that you are scanning the gcode and re-writing the m106 command to control the fan speed up and down against limits.   For this purpose the control could be limited to just an offset value, and none of the reporting would be necessary.  Thoughts?

ntoff, 2018-03-18:
M106 is usually a code on a line all by itself, G1/G0 Xnnn Ynnn Znnn Ennn could have any combination of X, Y, Z, and E commands. Most of the time you might find the Z on a line all by itself which would be fairly easy to find, but if a slicer were put into vase mode then Z is added to every command because Z constantly increases. 

I'm not sure what kind of effect that would have on performance, scanning and doing math work on every single line of gcode. I mean, you are just adding one number to an other so I don't think it's going to require a super computer but on low powered devices, every millisecond matters. Some people already experience performance issues with gcode that has a lot of very small movements.

I have a few plugins / doodads / widgets that I haven't registered / released because they only work "well enough for me" in very specific scenarios, if you felt like having a go at it then by all means do so. I'm not a programmer either, I'm more at home with a welding torch in my hand than a book on Javascript. Just have a go at it, ~~steal~~ borrow my fan speed plugin's code if it makes it any easier to get started and don't even ask me how you'd figure out the regex code to find out which number you'd need to modify if Z isn't on its own, regexp is moon language to me but I use this site to help figure it out: https://www.regexpal.com/ and test whether the regexp works.

mikekscholz, 2021-08-22:
```
controls:
- children:
  - children:
    - additionalClasses: getLiveZ
      command: M290
      confirm: null
      name: Get
    - additionalClasses: saveLiveZ
      command: M500
      confirm: null
      name: Save
    layout: horizontal
  - default: 'Current Z Offset: ???'
    regex: Probe Offset ?Z:?([-+]?[0-9.]+)
    template: 'Current Z Offset: {0}mm'
  - children:
    - additionalClasses: babystepUp
      command: M290 Z0.02
      confirm: null
      name: Babystep Up
    - additionalClasses: babystepDown
      command: M290 Z-0.02
      confirm: null
      name: Babystep Down
    - additionalClasses: microstepUp
      command: M290 Z0.01
      confirm: null
      name: Microstep Up
    - additionalClasses: microstepDown
      command: M290 Z-0.01
      confirm: null
      name: Microstep Down
    layout: horizontal
  layout: vertical
  name: Live-Z Probe Offset
```