OctoPrint/OctoPrint #2476: Octoprint Connect/Disconnect button disappeared

CanadianSnowman, 2018-03-12:
Hello All,

I had just completed installation of Octoprint and my printer was connecting and then I click the automatically connect box and since then the Connect/Disconnect button disappeared completely with the automatic connect option and the State of my printer is constantly offline.  It will not connect to the printer anymore and I don't know how to revert to the default settings.

Thanks

hashashin, 2018-03-12:
Hi @CanadianSnowman, assuming you're on linux edit: `/home/<put_your_user_here>/.octoprint/config.yaml`
and look for the serial config. 
It should look something like this depending of your settings:
```
serial:
  additionalBaudrates:
  - 1000000
  - 500000
  baudrate: 250000
  autoconnect: true
  port: AUTO
```
Delete or change to "false" the autoconnect line, then restart octoprint.

I hope this helps you, salud.

foosel, 2018-03-12:
For questions or other kind of help or support, please get in touch on the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2018-03-12 by foosel