OctoPrint/OctoPrint #2479: Problems with Logitech Webcam C270

raresbozdog, 2018-03-14:
Hi I'm using this camera on my RPI 3 model B and it's not working. It's tested camera but for me is not working even with guvcview. I am using also a tested powered usb hub D-LINK DUB-H4 and have all updated. Please help !

george-vieira, 2018-03-15:
Works for me, but I'm using an old PI B+. try a clean raspbian installation and see if it works on the same hardware

foosel, 2018-03-15:
For questions or other kind of help or support, please get in touch on the [community forum](https://discourse.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2018-03-15 by foosel

raresbozdog, 2018-03-15:
@george-vieira I tried but still not working....

george-vieira, 2018-03-24:
then this isn't octopi issue.