OctoPrint/OctoPrint #2482: Plugin Blacklist for TPLinkSmartplug

stampede-arm, 2018-03-17:
Seeking Clarification/Rectification of the Plugin Blacklist for OctoPrint-TPLinkSmartplug
It appears that (according to the last few comments) this serious issue was fixed and the plugin should no longer be on blacklist?
[Link to Issue 19 on Plugin's Github](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/19)

jneilliii, 2018-03-17:
I'm not completely sure, but from what I can tell the blacklist listing is for specific versions of the plugin, and specific version of OctoPrint.  This is to prevent people from installing an older version of the plugin I guess.

https://plugins.octoprint.org/blacklist/

stampede-arm, 2018-03-17:
Shoot, I didn't see that column on that page. Didn't realize it was for specific versions of the plugin. Closing this issue as that appears to be correct.

closed 2018-03-17 by stampede-arm