OctoPrint/OctoPrint #2483: PRINT_STARTED event - first gcode sent before event received

FormerLurker, 2018-03-17:
#### What were you doing?

1. I received the event PRINT_STARTED
2. I immediately call _printer.set_job_on_hold(True)
3. Some code runs, then I call _printer.set_job_on_hold(False)
4. The first Gcode I receive in on_gcode_queuing is actually line 2 of the gcode file.

Note:  I log all of the queued gcode starting after step three (before I release the job lock).  I DO receive the gcode queuing message for the first line, but it happens before the print started event gets called.

Thank you for reading!

#### What did you expect to happen?

I expected one of the following two things:
1.  I expected to see line 1 of the gcode file in step 4
2.  If item 1 is not possible, I would like to request a PRINT_STARTING event that is raised and finishes before any gcode is queued.

#### What happened instead?

I see line 2 of the gcode file

#### Did the same happen when running OctoPrint in safe mode?

I cannot run in safe mode because I am testing a plugin, but no other plugins are installed.

#### Version of OctoPrint

OctoPrint 1.3.7.dev132+g4c3e44d5

#### Operating System running OctoPrint

Windows 10

#### Printer model & used firmware incl. version

Virtual Printer

#### Browser and version of browser, operating system running browser

NA

#### Link to octoprint.log
[Link to octoprint.log, where G21 is the first line and M107 is on line 2](
https://gist.github.com/FormerLurker/e33fa5d57b8e0516df63740ea97e811b)

#### Link to contents of terminal tab or serial.log

[Link to terminal output](https://gist.github.com/FormerLurker/e5128ca62cf947b03b14c87084df92dd)

#### Link to contents of Javascript console in the browser

NA
#### Screenshot(s)/video(s) showing the problem:

NA

I have read the FAQ.

foosel, 2018-03-19:
The problem here is that event processing happens asynchronously, so even a `PRINT_STARTING` event would not solve this. Events aren't able to block processing in any way (that would actually be bad in most cases). This is actually what you are seeing here, `PRINT_STARTED` is fired inside `on_comm_print_started`, right before the first line of the actual job file gets enqueued:

https://github.com/foosel/OctoPrint/blob/8cdae335a2a8f21d7daad6d565c7a6c089bc377d/src/octoprint/util/comm.py#L1029-L1066

https://github.com/foosel/OctoPrint/blob/8cdae335a2a8f21d7daad6d565c7a6c089bc377d/src/octoprint/printer/standard.py#L1126-L1132

I can't change that without causing a lot of issues elsewhere (and open a whole can of worms with regards to misbehaviour from plugins). I can look into adding a hook, but not still in time for 1.3.7 because I'm actually about to finish the first RC of that.

So let's find a different solution: As you can see above the first command that is sent by starting a print has the tags `trigger:start_print` and `source:file`. Might you be able to do something with this?

FormerLurker, 2018-03-19:
I figured there was a good reason for the current behavior, and thanks for looking it over!  I should study comm.py a bit more, but it kind of frightens me :)  Thanks for highlighting the relevant sections, this made it easier to see.

Your idea regarding the first gcode command is an interesting one.  Maybe I can make my own print starting event this way.  I'll post some code here if/when I get it worked out!

foosel, 2018-03-19:
> I should study comm.py a bit more, but it kind of frightens me :)

That makes two of us 😆

FormerLurker, 2018-03-19:
This (slightly modified for general use) snippet seems to work:

```
on_gcode_queuing(self, cmd, cmd_type, gcode, tags):
        # detect print start
        if ( self_printer.is_printing() and
            'trigger:printer.start_print' in tags
        ):
            if self._printer.set_job_on_hold(True):
                try:
                    self.on_print_start(tags)
                    # resend the command while we have the job lock.  This is important else the
                    # second gcode will be queued immediately after releasing the lock, causing it
                    # to be sent before the first gcode!
                    self._printer.commands(cmd)
                    # suppress the current command so it doesn't get sent twice
                    return None,
                finally:
                    self._printer.set_job_on_hold(False)
            else:
                self.on_print_start_failed(
                    "Unable to acquire the job lock, cannot run on_print_start.."
                )

```

foosel, 2018-03-19:
Perfect, closing this then if you don't mind :)

closed 2018-03-19 by foosel

FormerLurker, 2018-03-19:
I've been testing and have found some interesting things.  If you use the above code, and you are printing from SD, things get hairy.  Several gcodes are sent that are marked as 'comm.start_print' or 'printer.start_print' AFTER I send _printer.cancel_print().  This isn't such a huge deal because I'm not interested in printing from SD.  However, I am interested in warning the user about printing from the SD, and canceling the print (depending on the user's settings so it is not intrusive).  With the above code I was getting several warning notifications as Octolapse attempted to start each time it saw comm.start_print.

What I did was add the following check:  `set(['trigger:comm.start_print', 'fileline:1', 'api:job']) <= tags`

This is only true if printing locally.  I then use the standard PRINT_STARTING event and check the payload for `payload["origin"] != "local"`, at which point I cancel the print.  Several gcodes slip by and are sent to the printer, however.  Any ideas how to do the following:

Detect print start.  If printing from SD, cancel and prevent any gcodes from being sent to the printer.

Here is what I'm trying now, but it doesn't prevent gcodes from being sent to the printer:

```
                if (
            self.Settings.is_octolapse_enabled and
            self.State == TimelapseState.Idle and
            self.OctoprintPrinter.is_printing() and
            set(['trigger:comm.start_print', 'api:job']) <= tags
        ):
            self.Settings.current_debug_profile().log_print_state_change(
                "Print Start Detected.  Command: {0}, Tags:{1}".format(cmd, tags)
            )
            if self.OctoprintPrinter.set_job_on_hold(True):
                try:
                    if 'fileline:1' not in tags:

                        self.OctoprintPrinter.cancel_print()
                    else:
                        self.on_print_start(tags)
                        # resend the command while we have the job lock.  This is important else your first
                        # two gcodes will run out of order.
                        if self.State == TimelapseState.WaitingForTrigger:
                            self.OctoprintPrinter.commands(cmd)
                        # suppress the current command so it doesn't get sent twice
                    return None,
                finally:
                    self.OctoprintPrinter.set_job_on_hold(False)
            else:
                self.on_print_start_failed(
                    "Octolapse was unable to start.  Unable to acquire the job lock."
                )
```

Let me know if there's an easy way to accomplish this.  Otherwise things seem to be working!

FormerLurker, 2018-03-19:
I forgot, here is the terminal log:
_______________
```
Changing monitoring state from "Operational" to "Printing from SD"
Changing monitoring state from "Printing from SD" to "Cancelling"
Recv: ok
Send: M23 /abs_20~1.gco
Recv: File opened: abs_20~1.gco  Size: 333605
Recv: File selected
Recv: ok
Send: M24
Recv: ok
[...]
Recv: ok
Send: M25
Recv: ok
[...]
Recv: ok
Send: M26 S0
Recv: ok
Send: M400
Recv: ok
Send: M114
[...]
Recv: ok X:210.0 Y:0.0 Z:0.25 E:100.0 Count: A:21000 B:0 C:25
Send: M84
Send: M104 T0 S0
Recv: ok
Send: M140 S0
Recv: ok
Send: M106 S0
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
Recv: ok
```
___________________________

I guess my only problem here really is that M24 is being sent and M25 right after.  It would be best if M24 is never sent in this case.

foosel, 2018-03-19:
Careful with checking for `fileline:1` - if your file contains comments at the beginning, line 1 will never be sent.

FormerLurker, 2018-03-19:
I think you are thinking about filepos, not fileline, right?  I've been testing it with comments and such at the beginning of the file, and it seems to always work.  Am I missing something?

foosel, 2018-03-19:
It's working for you because it's currently a bit buggy... `fileline` should be the actual line in the file. Right now it's not, it's counting actually returned lines only, after filtering out comments and such. I'll have to change that.

FormerLurker, 2018-03-19:
Can you add another tag then for GcodeLineNumber or something similar using the current method?  This bug is very good for me, lol!

foosel, 2018-03-19:
`filesentline` maybe?

FormerLurker, 2018-03-19:
Sounds great!  I'll add some conditional code to detect filesentline:1 so that it will work as soon as the tag is added.

foosel, 2018-03-19:
Hm... hold your horses for a bit, I just realized that I had forgotten about file seeks and that I can't actually even know the exact line in the file in all cases. So maybe I'll just leave it be as is now after all and give up on the idea to have `fileline` reflect the actual line number.

FormerLurker, 2018-03-19:
Too late, but its not a problem since I used this as my test:

`'fileline:1' not in tags and 'filesentline:1' not in tags:`

I assume this issue is due to resends?  Sounds to me like you'd need to keep a list of fileoffsets and match them with line numbers as you're going.  You could store them in a queue and pop them as necessary to keep the list short.  Sounds like a pain in the butt.

foosel, 2018-03-19:
Nah, the problem is rather since in principle I support something like "start printing file x at byte position y". Right now I just `seek` to that position. If I want to track the actual in-file line number I'd have to read it line by line and count. That's... not really a good option. So we'll just leave it for now as is. If I SHOULD change it after all, you'll be prepared I guess thanks to the above code ;)

The resend request handling is actually even worse of a pain in the butt thanks to not being able to rely on the order of `ok`s received from the firmware matching the order of sent commands. Ancient firmware, since fixed, but before that propagated through a couple hundred or thousand forks that didn't get the fix. Don't get me started on this topic ;)

FormerLurker, 2018-03-19:
That makes a lot of sense.  I won't get you started regarding the comm layer :)  Let me know when you start on your comm.py re-write.  I'd be interested in helping where I can.