OctoPrint/OctoPrint #2487: Using custom system commands with a script that do not return stalls the OctoPrint Web Interface

tiosolid, 2018-03-18:
#### What were you doing?

Creating a new command in the system menu while using a script that does not return (I.E. have something like an `while True:` inside it) stalls the interface since it waits forever for the `/api/system/commands/custom` request to finish.

Tried to run the python script using a bash sh file that `exit 0` but it still didnt work. I need this infinite loop since I'm using PWM and the script needs to run until killed via another script (I used a "flag" file so the scripts end gracefully).

#### Stuff that I tried (and didn't work)
- Running the .py script directly in the config.yaml;
- Running the .py script using a .sh script (like `python /home/pi/script.py &`);
- Running the .py script in the config.yaml using `python /home/pi/script.py &`;

Source code of the "start" script (running this stalls the interface forever):
```
import RPi.GPIO as IO
import time
import os.path

IO.setwarnings(False)
IO.setmode (IO.BCM)
IO.setup(18,IO.OUT)
p = IO.PWM(18,100)

open("/home/pi/lighton.flag", "a").close()
p.start(100)
p.ChangeDutyCycle(100)
while os.path.isfile("/home/pi/lighton.flag"):
        pass

p.ChangeDutyCycle(0)
p.stop()
```

Source code of the "stop" script (this one always works and does not stall the web interface):
```
import os

if os.path.isfile("/home/pi/lighton.flag"):
        os.remove("/home/pi/lighton.flag")
```

Source code of the bash script:
```
#!/bin/bash

python /home/pi/lighton.py &
exit 0
```

Source code of the relevant part of my config.yaml file:
```
system:
  actions:
  - action: streamon
    command: sudo service webcamd start
    confirm: false
    name: Start video stream
  - action: streamoff
    command: sudo service webcamd stop
    confirm: false
    name: Stop video stream
  - action: lighton
    command: /home/pi/lighton.sh
    confirm: false
    name: Turn printer lights on
  - action: lightoff
    command: python /home/pi/lightoff.py
    confirm: false
    name: Turn printer lights off
```

Note that running these scripts manually from the terminal **works**.

#### What did you expect to happen?
Normal system command behavior (run and return a green success message since the script do not raise an exception).

#### What happened instead?
Interface is stalled forever while waiting for the request to finish.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint
OctoPi with a Raspberry Pi 3.

#### Browser and version of browser, operating system running browser
Chrome 65.0 in Windows 10 64-bits

> Note: I removed all the other stuff about the printer and logs from the bug report since nothing different is logged and not having a printer connected does not change this behavior.

foosel, 2018-03-18:
You need to set the `async` flag to true for commands that don't return. Needs to be documented. Turning this into a documentation request.

tiosolid, 2018-03-18:
Using `async` worked flawlessly. Thanks very much!

PR #2498 (2018-03-20): Document the async flag of custom system commands
Fixes #2487

  * [x] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [ ] If your changes are large or otherwise disruptive: You have
    made sure your changes don't interfere with current development by
    talking it through with the maintainers, e.g. through a
    Brainstorming ticket
  * [x] Your PR targets OctoPrint's devel branch, or maintenance if it's
    a bug fix for an issue present in the current stable version (no PRs
    against master or anything else  [... cut]
commit 819d4d0f (2018-03-20): Expand the description of custom system commands

Fixes #2487
closed 2018-07-25 by foosel