OctoPrint/OctoPrint #2492: [RC Feedback] Feedback on the 1.3.7rc1 Release Candidate

foosel, 2018-03-19:
Please provide *general* feedback on your experience with the 1.3.7rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

b-morgan, 2018-03-20:
I have upgraded to 1.3.7rc1 and my first print is progressing nicely with one exception. 

The plugin "Detailed Progress Plugin" isn't outputting anything. It is supposed to inject M117 commands into the GCode stream. I've opened an issue against the plugin. Should I open one against OctoPrint as well?

kantlivelong, 2018-03-20:
@b-morgan Was it working in 1.3.6? I'd suggest providing the octoprint.log file in the [ticket you opened](https://github.com/dattas/OctoPrint-DetailedProgress/issues/26).

b-morgan, 2018-03-20:
@kantlivelong : Yes it was working on 1.3.4, 1.3.5, and 1.3.6. I uploaded the octoprint.log file as you suggested but I don't believe there is anything useful in the log.

I did, however, find another exception! I go to Settings, Logging and click the download icon next to octoprint.log. Instead of downloading the file, I get "500: Internal Server Error".

I guess that one deserves a ticket.

chippypilot, 2018-03-20:
Thanks for the RC. I like the sound of some of the new features.
To test this version I have created a fresh install using the latest stable link on the octopi page. 2017-06-25_2017-06-21-octopi-jessie-lite-0.14.0 on a Raspberry 3

I then  ran apt-get etc. and updated octoprint to this RC. I have connected to my <whisper>Anet A6</whisper> and have run a first print OK.

Auto connection works better now. Previously, I had to manually set the baudrate

Issues noted so far:
> It has detected my printer as Generic RepRap, not sure if the A6 is detectable.
> I have confirmed that the log download action (mentioned above by b morgan) produces the 500 error every time I tried it

Will continue testing over the next few days.

mrhanman, 2018-03-21:
I've printed several small things and been through all of the UI.  So far, I haven't encountered any issues.  The auto connection feature works for me now.

I do have a question about the Printer Safety Check plugin.  Does it run automatically on connection, and only display something if there is a problem detected, or is it meant to be run manually?  If it's the latter, then I can't seem to find it in the interface.

akurz42, 2018-03-21:
I'm seeing the exact same as @b-morgan: "Detailed Progress" and "Display ETA" (probably all plugins that try to display stuff on the LCD) don't seem to work anymore. Tested on Marlin 1.1.8 and Prusa MK3 3.1.3.

akurz42, 2018-03-21:
Apart from the above:

All is working fine :)

kantlivelong, 2018-03-21:
RE DetailsProgress 

It looks like the introduction of using a `frozendict` for printer state in 088b4a8059ecc172969971be1809b3e453f77f79 is causing the issue.

andrivet, 2018-03-21:
So far so good with RC1. Printer: Wanhao i3 Plus, Firmware: ADVi3++ (based on Marlin 1.1.8).
Note: Contrary to some people, I **do not have** problem with M117 messages. However, I have tested with the "Display Progress" plugin, not the "Detailed Progress" one.

akurz42, 2018-03-21:
Unfortunately I can't confirm @andrivet's  observation. None of the two plugins gives me any info during the print. The "Display Progress" plugin puts a "[##########]" on the display after the print is finished, though.

akurz42, 2018-03-21:
Another glitch I found on both my printers: when locally reset (press button on printer), Octoprint is correctly reporting that "A Reset has been detected", but state stays on "Canceling" until I disconnect/reconnect the printer.

foosel, 2018-03-22:
@akurz42 could you please create a bug report for the reset-vs-cancelling glitch?

foosel, 2018-03-22:
About the Detailed Progress issue. On first glance that plugin is doing something it shouldn't be doing, but I'll have to take a closer look and see if that can't be worked around again. Turning those internal state `dict`s into unmutable `frozendict`s is an attempt at narrowing down on the illusive #1951. 

About the logging 500, I think I already know the issue there. See #2501 which @b-morgan already was so kind to open.

foosel, 2018-03-22:
Hah... I think the fact that the Detailed Progress plugin is doing that modification there (which now fails due to the `frozendict` being in place) is actually the cause of #1951 😄

foosel, 2018-03-22:
@akurz42 might be that I have found it until then, currently narrowing it down. Might simply be because it tries to run the cancel script on external reset, won't switch from cancelling to cancelled until it has done that, but running it fails because of the line number reset caused by the printer reset.

akurz42, 2018-03-22:
@foosel: Sorry for not being of more help here, but I'm down to doing some actual work today and tomorrow. I may still be able to squeeze it in :)

foosel, 2018-03-22:
@akurz42 that wasn't meant as "why you no have time earlier!!1!" ;) just wanted you to know to best take another look here before you dive into it since that might safe you from log collecting and such if I've already found it.

akurz42, 2018-03-22:
@foosel: no offence taken :) I'll make sure to look here before diving into the thick of it.

foosel, 2018-03-22:
@b-morgan hoping for tomorrow. But depends a bit on this cancelling issue which so far is being stubborn.

If you want to test what's in the works beforehand you'll have to [manually switch to the `staging/maintenance` branch](https://discourse.octoprint.org/t/how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image/206) (that's where I'm preparing 1.3.7rc2, `maintenance` has turned into 1.3.8.dev with the release of 1.3.7rc1).

foosel, 2018-03-23:
@akurz42 I think the cancelling issue will stay a known bug for now. It's a corner case which should only arise when stuff is broken already anyhow thanks to the printer restart, and solving it involves more changes (with potential to introduce severe issues) than I feel comfortable right now. Getting something wrong here could cause severe communication issues.

I've got an approach, but it needs more fine tuning (still have to find a reliable solution for some race conditions). So this will have to wait until 1.3.8.

akurz42, 2018-03-23:
@foosel: thanks for dealing with this anyway. I don't think it's a crippling issue. A local reset is a last resort measure, so this shouldn't pop up every day and reconnecting isn't a problem either.