OctoPrint/OctoPrint #2493: [Request] iframe support for webcam view

ardenpm, 2018-03-19:
Currently webcam streams are handled as a configurable img tag in the main view of OctoPrint. This limits streaming to mjpg and there are many requests for additional features around webcams which probably shouldn't really be part of core OctoPrint, however plugin access to the webcam part may take some time. I would like to suggest that the stream URL setting be extended to offer two options:

1. MJPEG Stream URL (same as now)
2. IFRAME Stream URL

The second option would replace the img tag with an iframe tag of the given URL and then allow the user to easily place arbitrary content inside that area of the user interface.

While the user would need to be able to write their on HTML pages (or potentially use ones built into their cameras), for many this may be more straight forward than writing an OctoPrint plugin (or also more general if they are monitoring the stream multiple ways).

Given the wide range of different methods of getting video to the client-side, this one option would at least make it possible (if not simple) to implement them without further OctoPrint development. Trying to handle, for example, native video tags, WebRTC, WebSocket video streaming and others directly in OctoPrint would be much more work.

This could also potentially allow multiple webcam support using mostly existing tools for combining streams (at least on Linux I know that is feasible). It would still be necessary to have a webpage and server to serve this but I would imagine with this feature implemented some guides would spring up fairly quickly.

ntoff, 2018-03-19:
more or less a duplicate of https://github.com/foosel/OctoPrint/issues/2001

foosel, 2018-03-19:
I'd rather say an extension.

ardenpm, 2018-03-19:
Perhaps adding:

3. VIDEO Stream URL

Being the src property of a video tag would be helpful then. While obviously this could also be done inside an iframe if it's popular enough (sounds like there is definitely interest based on the discussion) this would be a good additional option.

One thing to note is that video tags are not live (especially when using HLS or DASH) and in almost all cases incorporate significant latency (often multiple seconds). Personally I would find that a pretty significant issue (particularly when jogging axis remotely) but I can see cases where you'd still want it.

Of course a totally general approach would be to let the user specify the tag and the properties of the tag (or just the entire tag string) but this feels like overkill.

jneilliii, 2018-04-20:
Going to take a stab at this plugin over the weekend.

jneilliii, 2018-04-21:
I've got the initial framework for multiple camera views and types. I know it's not exactly what you were trying to get to, but it might be a step in the right direction.

https://github.com/jneilliii/OctoPrint-MultiCamView

foosel, 2019-02-07:
Closing as duplicate of #2001 after all.

closed 2019-02-07 by foosel